MichaIng/DietPi #7516: Whiptail theme

guillaumeraffi, 2025-05-03:
#### Describe the solution you'd like:
Use the website color scheme in the whiptail menus
-

MichaIng, 2025-05-05:
Oh wow, I was not aware that it is even possible to color, but it is, via `NEWT_COLORS` environment variable: https://pagure.io/newt/blob/master/f/newt.h

Here some examples: https://gist.github.com/ymkins/bb0885326f3e38850bc444d89291987a
Here the defaults: https://pagure.io/newt/blob/master/f/newt.c

One problem is that for any defaults we're bound to at least the 4-bit ANSI color codes, to assure it works on all terminals: https://en.wikipedia.org/wiki/ANSI_escape_code#3-bit_and_4-bit
Though I see also "brown" in the source code in one place, which is not 4-bit ANSI color 🤔. Generally the question is how these color names are applied, and whether also other codes are possible than the ones we see in the source code/examples.

MichaIng, 2025-05-11:
I guess some general `whiptail` theming would be great: #7528

We would provide a default color palette, the mono colors and implement some `whiptail`-based UI to adjust element colors individually, applying the result right away.

guillaumeraffi, 2025-05-13:
[image] have also been playing with some logos, let me know if you are interested