MichaIng/DietPi #7517: NVMe SSD in M.2 slot not detected on NanoPi R5S

eggcllnt, 2025-05-04:
## Issue Description
After installing DietPi on my NanoPi R5S, the M.2 SSD that was previously detected and working properly under FriendlyWRT is no longer visible. The device doesn't appear in lsblk output, despite being physically installed and previously functional.

## System Information
Device: NanoPi R5S
Original OS: FriendlyWRT (SSD was detected)
Current OS: DietPi
Kernel version: 6.6.56-current-rockchip64

### Steps Taken

Verified SSD isn't showing up in basic disk utilities:

```bash
dietpi@DietPi:~$ lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk1      179:0    0 28.9G  0 disk 
|-mmcblk1p1  179:1    0  1.1G  0 part 
`-mmcblk1p2  179:2    0    1M  0 part 
mmcblk1boot0 179:32   0    4M  1 disk 
mmcblk1boot1 179:64   0    4M  1 disk 
mmcblk0      179:96   0 59.6G  0 disk 
`-mmcblk0p1  179:97   0 59.6G  0 part /
```

Checked if NVMe kernel modules are available:

```bash
dietpi@DietPi:~$ find /lib/modules/$(uname -r) -name "*nvme*"
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/target/nvmet.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/target/nvmet-tcp.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/target/nvme-fcloop.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/target/nvme-loop.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/target/nvmet-fc.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/host/nvme-fabrics.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/host/nvme-tcp.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvme/host/nvme-fc.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvmem
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvmem/nvmem-rmem.ko
/lib/modules/6.6.56-current-rockchip64/kernel/drivers/nvmem/nvmem_qcom-spmi-sdam.ko
```

Checked if NVMe modules are loaded:

```bash
dietpi@DietPi:~$ lsmod | grep -E 'nvme|ahci|sata|scsi'
[No output]
```

Tried manually loading NVMe modules:

```bash
dietpi@DietPi:~$ sudo modprobe nvme
dietpi@DietPi:~$ sudo modprobe nvme-core
dietpi@DietPi:~$ lsblk
[SSD still not detected]
```

Examined PCIe status in dmesg:

```bash
dietpi@DietPi:~$ sudo dmesg | grep -i -E 'pcie|nvme'
[    0.847519] rockchip-dw-pcie 3c0000000.pcie: host bridge /pcie@fe260000 ranges:
[    0.847617] rockchip-dw-pcie 3c0000000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    0.847669] rockchip-dw-pcie 3c0000000.pcie:      MEM 0x00f4200000..0x00f5ffffff -> 0x00f4200000
[    0.847709] rockchip-dw-pcie 3c0000000.pcie:      MEM 0x0300000000..0x033fffffff -> 0x0040000000
[    0.847994] rockchip-dw-pcie 3c0000000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    1.055022] rockchip-dw-pcie 3c0000000.pcie: PCIe Gen.2 x1 link up
[    1.055776] rockchip-dw-pcie 3c0000000.pcie: PCI host bridge to bus 0000:00
[    1.082499] pcieport 0000:00:00.0: PME: Signaling with IRQ 22
[    1.083143] pcieport 0000:00:00.0: AER: enabled with IRQ 22
[    2.191280] rockchip-dw-pcie 3c0400000.pcie: host bridge /pcie@fe270000 ranges:
[    2.191359] rockchip-dw-pcie 3c0400000.pcie:       IO 0x00f2100000..0x00f21fffff -> 0x00f2100000
[    2.191429] rockchip-dw-pcie 3c0400000.pcie:      MEM 0x00f2200000..0x00f3ffffff -> 0x00f2200000
[    2.191469] rockchip-dw-pcie 3c0400000.pcie:      MEM 0x0340000000..0x037fffffff -> 0x0040000000
[    2.203145] rockchip-dw-pcie 3c0400000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    2.410930] rockchip-dw-pcie 3c0400000.pcie: PCIe Gen.2 x1 link up
[    2.411180] rockchip-dw-pcie 3c0400000.pcie: PCI host bridge to bus 0001:00
[    2.434544] pcieport 0001:00:00.0: PME: Signaling with IRQ 73
[    2.435110] pcieport 0001:00:00.0: AER: enabled with IRQ 73
[    2.436425] rockchip-dw-pcie 3c0800000.pcie: host bridge /pcie@fe280000 ranges:
[    2.436498] rockchip-dw-pcie 3c0800000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[    2.436550] rockchip-dw-pcie 3c0800000.pcie:      MEM 0x00f0200000..0x00f1ffffff -> 0x00f0200000
[    2.436590] rockchip-dw-pcie 3c0800000.pcie:      MEM 0x0380000000..0x03bfffffff -> 0x0040000000
[    2.446555] rockchip-dw-pcie 3c0800000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    3.551144] rockchip-dw-pcie 3c0800000.pcie: Phy link never came up
[    3.551360] rockchip-dw-pcie 3c0800000.pcie: PCI host bridge to bus 0002:00
[    3.564057] pcieport 0002:00:00.0: PME: Signaling with IRQ 75
[    3.564570] pcieport 0002:00:00.0: AER: enabled with IRQ 75
```

Checked available PCI devices:

```bash
dietpi@DietPi:~$ sudo lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
```

Checked available PCIe and NVMe-related device tree overlays:

```bash
dietpi@DietPi:~$ find /boot/ -name "*nvme*.dtbo" -o -name "*m2*.dtbo" -o -name "*pcie*.dtbo"
/boot/dtb-6.6.56-current-rockchip64/rockchip/overlay/rockchip-rk3568-hk-pwm2.dtbo
/boot/dtb-6.6.56-current-rockchip64/rockchip/overlay/rockchip-rk3399-pcie-gen2.dtbo
```

Modified kernel parameters in DietPi configuration to try enabling PCIe:

```
extraargs=net.ifnames=0 pcie_aspm=off pci=nomsi
```

After reboot, the issue persists with the same error message:

```
[    3.539542] rockchip-dw-pcie 3c0800000.pcie: Phy link never came up
```

### Observations

The M.2 SSD worked properly under FriendlyWRT on the same hardware
NVMe kernel modules are available in DietPi but not automatically loaded
Manually loading NVMe modules does not resolve the issue
PCIe port 3c0800000.pcie reports "Phy link never came up" which appears to be the port connected to the M.2 slot
Two RTL8125 Ethernet controllers are properly detected on PCIe buses 0000:00 and 0001:00
No device appears to be detected on PCIe bus 0002:00, which is likely the one connected to the M.2 slot

### Questions

Are there specific device tree configurations needed for NanoPi R5S M.2 slot in DietPi?
Is there a known issue with DietPi's PCIe initialization for the NanoPi R5S M.2 slot?
Are there specific boot parameters or overlays that need to be enabled for M.2 NVMe support?

Any assistance would be greatly appreciated.

MichaIng, 2025-05-06:
I pushed new kernel and bootloader packages to our APT repo. Could you try it with those:
```sh
sudo apt update
sudo apt update
sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
sudo reboot
```

eggcllnt, 2025-05-07:
@MichaIng thanks for looking into that issue!
I've updated my existing system with the instractions that you've provided, but there were no result. Same missing nvme drive.

I've reinstalled system from scratch. It worked as expected, booted, updated, but no nvme again. One weird thing happened is it was rebooted and after ethernet ports were unresponsive. so, I've powered off, then powered on and things booted as usual asking to optout data collection. No nvme though.

I've added to /boot/dietpiEnv.txt `extraargs` `pci=realloc pcie_aspm=off` params, rebooted and it showed up. But this behaviour I was observed and previously.

### logs from successful boot

```bash
root@DietPi:~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk1      179:0    0  28.9G  0 disk 
└─mmcblk1p1  179:1    0  28.9G  0 part /
mmcblk1boot0 179:32   0     4M  1 disk 
mmcblk1boot1 179:64   0     4M  1 disk 
nvme0n1      259:0    0 465.8G  0 disk 
└─nvme0n1p1  259:1    0 465.8G  0 part
```

```bash
root@DietPi:~# dmesg | grep -i -E 'pcie|nvme'
[    0.000000] Kernel command line: root=UUID=43620ab3-751b-4f33-91a2-14ca0a8027d3 rootfstype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0 pci=realloc pcie_aspm=off
[    0.000000] PCIe ASPM is disabled
[    0.146022] /pcie@fe260000: Fixed dependency cycle(s) with /pcie@fe260000/legacy-interrupt-controller
[    0.174869] /pcie@fe270000: Fixed dependency cycle(s) with /pcie@fe270000/legacy-interrupt-controller
[    0.175797] /pcie@fe280000: Fixed dependency cycle(s) with /pcie@fe280000/legacy-interrupt-controller
[    1.979652] rockchip-dw-pcie 3c0000000.pcie: host bridge /pcie@fe260000 ranges:
[    1.979752] rockchip-dw-pcie 3c0000000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    1.979805] rockchip-dw-pcie 3c0000000.pcie:      MEM 0x00f4200000..0x00f5ffffff -> 0x00f4200000
[    1.979847] rockchip-dw-pcie 3c0000000.pcie:      MEM 0x0300000000..0x033fffffff -> 0x0040000000
[    1.980111] rockchip-dw-pcie 3c0000000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    2.176964] rockchip-dw-pcie 3c0000000.pcie: PCIe Gen.2 x1 link up
[    2.177683] rockchip-dw-pcie 3c0000000.pcie: PCI host bridge to bus 0000:00
[    2.177920] pci 0000:00:00.0: [1d87:3566] type 01 class 0x060400 PCIe Root Port
[    2.187782] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000 PCIe Endpoint
[    2.193190] pcieport 0000:00:00.0: PME: Signaling with IRQ 22
[    2.193768] pcieport 0000:00:00.0: AER: enabled with IRQ 22
[    4.171029] rockchip-dw-pcie 3c0400000.pcie: host bridge /pcie@fe270000 ranges:
[    4.171793] rockchip-dw-pcie 3c0400000.pcie:       IO 0x00f2100000..0x00f21fffff -> 0x00f2100000
[    4.172629] rockchip-dw-pcie 3c0400000.pcie:      MEM 0x00f2200000..0x00f3ffffff -> 0x00f2200000
[    4.173586] rockchip-dw-pcie 3c0400000.pcie:      MEM 0x0340000000..0x037fffffff -> 0x0040000000
[    4.182905] rockchip-dw-pcie 3c0400000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    4.382822] rockchip-dw-pcie 3c0400000.pcie: PCIe Gen.2 x1 link up
[    4.383749] rockchip-dw-pcie 3c0400000.pcie: PCI host bridge to bus 0001:00
[    4.388854] pci 0001:00:00.0: [1d87:3566] type 01 class 0x060400 PCIe Root Port
[    4.411384] pci 0001:01:00.0: [10ec:8125] type 00 class 0x020000 PCIe Endpoint
[    4.429076] pcieport 0001:00:00.0: PME: Signaling with IRQ 79
[    4.430234] pcieport 0001:00:00.0: AER: enabled with IRQ 79
[    4.451745] rockchip-dw-pcie 3c0800000.pcie: host bridge /pcie@fe280000 ranges:
[    4.452473] rockchip-dw-pcie 3c0800000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[    4.453756] rockchip-dw-pcie 3c0800000.pcie:      MEM 0x00f0200000..0x00f1ffffff -> 0x00f0200000
[    4.454583] rockchip-dw-pcie 3c0800000.pcie:      MEM 0x0380000000..0x03bfffffff -> 0x0040000000
[    4.463797] rockchip-dw-pcie 3c0800000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    4.853010] rockchip-dw-pcie 3c0800000.pcie: PCIe Gen.1 x1 link up
[    4.853877] rockchip-dw-pcie 3c0800000.pcie: PCI host bridge to bus 0002:00
[    4.857686] pci 0002:00:00.0: [1d87:3566] type 01 class 0x060400 PCIe Root Port
[    4.880205] pci 0002:01:00.0: [1344:5416] type 00 class 0x010802 PCIe Endpoint
[    4.882880] pci 0002:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0002:00:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    4.899225] pcieport 0002:00:00.0: enabling device (0000 -> 0002)
[    4.904000] pcieport 0002:00:00.0: PME: Signaling with IRQ 82
[    4.905363] pcieport 0002:00:00.0: AER: enabled with IRQ 82
[    4.907149] nvme nvme0: pci function 0002:01:00.0
[    4.907656] nvme 0002:01:00.0: enabling device (0000 -> 0002)
[    5.035223] nvme nvme0: allocated 64 MiB host memory buffer.
[    5.142951] nvme nvme0: 4/0/0 default/read/poll queues
[    5.163016]  nvme0n1: p1
```

```bash
[... 215 lines cut ...]
[    4.882880] pci 0002:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0002:00:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    4.885064] pci_bus 0002:01: busn_res: [bus 01-0f] end is updated to 01
[    4.885747] pci_bus 0002:00: max bus depth: 1 pci_try_num: 2
[    4.886329] pci 0002:00:00.0: BAR 0 [mem 0x380000000-0x3bfffffff]: assigned
[    4.886999] pci 0002:00:00.0: BAR 1 [mem size 0x40000000]: can't assign; no space
[    4.887699] pci 0002:00:00.0: BAR 1 [mem size 0x40000000]: failed to assign
[... cut ...]
[    4.896445] pci_bus 0002:00: resource 4 [io  0x200000-0x2fffff]
[    4.897049] pci_bus 0002:00: resource 5 [mem 0xf0200000-0xf1ffffff]
[    4.897657] pci_bus 0002:00: resource 6 [mem 0x380000000-0x3bfffffff]
[    4.898259] pci_bus 0002:01: resource 1 [mem 0xf0200000-0xf02fffff]
[    4.899225] pcieport 0002:00:00.0: enabling device (0000 -> 0002)
[    4.904000] pcieport 0002:00:00.0: PME: Signaling with IRQ 82
[    4.905363] pcieport 0002:00:00.0: AER: enabled with IRQ 82
[    4.907149] nvme nvme0: pci function 0002:01:00.0
```

StephanStS, 2025-05-18:
My R5S (2 GB RAM) with my NVMe works: Booting from an 8 GB SD card and using the NVMe as a mounted ext4 drive.

**lsblk give my nvme0n1 disk:**

```
root@dietpi:~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk1      179:0    0   7,3G  0 disk
└─mmcblk1p1  179:1    0   3,7G  0 part /
mmcblk1boot0 179:32   0     4M  1 disk
mmcblk1boot1 179:64   0     4M  1 disk
nvme0n1      259:0    0 238,5G  0 disk
└─nvme0n1p1  259:1    0 238,5G  0 part /mnt/nvme_1

```

Latest kernel:

```
root@dietpi:~# uname -r
6.12.25-current-rockchip64

```

**lspci gives the NVMe controller:** 

```
root@dietpi:~# lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983

```

dmesg output:

```
root@dietpi:~# dmesg | grep -i -E 'pcie|nvme'
[    0.143610] /pcie@fe260000: Fixed dependency cycle(s) with /pcie@fe260000/legacy-interrupt-controller
[    0.171861] /pcie@fe270000: Fixed dependency cycle(s) with /pcie@fe270000/legacy-interrupt-controller
[    0.172754] /pcie@fe280000: Fixed dependency cycle(s) with /pcie@fe280000/legacy-interrupt-controller
[    1.957517] rockchip-dw-pcie 3c0000000.pcie: host bridge /pcie@fe260000 ranges:
[    1.957634] rockchip-dw-pcie 3c0000000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    1.957686] rockchip-dw-pcie 3c0000000.pcie:      MEM 0x00f4200000..0x00f5ffffff -> 0x00f4200000
[    1.957725] rockchip-dw-pcie 3c0000000.pcie:      MEM 0x0300000000..0x033fffffff -> 0x0040000000
[    1.958067] rockchip-dw-pcie 3c0000000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    2.157522] rockchip-dw-pcie 3c0000000.pcie: PCIe Gen.2 x1 link up
[    2.158257] rockchip-dw-pcie 3c0000000.pcie: PCI host bridge to bus 0000:00
[    2.158484] pci 0000:00:00.0: [1d87:3566] type 01 class 0x060400 PCIe Root Port
[    2.168331] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000 PCIe Endpoint
[    2.181011] pcieport 0000:00:00.0: PME: Signaling with IRQ 22
[    2.181680] pcieport 0000:00:00.0: AER: enabled with IRQ 22
[    4.223495] rockchip-dw-pcie 3c0800000.pcie: host bridge /pcie@fe280000 ranges:
[    4.224220] rockchip-dw-pcie 3c0800000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[    4.225044] rockchip-dw-pcie 3c0800000.pcie:      MEM 0x00f0200000..0x00f1ffffff -> 0x00f0200000
[    4.225924] rockchip-dw-pcie 3c0800000.pcie:      MEM 0x0380000000..0x03bfffffff -> 0x0040000000
[    4.235450] rockchip-dw-pcie 3c0800000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    4.433517] rockchip-dw-pcie 3c0800000.pcie: PCIe Gen.3 x1 link up
[    4.433815] rockchip-dw-pcie 3c0800000.pcie: PCI host bridge to bus 0002:00
[    4.433934] pci 0002:00:00.0: [1d87:3566] type 01 class 0x060400 PCIe Root Port
[    4.444752] pci 0002:01:00.0: [144d:a808] type 00 class 0x010802 PCIe Endpoint
[    4.445729] pci 0002:01:00.0: 7.876 Gb/s available PCIe bandwidth, limited by 8.0 GT/s PCIe x1 link at 0002:00:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    4.456727] pcieport 0002:00:00.0: PME: Signaling with IRQ 79
[    4.458124] pcieport 0002:00:00.0: AER: enabled with IRQ 79
[    4.459238] nvme nvme0: pci function 0002:01:00.0
[    4.459283] nvme 0002:01:00.0: enabling device (0000 -> 0002)
[    4.462345] nvme nvme0: missing or invalid SUBNQN field.
[... 32 lines cut ...]
/boot/dtb-6.12.25-current-rockchip64/rockchip/overlay/rockchip-rk3588-pwm5-m2.dtbo
/boot/dtb-6.12.25-current-rockchip64/rockchip/overlay/rockchip-rk3588-pwm13-m2.dtbo
/boot/dtb-6.12.25-current-rockchip64/rockchip/overlay/rockchip-rk3588-pwm14-m2.dtbo
/boot/dtb-6.12.25-current-rockchip64/rockchip/overlay/rockchip-rk3588-uart7-m2.dtbo
/boot/dtb-6.12.25-current-rockchip64/rockchip/overlay/rockchip-rk3588-pwm6-m2.dtbo

```

**dietpiEnv.txt:**

```
root@dietpi:~# cat /boot/dietpiEnv.txt
rootdev=UUID=2a524dbc-55ed-4136-a984-bd87eb4222a5
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=ttyS2,1500000 console=tty1
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=

```

MichaIng, 2025-05-20:
Thanks for testing. Good to know it is no general issue.

@eggcllnt
I guess this is the essential problem:
```
[    4.886999] pci 0002:00:00.0: BAR 1 [mem size 0x40000000]: can't assign; no space
[    4.887699] pci 0002:00:00.0: BAR 1 [mem size 0x40000000]: failed to assign
```
Could you filter your `dmesg` output to include `pci` for a successful start as well, to check how it tries to assign the BAR memory in that case?

Seems to be some conflict for assigned resources: https://stackoverflow.com/questions/65241905/linux-pcie-preallocate-bar-space

Do you have 2 GB or 4 GB RAM?

MichaIng, 2025-05-29:
Since I was digging into this PCI(e) device tree nodes in another context, here the BAR address range and size defined in the device tree for this PCIe bus: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3568.dtsi?h=linux-6.12.y#n116

This part `<0x03000000 0x0 0x40000000 0x3 0x80000000 0x0 0x40000000>;`, the first being only the bit field, the second two (64-bit high and low value) the PCI bus address, the next two the CPU bus address to map to, and the last two the size of the segment to map to, which is 1 GiB. Seems like in some boot sessions this is too large 🤔. It is the same for all 3 PCIe busses (2x 2.5 Gbit Ethernet adapters, and 1x M.2 slot) in this last range block, all 1 GiB segment size from the same PCI bus address, but mapped to different CPU bus addresses, with an offset of 1 GiB one after another.

Others can probably debug this faster. But if I shall help, I need the full `dmesg` output of a positive and a negative boot session, to compare values (in case they differ) and order in which devices have been loaded, to narrow down which could block the needed space on the CPU bus,

If you do not need both 2.5 Gbit Ethernet adapters, we could also disable one and see whether this helps. We could also change the mapping of the M.2 slot to use the CPU bus address previously used by the disabled Ethernet adapter. But I am not sure whether it can simply be done like that, would be experimenting 😄.