MichaIng/DietPi #7519: Please add support Orange Pi RV2 (Riscv64)

misterhsp, 2025-05-05:
### Creating an image request
Please add support Orange Pi RV2 (Riscv64)

#### Formal device information
- Device name | OrangePi RV2
- Official product URL | http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-RV2.html
- Image download URL | http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-RV2.html

#### Is the SBC officially supported by the Debian installer?
I don't know

#### If not, is a reliable 3rd party Debian image available for this SBC?
I don't know

MichaIng, 2025-05-05:
Yeah, one just arrived. Though I will add support for the Orange Pi 5 Ultra/CM5/4A first.

misterhsp, 2025-05-06:
An image would be really good. The Ubuntu image available for download 
is really bad. Many broken dependencies. 
Someappscan'tbeinstalledbecausethedependenciesarebroken.

misterhsp, 2025-05-06:
An image would be really good. The Ubuntu image available for download 
is really bad. Many broken dependencies.
Some apps cannot be installed because the dependencies are broken.

MichaIng, 2025-06-09:
U-Boot source: https://github.com/orangepi-xunlong/u-boot-orangepi/tree/v2022.10-ky
Kernel sources: https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.6-ky

Not sure whether anyone is aiming for mainline kernel support. In case to monitor:
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/riscv/boot/dts

There seem to be no other board using this "Ky X1" SoC. Also I couldn't find any info about the SoC manufacturer, probably Xunlong own/exclusive production? Quite possible that this will never see mainline Linux or U-Boot support, under these circumstances, considering the efforts, and it being a single low-power SoC only. The StarFive JH7110 used in the Orange Pi RV (1) is as powerful, seems to have even more single-core power, and StarFive had a transparent agenda to mainline it right from the start, which has worked pretty well.

However, let's hope for the best. Until then, I'll add support to our Armbian fork, using the Xunlong sources as is.

fethomm, 2025-06-11:
From what I read, the Ky X1 is supposed to be identical with the SpacemiT K1 8, that powers the Banana Pi BPI-F3. But I can't be sure.

fethomm, 2025-06-11:
> Do you have a link for this info?
> 
> Would be good, since that Banana Pi BPI-F3 is supported already, hence basic drivers and stuff should be there:
> 
> * https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/riscv/boot/dts/spacemit
> * https://github.com/u-boot/u-boot/tree/master/arch/riscv/dts
> 
> With some luck, other board features can be ported via device tree only. However, for now, WiFi (due to missing driver in mainline Linux) and stuff like the NPU will work with vendor kernel only anyway. So it is just about efforts and hence changes this board gets mainline support, or at least whether mainline support can be patched with reasonable efforts.

I can't find the original quote, but on page 6 of https://www.androidpimp.com/embedded/orange-pi-rv2/6/ it says: "After thorough examination, we can confidently assert that the RV2 Chipset is crafted by SpacemiT, an emerging Chinese company dedicated to the innovation and development of high-performance RISC-V CPUs." 

Another review at https://boilingsteam.com/orange-pi-rv2-new-risc-v-board-review/ mentions: "The Ky X1 processor appears to be the same as the Key Stone K1 chip from the chinese company SpacemiT"

romanrm, 2025-08-17:
> create sid/trixie image (need root)

@misterhsp thank you for your posts here, I made a [Trixie image](https://romanrm.net/rv-debian) for this board just now.