MichaIng/DietPi #7522: tinkerboard-2 changes MAC address each boot after upgrade linux-dtb-current-rockchip64 linux-image-current-rockchip64

briney83, 2025-05-06:
### Creating a bug report/issue
- [ *] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.12.1
- Distro version | echo $G_DISTRO_NAME $G_RASPBIAN
- Kernel version | 6.12.25-current-rockchip64
- SBC model | ASUS Tinker Board 2 (aarch64)
- Power supply used | 19V 2.37A
- SD card used | Samsung 128Gb Class10 UHS-I U3+

#### Additional Information (if applicable)

#### Steps to reproduce

1. sudo dietpi-update 1

#### Expected behaviour

- MAC address should not be changed

#### Actual behaviour

- Every boot changes IP address and MAC address

#### Extra details

- dietpi@tinkerboard-2:~$ cat /var/tmp/dietpi/logs/dietpi-upgrade_apt.log 
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  linux-dtb-current-rockchip64 linux-image-current-rockchip64
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 49.3 MB of archives.
After this operation, 77.7 MB of additional disk space will be used.
Get:1 https://dietpi.com/apt all/tinkerboard2 arm64 linux-dtb-current-rockchip64 arm64 25.05.0-trunk-dietpi2 [766 kB]
Get:2 https://dietpi.com/apt all/tinkerboard2 arm64 linux-image-current-rockchip64 arm64 25.05.0-trunk-dietpi2 [48.5 MB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 49.3 MB in 6s (7633 kB/s)
(Reading database ... 22064 files and directories currently installed.)
Preparing to unpack .../linux-dtb-current-rockchip64_25.05.0-trunk-dietpi2_arm64.deb ...
Armbian 'linux-dtb-current-rockchip64' for '6.12.25-current-rockchip64': 'preinst' starting.
Armbian 'linux-dtb-current-rockchip64' for '6.12.25-current-rockchip64': 'preinst' finishing.
Unpacking linux-dtb-current-rockchip64 (25.05.0-trunk-dietpi2) over (24.11.0-trunk-dietpi1) ...
Preparing to unpack .../linux-image-current-rockchip64_25.05.0-trunk-dietpi2_arm64.deb ...
Armbian 'linux-image-current-rockchip64' for '6.6.56-current-rockchip64': 'prerm' starting.
Armbian 'linux-image-current-rockchip64' for '6.6.56-current-rockchip64': 'prerm' finishing.
Armbian 'linux-image-current-rockchip64' for '6.12.25-current-rockchip64': 'preinst' starting.
Armbian 'linux-image-current-rockchip64' for '6.12.25-current-rockchip64': 'preinst' finishing.
Unpacking linux-image-current-rockchip64 (25.05.0-trunk-dietpi2) over (24.11.0-trunk-dietpi1) ...
Armbian 'linux-image-current-rockchip64' for '6.6.56-current-rockchip64': 'postrm' starting.
Removing obsolete initramfs images
Armbian 'linux-image-current-rockchip64' for '6.6.56-current-rockchip64': 'postrm' finishing.
Setting up linux-image-current-rockchip64 (25.05.0-trunk-dietpi2) ...
Armbian 'linux-image-current-rockchip64' for '6.12.25-current-rockchip64': 'postinst' starting.
Removing obsolete initramfs images
removed '/boot/uInitrd-6.6.56-current-rockchip64'
removed '/boot/initrd.img-6.6.56-current-rockchip64'
update-initramfs: Generating /boot/initrd.img-6.12.25-current-rockchip64
update-initramfs: Converting to U-Boot format
Image Name:   uInitrd
Created:      Tue May  6 22:18:13 2025
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    14876041 Bytes = 14527.38 KiB = 14.19 MiB
Load Address: 00000000
Entry Point:  00000000
'/boot/uInitrd' -> 'uInitrd-6.12.25-current-rockchip64'
Armbian: update last-installed kernel symlink to 'Image'...
'/boot/Image' -> 'vmlinuz-6.12.25-current-rockchip64'
Armbian: Debian compat: linux-update-symlinks install 6.12.25-current-rockchip64 boot/vmlinuz-6.12.25-current-rockchip64
Armbian 'linux-image-current-rockchip64' for '6.12.25-current-rockchip64': 'postinst' finishing.
Setting up linux-dtb-current-rockchip64 (25.05.0-trunk-dietpi2) ...
Armbian 'linux-dtb-current-rockchip64' for '6.12.25-current-rockchip64': 'postinst' starting.
Armbian: DTB: symlinking /boot/dtb to /boot/dtb-6.12.25-current-rockchip64...
'dtb' -> 'dtb-6.12.25-current-rockchip64'
Armbian 'linux-dtb-current-rockchip64' for '6.12.25-current-rockchip64': 'postinst' finishing.
[  OK  ] DietPi-Update | APT upgrade

MichaIng, 2025-05-06:
I think I found it. Guess it is this commit: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?id=5d90cb1edcf7c1854e4cecb52871421f29d3d849

This alias does not exist, does it?
```sh
cat /proc/device-tree/aliases/ethernet0; echo
```

briney83, 2025-05-06:
cat: /proc/device-tree/aliases/ethernet0: Нет такого файла или каталога

MichaIng, 2025-05-06:
I patched the device trees to add the alias: https://github.com/MichaIng/build/commit/c1d5f10
Build runs: https://github.com/MichaIng/DietPi/actions/runs/14865331642
Can you test that build once this build finished:
```sh
cd /tmp
wget https://dietpi.com/download/binaries/linux-{image,dtb}-current-rockchip64.deb
sudo dpkg -i linux-{image,dtb}-current-rockchip64.deb
sudo reboot
```

briney83, 2025-05-06:
dietpi@tinkerboard-2:/tmp$ sudo dpkg -i linux-{image,dtb}-current-rockchip64.deb
dpkg-deb: ошибка: «linux-image-current-rockchip64.deb» не является архивом в формате Debian
dpkg: ошибка при обработке архива linux-image-current-rockchip64.deb (--install):
 подпроцесс dpkg-deb --control возвратил код ошибки 2
dpkg-deb: ошибка: «linux-dtb-current-rockchip64.deb» не является архивом в формате Debian
dpkg: ошибка при обработке архива linux-dtb-current-rockchip64.deb (--install):
 подпроцесс dpkg-deb --control возвратил код ошибки 2
При обработке следующих пакетов произошли ошибки:
 linux-image-current-rockchip64.deb
 linux-dtb-current-rockchip64.deb

MichaIng, 2025-05-06:
Sorry I missed an `s` in the URL:
```sh
cd /tmp
rm *.deb
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
sudo dpkg -i linux-{image,dtb}-current-rockchip64.deb
sudo reboot
```

briney83, 2025-05-06:
Yep, now faithful MAC

thanks for quick fix

closed 2025-05-25 by MichaIng