MichaIng/DietPi #7523: rules in /etc/iptables/rules.v4

z3huti, 2025-05-09:
what's the purpose of all the rules in /etc/iptables/rules.v4 if first 3 rules allow all anyway, question is rhetorical 

# Generated by iptables-save v1.8.9 (nf_tables) 
*filter
:INPUT ACCEPT [2900:1886009]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [2652:476848]

-A INPUT -i wg0 -p udp -m udp --dport 53 -m comment --comment pihole-DNS-rule -j ACCEPT

should it be better to block all in the first place, and then when user install programs write rules for it?

example:

*filter
:INPUT DROP [0:0]
:FORWARD DROP [0:0]
:OUTPUT ACCEPT [0:0]

#  (loopback)
-A INPUT -i lo -j ACCEPT
-A OUTPUT -o lo -j ACCEPT

# Allow already established connections (for responding to requests)
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT

# Allow SSH 
-A INPUT -p tcp --dport 22 -j ACCEPT

# Pi-hole Web Interface
-A INPUT -p tcp --dport 8089 -m comment --comment "Pi-hole Web Interface" -j ACCEPT

# Allow DNS for Pi-hole/Unbound (53 UDP and TCP)
-A INPUT -p udp --dport 53 -m comment --comment dns-udp -j ACCEPT
-A INPUT -p tcp --dport 53 -m comment --comment dns-tcp -j ACCEPT

MichaIng, 2025-05-10:
Where is `/etc/iptables/rules.v4` coming from? It is neither part of the `iptables` package nor of any `dietpi-software` installation. So you would need to check back about it with what/who created this rule/file. Probably just a failsafe step, which skipped checking whether it is actually needed or not.

It is however expected that installing `iptables` does not restrict any connections at first, but one needs to apply rules to restrict traffic. Your rule looks reasonable, to restrict incoming connections for Pi-hole DNS, web UI and SSH only. `-A OUTPUT -o lo -j ACCEPT` is not needed either, of course.

z3huti, 2025-05-10:
> Where is `/etc/iptables/rules.v4` coming from? It is neither part of the `iptables` package nor of any `dietpi-software` installation. So you would need to check back about it with what/who created this rule/file. Probably just a failsafe step, which skipped checking whether it is actually needed or not.
> 
> It is however expected that installing `iptables` does not restrict any connections at first, but one needs to apply rules to restrict traffic. Your rule looks reasonable, to restrict incoming connections for Pi-hole DNS, web UI and SSH only. `-A OUTPUT -o lo -j ACCEPT` is not needed either, of course. There might be however issues with IPv6 router advertisements, blocked pings and such.

could be that rules.v4 created when pihole, unbound and wireguard was installed on a clean dietpi install

MichaIng, 2025-05-10:
I checked back at the Pi-hole installer, and it created `iptables` rules only until including v4. v5 released 5 years ago removed this. Now it is only documented in the Pi-hole docs.

Our WireGuard server implementation creates rules only at VPN up/down script, not with such a persistent config file. Did you install PiVPN? The rule allows the DNS connection from WireGuard interface explicitly, so indeed looks like some managed Pi-hole + WireGuard setup, but not managed by us/`dietpi-software`.

z3huti, 2025-05-10:
> Did you install PiVPN?

yes

MichaIng, 2025-05-11:
Yeah I edited the found PiVPN installer code into my post above, which adds the rule. It is somewhat weird that for all other `iptables` and UFW rules the installer applies, complex checks are done about which one is required, but for the Pi-hole rule no checks are done at all. Wouldn't be hard to add it right to the block which applies if INPUT is not generally accepted: https://github.com/pivpn/pivpn/blob/8cb0a4e/auto_install/install.sh#L3481-L3493

However, it does not really hurt either to have a redundant rule. In any case, something which would need to be discussed at the PiVPN repo. I'll mark this issue as closed then.

closed 2025-05-11 by MichaIng