MichaIng/DietPi #7524: dietpi-update failure

studiokeywi, 2025-05-09:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
  **`G_DIETPI_VERSION_CORE=9`
  `G_DIETPI_VERSION_SUB=7`
  `G_DIETPI_VERSION_RC=1`
  `G_GITBRANCH='master'`
  `G_GITOWNER='MichaIng'`**
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
  **`bookworm`**
- Kernel version | `uname -a`
  **`Linux TGDOrangePi 6.6.31-current-sunxi64 #1 SMP Fri May 17 10:02:40 UTC 2024 aarch64 GNU/Linux`**
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
  **`Orange Pi Zero 3 (aarch64)`**
- Power supply used | (EG: 5V 1A RAVpower)
  Unsure
- SD card used | (EG: SanDisk ultra)
  Unsure

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`  
  **`61d63ada-a752-440f-8745-06986b01ac27`**

#### Steps to reproduce

1. Ran `sudo dietpi-update`

#### Expected behaviour

- Process should complete without failure

#### Actual behaviour

- Process failed to complete

#### Extra details

- Appears to be issue with `dkpg` and `dpkg: unrecoverable fatal error, aborting:database ... files list file for package 'libxcb-glx0:arm64' is missing final newline`

Slayingripper, 2025-05-10:
Server is down for everyone cant even access the site: 

https://www.isitdownrightnow.com/dietpi.com.html

MichaIng, 2025-05-10:
Your `/var/lib/dpkg/info/libxcb-glx0\:arm64.list` seems to be damaged. Can you show its content:
```sh
cat /var/lib/dpkg/info/libxcb-glx0\:arm64.list
```

studiokeywi, 2025-05-10:
> Your `/var/lib/dpkg/info/libxcb-glx0\:arm64.list` seems to be damaged. Can you show its content:
> 
> cat /var/lib/dpkg/info/libxcb-glx0\:arm64.list
```
dietpi@TGDOrangePi:~$ cat /var/lib/dpkg/info/libxcb-glx0\:arm64.list
/.
/usr
/usr/lib
/usr/lib/aarch64-linux-gnu
/usr/lib/aarch64-linux-gnu/libxcb-glx.so.0.0.0
/usr/share
/usr/share/doc
/usr/share/doc/libxcb-glx0
/usr/share/doc/libxcb-glx0/changelog.Debian.gz
/usr/share/doc/libxcb-glx0/changelog.gz
/usr/share/doc/libxcb-glx0/copyright
/usr/lib/aarch64-linux-gnu/libxcb-glx.so.0
dietpi@TGDOrangePi:~$
```

studiokeywi, 2025-05-10:
I'll spare the bulk of the output but the final lines: 
```
Processing triggers for hicolor-icon-theme (0.17-2) ...
gtk-update-icon-cache: **
Gtk:ERROR:../../../gtk/updateiconcache.c:1357:write_dir_index: assertion failed: (offset == ftell (cache))
Bail out! Gtk:ERROR:../../../gtk/updateiconcache.c:1357:write_dir_index: assertion failed: (offset == ftell (cache))
Aborted
WARNING: icon cache generation failed
dpkg: unrecoverable fatal error, aborting:
 failed to open diversions file: Input/output error
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

MichaIng, 2025-05-10:
Okay there is some problem with the filesystem. Check
```sh
dmesg -l 0,1,2,3
```
which will reveal I/O errors. Did you have some system crashes recently?

You can try to repair the filesystem:
```sh
> /forcefsck
reboot
```
And after reboot to check what has been done:
```sh
cat /run/initramfs/fsck.log
```

studiokeywi, 2025-05-10:
I ran the commands you suggested but forgot to save the output. However, I then was able to run `sudo apt upgrade` again to be told that I needed to run `sudo dpkg --configure -a`

After this, `sudo apt upgrade` was able to succeed, and I tried `sudo dietpi-update` which succeeded!

Much appreciated that you helped me with being able to restore the ability to update. Offhand, I couldn't tell you when the last time the board crashed would be, but it has happened lol.

Re-running `dmesg -l 0,1,2,3` now gave me this information, not sure if it's of use as it doesn't sound like this was actually an issue with DietPi itself, but trying to capture information in case anyone in the future encounters a similar situation:

```
dietpi@TGDOrangePi:~$ dmesg -l 0,1,2,3
[    1.480769] dw-apb-uart 5000000.serial: Error applying setting, reverse things back
[    1.481021] sun6i-spi 5010000.spi: Error applying setting, reverse things back
[    1.486283] sunxi-mmc 4020000.mmc: Error applying setting, reverse things back
[    1.486759] sunxi-mmc 4021000.mmc: Error applying setting, reverse things back
[    1.486959] dw-apb-uart 5000000.serial: Error applying setting, reverse things back
[    1.487211] sun6i-spi 5010000.spi: Error applying setting, reverse things back
[    4.770129] WCN_ERR: dts node for bt_wake not found
[    6.127461] wifi ini path = /lib/firmware/wifi_2355b001_1ant.ini
[    6.158197] sprdwl:sprdwl_get_fw_info length mismatch: len_count=83, r_len=89
[    6.158240] sprdwl:sprdwl_get_fw_info, drv_version=1, fw_version=2, compat_ver=0
[    6.158254] sprdwl:chip_model:0x2355, chip_ver:0x0
[    6.158263] sprdwl:fw_ver:0, fw_std:0x7f, fw_capa:0x120f7f
[    6.158274] sprdwl:mac_addr:24:b7:2a:46:3b:4f
[    6.158285] sprdwl:credit_capa:TX_WITH_CREDIT
[    6.158293] sprdwl:ott support:0
[   12.005895] unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection No More Mr WiFi failed status code:1![   13.280344] sprdwl:Wait disconnect event timeout. [mstime = 13276]
dietpi@TGDOrangePi:~$
```

MichaIng, 2025-05-11:
Okay looks good. No filesystem or I/O errors found/reported anymore. The other kernel errors are (sadly) common on the Orange Pi Zero 3, especially with this UNISOC WiFi driver which is not part of mainline Linux but patched inside and not super well maintained or of good coding quality.

I'll mark the issue as closed then.

closed 2025-05-11 by MichaIng

studiokeywi, 2025-05-11:
> Okay looks good. No filesystem or I/O errors found/reported anymore. The other kernel errors are (sadly) common on the Orange Pi Zero 3, especially with this UNISOC WiFi driver which is not part of [...]

Yeah, I do recall the WiFi is a bit of "well that's what I get for being cheap" for me -- from what I remember it doing, it mostly connects via Ethernet on first boot and then if/when that fails, it only ever connects to WiFi; plays hell with me regularly having to SSH into the SBC (it lives in my kitchen with a 6" Cat6 cable directly to one of my Eero beacons) since I have to switch which local IP it is on currently... but otherwise no complaints since it only hosts two seldom used Discord bots and soon some Plex usage