MichaIng/DietPi #7525: Home Assistant stuck?

whyisthisbroken, 2025-05-09:
Hey guys, 

I've installed Home Assistant and the latest version is 2025.4.4
I can't get the new update - is there any bug in the Update process? (sh)

Anyone have the same experience?

How can I see the logs from the Update process?

Yougoshatenshi, 2025-05-09:
With 2025.5.X they updated the required Python version from Python 3.13.0  to  Python >=3.13.2.
Currently DietPi only installs 3.13.1

whyisthisbroken, 2025-05-09:
Ah damn thanks... 
I've overlooked it... I'll try to push it manually. Thanks for the hint...

MichaIng, 2025-05-09:
Many thanks for the report.

Fixed with: https://github.com/MichaIng/DietPi/commit/bb4bf98

I wonder whether it causes more or less issue if `dietpi-software` installed always the latest (stable) Python version available for `pyenv` 🤔.

whyisthisbroken, 2025-05-09:
I think pushing the latest stable should not be a problem.
Never had any problems with it so far. Maybe test run and if several users complain about problems on the old method again🤔Thanks for the info that it will be fixed in the next release.

Yougoshatenshi, 2025-05-09:
> Anything we need to do to get this working?
> 
> The update script says everything is satisfied?
> 
> **/home/homeassistant/homeassistant-update.sh**

Waiting till its available on other branches (currently only available on dev).
A reinstall via `dietpi-software reinstall 157` will be needed to upgrade the python version then as usual, as long there is no change to that.

Dynamic5912, 2025-05-10:
So do I need to reinstall Home Assistant again?

The update script still says all requirements satisfied.

Joulinar, 2025-05-10:
As stated above, you need to wait until is has been moved into master branch or you switch into dev branch and reinstall HA afterwards.

MichaIng, 2025-05-14:
The patch can be applied also manually, nothing risky:
```sh
sudo sed -i "s/ha_python_version='3.13.1'/ha_python_version='3.13.3'/" /boot/dietpi/dietpi-software
sudo dietpi-software reinstall 157
```

Dynamic5912, 2025-05-14:
> The patch can be applied also manually, nothing risky:
> ```sh
> sudo sed -i "s/ha_python_version='3.13.1'/ha_python_version='3.13.3'/" /boot/dietpi/dietpi-software
> sudo dietpi-software reinstall 157
> ```

Works a treat - thanks!

whyisthisbroken, 2025-05-15:
Is it possible to add zlib-ng and isal by default? I have to install them separately when reinstalling HA via Dietpi.
HA gives a Warning:
```
Logger: aiohttp_fast_zlib
Quelle: components/http/__init__.py:208
Erstmals aufgetreten: 07:47:03 (1 Vorkommnis)
Zuletzt protokolliert: 07:47:03

zlib_ng and isal are not available, falling back to zlib, performance will be degraded.
```

MichaIng, 2025-05-15:
You can use `SOFTWARE_HOMEASSISTANT_PIP_DEPS` in `dietpi.txt` to have modules installed automatically on HA (re)installs.

Hmm, those are two alternative compression libraries:
- https://pypi.org/project/zlib-ng/
- https://pypi.org/project/isal/

`isal` is preferred by `aiohttp_fast_zlib`, we I added this to the default installation, which indeed silences the warning: https://github.com/MichaIng/DietPi/commit/c82aa45

closed 2025-05-25 by MichaIng

GrimXer0, 2025-05-29:
Heyo, I tried that command and now Im getting the error "pyenv: version `3.13.1' is not installed (set by /home/homeassistant/.python-version)" How would I fix this?

GrimXer0, 2025-05-29:
> The fix ist noch longer needed If you are in latest DietPi version. Simply remove the file and try again

Yep, tried that. reset backup. still the same error

Joulinar, 2025-05-29:
Can you remove the file without restoring a backup

GrimXer0, 2025-05-29:
> Can you remove the file without restoring a backup

I believe removing it deletes the entirity of dietpi-software since I tried replacing it and lost acces to dietpi-software

Joulinar, 2025-05-29:
?? Not sure what you mean, but you will not lose anything by removing `/home/homeassistant/.python-version`

MichaIng, 2025-06-23:
Then you are running a Bullseye system, where newer Home Assistant is not supported due to raised minimum SQLite version requirement.

Check out how to upgrade to Bookworm with our script: https://dietpi.com/blog/?p=3128#upgrade