MichaIng/DietPi #7526: dietpi.com down?

underwoodblog, 2025-05-10:
As the header says. dietpi.com is not reachable. Tried different locations.

https://www.reddit.com/r/dietpi/comments/1kj28gl/dietpicom_down_over_ipv4/

Slayingripper, 2025-05-10:
Seems to be down for everyone. 

https://www.isitdownrightnow.com/dietpi.com.html

sbadux, 2025-05-10:
I can't even find any mirror to download DietPi. Also LinuxTracker has no ISOs.
For the future, maybe it would be useful to have a link to some mirror or torrent magnet? For example in this Github Readme page?

DrCWO, 2025-05-10:
Same for me 👎

Joulinar, 2025-05-10:
I will contact @MichaIng and our new VPS provider

EmEiBee, 2025-05-10:
Good idea. Same problem here.
No pong on a ping.

Tracert / traceroute: 
```
1 1 ms <1 ms <1 ms fritz.box [192.168.1.111] 
2 18 ms 16 ms 17 ms p3e9bf33b.dip0.t-ipconnect.de [62.155.243.59] 
3 25ms 23ms 24ms 217.5.67.50 
4 24ms 23ms 23ms 62,157,251,167 
5 30 ms 29 ms 30 ms core21.fsn1.hetzner.com [213.239.224.105] 
6 28 ms 29 ms 28 ms ex9k2.dc11.fsn1.hetzner.com [213.239.254.94]
7 * * * Request timed out.
8 * * * Request timed out.
```

Joulinar, 2025-05-10:
We have restarted our VM, should be working again. Can you check?

sbadux, 2025-05-10:
it works, thanks!

EmEiBee, 2025-05-10:
well done

Joulinar, 2025-05-10:
Ok I will close the case for now. Feel free to reopen if needed.

closed 2025-05-10 by Joulinar

MichaIng, 2025-05-10:
As far as we can see it was my mistake switching from DHCP to static IP in a not too failsafe way, so that the default route or address could still expire.