MichaIng/DietPi #7527: Error using image script to create backup of SD-Card

DrCWO, 2025-05-10:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
I see an ERROR creating a backup image file from a SD-Card.
Complete Log attached.

Any ideas?

```
login as: root
root@192.168.0.196's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v8.24.1 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (aarch64)
 - CPU temp : 24 °C / 75 °F : Who put me in the freezer!
 - LAN IP : 192.168.0.196 (eth0)
 - MOTD : DietPi v9.12 has been released. Check out all changes:
          https://dietpi.com/docs/releases/v9_12/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v8.24.1 to v9.12.1

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# ls -al /dev/sd*
brw-rw---- 1 root disk 8, 0 May 10 08:58 /dev/sda
brw-rw---- 1 root disk 8, 1 May 10 08:58 /dev/sda1
brw-rw---- 1 root disk 8, 2 May 10 08:58 /dev/sda2
root@DietPi:~#
root@DietPi:~# export SKIP_ARCHIVE=1
root@DietPi:~# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
[  OK  ] DietPi-Imager | cd /root
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree xz-utils
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sda ...
[ INFO ] DietPi-Imager | MBR partition table detected
[ INFO ] DietPi-Imager | e2fsck -fyD /dev/sda2, please wait...
e2fsck 1.46.2 (28-Feb-2021)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 3A: Optimizing directories
Inode 2100 extent tree (at level 1) could be shorter.  Optimize? yes

Inode 134690 extent tree (at level 1) could be shorter.  Optimize? yes

Pass 4: Checking reference counts
Pass 5: Checking group summary information

/dev/sda2: ***** FILE SYSTEM WAS MODIFIED *****
/dev/sda2: 180442/962880 files (0.7% non-contiguous), 2391797/3855744 blocks
[  OK  ] DietPi-Imager | e2fsck -fyD /dev/sda2
[  OK  ] DietPi-Imager | mkdir /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | mount /dev/sda2 /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | rm -f /tmp/DietPi-Imager_mnt/root/.bash_history /tmp/DietPi-Imager_mnt/home/pi/.bash_history /tmp/DietPi-Imager_mnt/var/lib/dhcp/*.leases
[  OK  ] DietPi-Imager | mkdir /tmp/DietPi-Imager_mnt/etc/systemd/system/local-fs.target.wants
[  OK  ] DietPi-Imager | ln -s /etc/systemd/system/dietpi-fs_partition_resize.service /tmp/DietPi-Imager_mnt/etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | partprobe /dev/sda
[  OK  ] DietPi-Imager | partx -u /dev/sda
[ INFO ] DietPi-Imager | e2fsck -fyD /dev/sda2, please wait...
e2fsck 1.46.2 (28-Feb-2021)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 3A: Optimizing directories
Pass 4: Checking reference counts
Pass 5: Checking group summary information

/dev/sda2: ***** FILE SYSTEM WAS MODIFIED *****
/dev/sda2: 180442/962880 files (0.7% non-contiguous), 2391795/3855744 blocks
[  OK  ] DietPi-Imager | e2fsck -fyD /dev/sda2
[ INFO ] DietPi-Imager | Overriding filesystems free space with zeros to purge removed data and allow better compression...
[  OK  ] DietPi-Imager | mount /dev/sda1 /tmp/DietPi-Imager_mnt
[ INFO ] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt, please wait...
[FAILED] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[ INFO ] DietPi-Imager | fsck -y /dev/sda1, please wait...
fsck from util-linux 2.36.1
fsck.fat 4.2 (2021-01-31)
/dev/sda1: 373 files, 65118/258077 clusters
[  OK  ] DietPi-Imager | fsck -y /dev/sda1
[  OK  ] DietPi-Imager | tune2fs -O ^has_journal /dev/sda2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | tune2fs -O has_journal /dev/sda2
[  OK  ] DietPi-Imager | sync
[ INFO ] DietPi-Imager | zerofree -v /dev/sda2, please wait...
333540/1463949/3855744
[  OK  ] DietPi-Imager | zerofree -v /dev/sda2
[ INFO ] DietPi-Imager | e2fsck -fyD /dev/sda2, please wait...
e2fsck 1.46.2 (28-Feb-2021)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 3A: Optimizing directories
Pass 4: Checking reference counts
Pass 5: Checking group summary information

/dev/sda2: ***** FILE SYSTEM WAS MODIFIED *****
/dev/sda2: 180442/962880 files (0.7% non-contiguous), 2391795/3855744 blocks
[  OK  ] DietPi-Imager | e2fsck -fyD /dev/sda2
[  OK  ] DietPi-Imager | rmdir /tmp/DietPi-Imager_mnt
environment: line 514: ((: 262144 >  : syntax error: operand expected (error token is ">  ")
[FAILED] DietPi-Imager | Free space check: path=. | available=4734 MiB | required=15293 MiB
root@DietPi:~#
```

DrCWO, 2025-05-10:
Uuh, seems to be my fault, no Disk Space 🥇

DrCWO, 2025-05-10:
I deleted the old backup files (two times 4.5GB) and restarted. 
Same issue. It now reported 13GB free and 15293 MiB required. 

Now I mounted the two partitions on the SD card to backup to see how much space is filled:
```
root@DietPi:~# mount /dev/sda1 /mnt
root@DietPi:~# df -h /mnt
Filesystem      Size  Used Avail Use% Mounted on
/dev/sda1       127M   32M   95M  26% /mnt
root@DietPi:~# umount /mnt
root@DietPi:~# mount /dev/sda2 /mnt
root@DietPi:~# df -h /mnt
Filesystem      Size  Used Avail Use% Mounted on
/dev/sda2        15G  8.9G  5.6G  62% /mnt
root@DietPi:~#
```

I now ask myself why the backup script asks for 15GB free space as only 9GB are used?
So there SHOULD be enough space on the drive????

DrCWO, 2025-05-10:
Next I deleted aprox. 3GB of unnecessary files on the SD-Card to save.
Also using `dietpi-update` hoping that helps.

Still same issue 👎

MichaIng, 2025-05-10:
The last partition detection somehow failed, so that it could not be downsized along with the filesystem, and also caused the syntax error at line 514.

Can you show the output of this:
```sh
lsblk -rnpo NAME /dev/sda | tail -1
```

A bit weird as later
```sh
lsblk -rnpo NAME,FSTYPE /dev/sda?*
```
successfully printed all partitions of the drive.

While the root partition is selected in the script, the _last_ partition to shrink is auto-detected, since it is not assured that the root partition is the last one. And if not, shrinking the root partition has no point.

DrCWO, 2025-05-10:
This is what I get. I have no idea what this should tell me hope it may help you to fix the issue 👍 
```
root@DietPi:~# ls -al /dev/sd*
brw-rw---- 1 root disk 8, 0 May 10 10:35 /dev/sda
brw-rw---- 1 root disk 8, 1 May 10 10:35 /dev/sda1
brw-rw---- 1 root disk 8, 2 May 10 10:35 /dev/sda2
root@DietPi:~# lsblk -rnpo NAME /dev/sda | tail -1
/dev/sda1
root@DietPi:~# lsblk -rnpo NAME,FSTYPE /dev/sda?*
/dev/sda1 vfat
/dev/sda2 ext4
root@DietPi:~#
```
If you need further output for debugging please let me know.

DrCWO, 2025-05-10:
Maybe this additional information my help you debugging.

**With exactly the same SD-Card the image creation already worked in the past with an earlier release of the script 👍** 
I only changed some file content on the SD-Card but nothing else.

MichaIng, 2025-05-20:
It detects the first partition as last 🤔. Can you show this again:
```sh
lsblk -rnpo NAME /dev/sda
lsblk -rnpo NAME /dev/sda | tail -1
lsblk -npo NAME /dev/sda | tail -1
```
I remember there was some other case where the `-r` option, maybe even just in combination with a pipe, lead to a changed output order.

DrCWO, 2025-05-22:
I'm on holiday till Wednesday and can't try at the moment. 
Will get back to you with a response after I returned 👍

MichaIng, 2025-05-22:
Is it by chance a Bullseye system? I found again the report and debugging of `-r` messing up partition order: https://github.com/MichaIng/DietPi/issues/2630#issuecomment-2323534747

In this case, without `-r`, the partition name is prefixed with `└─`. We could also pipe it through `sort`. Or we just use `sfdisk`:
```sh
sfdisk -qlo Device /dev/mmcblk0 | tail -1
```

MichaIng, 2025-05-22:
Fixed with: https://github.com/MichaIng/DietPi/commit/447816c
Would be great if you could give it a try. The output above + whether it is a Bullseye system would be still good to know, just to assure I derived the underlying issue correctly, and we narrowed down the cases correctly, in which it occurs: https://github.com/MichaIng/DietPi/issues/7527#issuecomment-2895061240

closed 2025-06-22 by MichaIng