MichaIng/DietPi #7530: SkyConnect is not visible after update to v9_12

olehb007, 2025-05-13:
Reference code: 1047ba21-4584-4321-9efe-d5b9b9b51005

lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

No such file or directory: '/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_6cbbe40cd496ed11af82c298a7669f5d-if00-port0'

MichaIng, 2025-05-13:
Could you please fill the bug report template, so we know at least which hardware and distro version you use?

I don't know what SkyConnect is, but I guess some remote USB share, like VirtualHere and USB/IP? How is it installed and started?

You'd need to check its logs, to find out why it fails. Could be related to a kernel upgrade, but depends on your hardware whether there could have been any breaking change with a kernel upgrade.

MichaIng, 2025-05-31:
Marking as closed due to outstanding reply. Feel free to reopen if the issue persists, and you find time to provide some info about what SkyConnect is and check its logs on both sides.

olehb007, 2025-07-09:
Resolved after update to v9_14