MichaIng/DietPi #7531: Rock 5c image conversion and "extraction"

samolego, 2025-05-15:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

Hi there! I got a new radxa rock 5c and I tried converting armbian image to dietpi, since dietpi doesn't support this sbc (yet?). However, I run into some issues when trying to extract the converted image from sd card to .img file, similar to #7024

#### Details:
- Date           | 2025-05-15 09:07:45
- Program name   | DietPi-Imager
- Command        | `sgdisk -e /home/samoh/Downloads/radxa_rock5c.img`
- Exit code      | 4
- DietPi version | v9.12.1 (MichaIng/master)
- Distro version |  (ID=7)
- Kernel version | `Linux legopad 6.14.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Sun Apr  6 15:05:05 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux`
- Architecture   | `amd64`
- Hardware model |  (ID=)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. Download & flash https://www.armbian.com/radxa-rock-5c/ (minimal)
2. Run the dietpi conversion script
3. Poweroff the sbc and attach sd card to main PC
4. Run the image extraction script
#### Expected behaviour:
Image should be extracted successfully.
#### Actual behaviour:
There's a failure due to some size issues?
#### Additional logs:
```
Warning! Disk size is smaller than the main header indicates! Loading
secondary header from the last sector of the disk! You should use 'v' to
verify disk integrity, and perhaps options on the experts' menu to repair
the disk.
Caution: invalid backup GPT header, but valid main header; regenerating
backup header from main header.

Warning! Error 25 reading partition table for CRC check!
Warning! One or more CRCs don't match. You should repair the disk!
Main header: OK
Backup header: ERROR
Main partition table: OK
Backup partition table: ERROR

****************************************************************************
Caution: Found protective or hybrid MBR and corrupt GPT. Using GPT, but disk
verification and recovery are STRONGLY recommended.
****************************************************************************

Warning! Secondary partition table overlaps the last partition by
431 blocks!
Try reducing the partition table size by 1724 entries.
(Use the 's' item on the experts' menu.)
Aborting write of new partition table.
```

samolego, 2025-05-15:
Tried booting anyways without extraction and (at least) wifi module isn't found :(, so I guess this won't be really relevant.

MichaIng, 2025-05-15:
I remember this from another report when using certain base images. The GPT table seems to be unexpected huge: https://github.com/MichaIng/DietPi/issues/7024#issuecomment-2068216185

So to fix this, we'd need to check the actual GPT size, instead of assuming the default 34 sectors. You could replace the command with `:` or `true` from the error handler. It is good to have a valid backup partition table, but not necessary if the main one is intact. That way you can create the image for now.

MichaIng, 2025-06-02:
I see now I implemented this already: https://github.com/MichaIng/DietPi/commit/079710a
So the GPT size is obtained from the first usable sector. The GPT _can_ not be larger than that 🤔.

Do you have a chance to run these commands on the image after `dietpi-installer` finished?
```sh
sgdisk -p "$(findmnt -Ufnro SOURCE -M /)" 2>&1 | mawk -F[\ ,] '/^First usable sector/{print $5}'
sgdisk -p "$(findmnt -Ufnro SOURCE -M /)"
```

MichaIng, 2025-06-03:
Okay no problem. I implemented this when facing the issue in #7024 and I see you also found that issue before reporting. For the images I created the new method works, but I also never had a GPT size that was actually larger than 34 sectors.

I'll try to replicate once with the latest minimal ROCK 3C image from Armbian.