MichaIng/DietPi #7536: Dietpi update fails, apt error

MicDG, 2025-05-18:
#### Details:
- Date           | 2025-05-18 19:16:37
- Program name   | DietPi-Update
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- DietPi version | v9.11.2 (MichaIng/master)
- Distro version | bullseye (ID=6)
- Kernel version | `Linux HomeDroid 4.9.312-arm64 #1 SMP PREEMPT Tue Nov 22 21:32:44 CET 2022 aarch64 GNU/Linux`
- Architecture	 | `arm64`
- Hardware model | Odroid C4/HC4 (aarch64) (ID=16)
- Power supply   | NorthPada 5V 3A
- SD card        | Lexar Micro SD 64 GB
- Pre-image      | Meveric
#### Steps to reproduce:

1. Run dietpi-update as root
#### Expected behaviour:

- Update DietPi
#### Actual behaviour:

- apt update throws error
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://dietpi.com/apt bullseye InRelease
Hit:6 https://dietpi.com/apt all InRelease
Ign:7 https://dietpi.com/meveric bullseye InRelease
Ign:8 https://dietpi.com/meveric all InRelease
Err:9 https://dietpi.com/meveric bullseye Release
  404  Not Found [IP: 188.114.97.7 443]
Err:10 https://dietpi.com/meveric all Release
  404  Not Found [IP: 188.114.97.7 443]
Reading package lists...
E: The repository 'https://dietpi.com/meveric bullseye Release' no longer has a Release file.
E: The repository 'https://dietpi.com/meveric all Release' no longer has a Release file.
```

MichaIng, 2025-05-18:
I indeed removed our mirror of Meveric's APT repository, as we do not ship Odroids with its kernel anymore since years. Not sure whether a migration to our new kernel packages is possible, as of the possibly too small boot partition. Can you check this?
```sh
df -h
```
As long as you do not use specific GUI/video applications, you can remove the repository, or migrate back to the original one:
```sh
sed -i 's|https://dietpi.com|https://oph.mdrjr.net|' /etc/apt/sources.list.d/meveric*.list
```
resp. to remove them:
```sh
rm /etc/apt/sources.list.d/meveric*.list
```

MichaIng, 2025-05-20:
That boot partition is big enough, so we can try the migration. Do you have a possibility to backup the whole SD card, just in case something goes wrong?

RyoShinzo, 2025-05-21:
> As long as you do not use specific GUI/video applications, you can remove the repository, or migrate back to the original one:
> 
> sudo sed -i 's|https://dietpi.com|https://oph.mdrjr.net|' /etc/apt/sources.list.d/meveric*.list

I had the same issue (I'm running DietPi on an Odroid HC4). Executing this command fixed it, thanks.

MichaIng, 2025-05-21:
Will be done automatically next DietPi update: https://github.com/MichaIng/DietPi/commit/f6f9ec1

If someone wants to try the migration to our new kernel stack (have an SD card backup, at best):
```sh
G_EXEC curl -sSfo /boot/boot.cmd 'https://raw.githubusercontent.com/MichaIng/DietPi/refs/heads/dev/.build/images/U-Boot/boot.cmd'
G_EXEC curl -sSfo /boot/dietpiEnv.txt 'https://raw.githubusercontent.com/MichaIng/DietPi/refs/heads/dev/.build/images/U-Boot/dietpiEnv.txt'
G_EXEC mkdir -p /etc/kernel/post{inst,rm}.d /etc/initramfs/post-update.d
G_EXEC curl -sSfo /etc/kernel/postinst.d/dietpi-initramfs_cleanup 'https://raw.githubusercontent.com/MichaIng/DietPi/refs/heads/dev/.build/images/U-Boot/dietpi-initramfs_cleanup'
G_EXEC ln -sf /etc/kernel/post{inst,rm}.d/dietpi-initramfs_cleanup
G_EXEC curl -sSfo /etc/initramfs/post-update.d/99-dietpi-uboot 'https://raw.githubusercontent.com/MichaIng/DietPi/refs/heads/dev/.build/images/U-Boot/99-dietpi-uboot'

G_CONFIG_INJECT 'rootdev=' "rootdev=UUID=$(findmnt -Ufnro UUID -M /)" /boot/dietpiEnv.txt
G_CONFIG_INJECT 'rootfstype=' "rootfstype=$(findmnt -Ufnro FSTYPE -M /)" /boot/dietpiEnv.txt
G_EXEC sed --follow-symlinks -i '1i[main]' /boot/dietpiEnv.txt
G_EXEC sed --follow-symlinks -i '/^setenv overlay_error/a\
\n# petitboot workarounds\
setenv fdtfile "amlogic/meson-sm1-odroid-${variant}.dtb"\
setenv rootuuid "true"' /boot/boot.cmd
G_EXEC sed --follow-symlinks -i '/env import/c\	ini main ${scriptaddr} || env import -t ${scriptaddr} ${filesize}' /boot/boot.cmd
G_EXEC sed --follow-symlinks -i '/test -e/s/fixup.scr;/fixup.scr || test "${petitboot_active}" = "1";/' /boot/boot.cmd
G_EXEC sed --follow-symlinks -i '/^ln -s/c\mv -v "/boot/uInitrd-$1" /boot/uInitrd' /etc/initramfs/post-update.d/99-dietpi-uboot

G_EXEC eval 'echo '\''do_symlinks=0'\'' > /etc/kernel-img.conf'
G_AGI initramfs-tools u-boot-tools armbian-firmware zstd
G_EXEC mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
G_CONFIG_INJECT 'COMPRESS=' 'COMPRESS=zstd' /etc/initramfs-tools/initramfs.conf
G_AGI linux-{image,dtb}-current-meson64 linux-u-boot-odroidc4-current
# If above command fails with insufficient space, probably it works when running below first:
G_AGP linux-image-arm64-odroid-c4 meveric-keyring u-boot
G_EXEC rm -f /boot/Image.gz /boot/Image.gz.bak /boot/boot.ini
/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```
Before rebooting, check `/boot` that there is a kernel `Image` and `uInitrd` file as well as `dtb` dir freshly created. `/boot/dietpiEnv.txt` should contain `rootdev=UUID=` with the correct UUID of your root filesystem.

closed 2025-05-25 by MichaIng

MicDG, 2025-05-26:
Hello MichaIng, sorry for answering just now but I have been quite busy at work.

I tried the steps you suggested to migrate to the new kernel and couldn't boot the system anymore, lucky me the SD card died in a timely manner. So I just restored the backup on another SD card with the new v9.12.1 flashed. 

I got some errors though when executing the following command, if it can be of any use:

```
G_AGI initramfs-tools u-boot-tools armbian-firmware zstd
...
...
Unpacking armbian-firmware (25.02.0-trunk-dietpi1) ...
dpkg: error processing archive
/var/cache/apt/archives/armbian-firmware_25.02.0-trunk-dietpi1_all.deb
(--unpack):
trying to overwrite '/lib/firmware/ath10k/QCA9377/hw1.0/board-2.bin', which
is also in package firmware-atheros 20210315-3
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
/var/cache/apt/archives/armbian-firmware_25.02.0-trunk-dietpi1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Thanks for the help!

MichaIng, 2025-05-26:
Some missing package dependencies. I just added them: https://github.com/MichaIng/build/commit/56988f5

`apt purge firmware-atheros` would have solved it.

So you mean you restored a `dietpi-backup` on a new image? That would restore the old kernel etc as well 😄. Luckily the new bootloader seems to support the old `boot.ini` script, so that it can boot. So if that, please rerun the steps. The firmware package install would now succeed.

MichaIng, 2025-05-26:
`firmware-iwlwifi` was also missing, so in that that would need to be removed as well.

> but maybe again the SD was already starting to die so something was corrupted...?

No indication for that. Simply a package conflict that was not reflected in its meta data.

> Anyway after a few tries the installation did get to an end eventually, just didn't reboot when it had to, so I have no means to say whether it worked or not at the time.

Hmm, no idea how it could go through, without `G_AGI initramfs-tools u-boot-tools armbian-firmware zstd` being either skipped or left at a failed state, like one or more of these packages not being installed and/or configured correctly. This could have also led to the failing reboot, or at least failing network due to missing firmware blobs. But could be also something else. In any case good to have the migration steps collected here, for others who might be in the same situation. Then we can check back again whether something is missing.

> Sorry I said "restored", what I actually meant is I copied over the various configuration files for the software I had installed, as using dietpi-restore directly resulted in an unbootable SD card.

Ah yeah that makes sense. I guess no bad to be based on a fresh new image now, even if the migration would be theoretically possible. I always take such as opportunity to document and/or script my personal setup steps, so it can be easier redone and mostly automated in the future 🙂.