MichaIng/DietPi #7539: Open Beta v9.13 | Please help testing and hardening our upcoming release

MichaIng, 2025-05-19:
RC version | v9.13.0
----------- | -------
v9.13.0 PR | #7529
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2025-05-24**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av9.13
____
### Beta v9.13.0
_(2025-05-19)_

#### Removed software
- Google AIY | The Google AIY Voice Kit is not produced or sold anymore, and the related software repositories have been archived years ago. Furthermore we used install steps which were deprecated even years before the repository has been archived, and no sample to test whether it still works or not. If you still have an instance running, or aim to still try to get one running, here are uninstall instructions, as well as a link to the latest install instructions: https://github.com/MichaIng/DietPi/pull/7533
- Mycroft AI | As we state on our docs since over a year, Mycroft AI has shut down: https://mailchi.mp/cccbfdd464c7/mycroft-is-winding-down-its-operations-6268807. There are two forks we might want to implement as alternative: OpenVoiceOS and Neon AI. Let us know if you are interested, and we will help with the implementation. If you have a left dysfunctional Mycroft AI instance on your DietPi system, here is how to uninstall it manually: https://github.com/MichaIng/DietPi/pull/7534

#### Enhancements
- Orange Pi 3B | Added support for the PWM fan connector to mainline kernel builds. Many thanks to @Vyacheslav-S for reporting this missing kernel feature: https://github.com/MichaIng/DietPi/issues/7490#issuecomment-2829426560
- Radxa ZERO 3 | An upgrade to the Linux 6.1 based vendor kernel will be offered on DietPi update an implemented into new images from now on. Among other enhancements, this enables HDMI audio. Many thanks to @simpL3gr for reporting the missing HDMI audio functionality: https://github.com/MichaIng/DietPi/issues/7520
- ASUS Tinker Board 2 | Resolved an issue where the onboard Ethernet MAC address changed each boot with the new Linux 6.12 based kernel. Many thanks to @briney83 for reporting this issue: https://github.com/MichaIng/DietPi/issues/7522
- Virtual Machines | Our future VM images will ship with initramfs-tools instead of tiny-initramfs. The latter generates *much* smaller initramfs images (~1 MiB vs ~35 MiB), which is partly achieved by adding only those kernel modules actually used by the host. This auto-detection cannot work when generating images via container on other host systems, hence we maintained a small list of block device controller drivers known to be used by the virtualizers we provide images for. The default controller can however be changed, which then renders the image unbootable, and a static list of controller drivers reduces the compatibility of our VM images. For our Clonezilla-based installer images it causes the additional issue that Clonezilla aims to regenerate the initramfs via `update-initramfs` command, which would add possibly missing drivers, but tiny-initramfs provides `update-tirfs` instead. Furthermore, while Debian still maintains a tiny-initramfs package, there is actually no upstream development since years. We love small images with low disk I/O (for initramfs image generation), but sadly the downsides have become too large to keep using tiny-initramfs. Many thanks to @TheLinuxGuy for reporting the issue with Clonezilla, and an in turn unbootable VM image on Proxmox: https://github.com/MichaIng/DietPi/issues/7468
- First run setup/DietPi-Config | The keyboard layout dialogue has been simplified. It shows a single list of supported layouts now, instead of the multiple steps to select keyboard model, navigate back and forth for the correct language, layout, and layout variant, and additional options like AltGr and compose key functions, presented by `dpkg-reconfigure keyboard-configuration`. If additional options are needed, it is still possible to start the extended dialogue by selecting "Advanced" in our simplified menu. Else it will select the genetic 105-key keyboard model, default layout variant and options, which will work fine for most users, including most laptop keyboards, with and without keypad. The goal is to simplify and speed up the first run setup, as well as to unify the code sections which trigger a keyboard configuration.
- DietPi-Software | Gitea: This software option has been enabled for RISC-V systems, supported since v1.23.8.

#### Bug fixes
- Orange Pi 3B v1.1 | Resolved an issue where Ethernet did not work with the old revision of this board, when booting with the vendor bootloader from SPI. Many thanks to @iivailo for reporting this issue, and @Kwiboo for valuable insights based on ROCK 3B, which suffered from the same issue: https://github.com/MichaIng/DietPi/issues/7490, https://github.com/MichaIng/DietPi/issues/7327#issuecomment-2589181487
- DietPi-Set_CPU | Resolved an issue where on certain rare systems, the maximum CPU scheduling frequency was set to the min possible frequency, or the other way round, if not explicitly set differently. Many thanks to `@fat-pi2` and `@waym` for reporting this issue: https://dietpi.com/forum/t/6560
- DietPi-Software | RPi.GPIO: Resolved a regression where the installation failed on Bullseye systems as the `python3-rpi-lgpio` compatibility package exists since Bookworm only.
- DietPi-Software | fish: Resolved a syntax error on uninstall, and assured that all users who use fish as login shell are altered to use bash instead. Many thanks to @echtfrank for reporting these issues: https://github.com/MichaIng/DietPi/issues/7500
- DietPi-Software | Home Assistant: Resolved an issue where latest HA could not be installed/updated due to a raised Python requirement on v3.13.2. Many thanks to @whyisthisbroken and @Yougoshatenshi for reporting this issue: https://github.com/MichaIng/DietPi/issues/7525

TheLinuxGuy, 2025-05-19:
Hi @MichaIng is it possible for you to generate proxmox .iso of this release?

I can send the .iso to my VPS provider to upload and then i can redo the tests from #7468

MichaIng, 2025-05-19:
Sure, build is running: https://github.com/MichaIng/DietPi/actions/runs/15124589206
Can be found here once done: https://dietpi.com/downloads/images/testing/
This will be still DietPi v9.12, but built with the respective changes in our installer, hence with `initramfs-tools`.

MichaIng, 2025-05-22:
Hmm weird, do you have a chance to get some earlier logs?

The Hyper-V SCSI controller driver is present in the initramfs:
```console
root@VM-Bookworm:/mnt/boot# lsinitramfs initrd.img-6.1.0-35-amd64 | grep hv_storvsc
usr/lib/modules/6.1.0-35-amd64/kernel/drivers/scsi/hv_storvsc.ko
```
However, it should not even be used by Proxmox, but the Virtio-SCSI controller driver instead (at least recommended), and it also shows up in this part of your logs:
```console
root@VM-Bookworm:/mnt/boot# lsinitramfs initrd.img-6.1.0-35-amd64 | grep virtio_scsi
usr/lib/modules/6.1.0-35-amd64/kernel/drivers/scsi/virtio_scsi.ko
```
When Clonezilla was used, did you see it re-generating the initramfs?

MichaIng, 2025-05-23:
Beta v9.13.1 has been pushed, which another larger change: #7541

MichaIng, 2025-05-24:
Fixed with: #70dd123

MichaIng, 2025-05-24:
I see now that the problem is the `subnetcache` is enabled, which should usually not be the case. So there is `module-config: subnetcache` defined somewhere, hence `validator` and `iterator` modules are not leaded. What you do is disabling the `subnetcache` module, while alternatively `module-config: "subnetcache validator iterator"` would be possible to enable all 3 of them.

But it should not be enabled, usually: https://unbound.docs.nlnetlabs.nl/en/latest/manpages/unbound.conf.html#edns-client-subnet-module-options
This should only be used if clients can belong to different subnets, where you might want different cache instances per subnet, AFAIU.

Let me see whether Debian overrides defaults somehow.

MichaIng, 2025-05-24:
Okay understand it better now:
- The two cache settings do not apply to the `subnetcache` module, and the warning is not an error, but just tells you exactly that, while the settings do apply to other modules just well. So it is not wrong to have all 3 modules enabled and accept the warnings.
- Earlier, the code did not check/was away of that, and did not emit the warning, but this changed in newer Unbound.
- Unbound decided just recently to disable the `subnetcache` by default. Previously, and with the versions Debian currently serves, it is enabled by default, as/when compiled support for this module in general. Here the commit which changed the default: https://github.com/NLnetLabs/unbound/commit/35dbbcb

So what @z3huti did is matching the new Unbound default, and given the arguments for the commit, we should change it in our config pre-actively as well, not only to mute the warnings.

MichaIng, 2025-05-24:
Fixed with: https://github.com/MichaIng/DietPi/commit/778859f
Also patched on existing instances via DietPi update.

MichaIng, 2025-05-25:
Beta v9.13.2 has been merged: #7547

I'll prepare the release now.

closed 2025-05-25 by MichaIng

MichaIng, 2025-05-26:
DietPi v9.13 has been released: #7540

Thanks everyone for testing! ❤