MichaIng/DietPi #7542: Check and warn when static IP, gateway, and DNS are in inconsistent subnets

ralph303, 2025-05-23:
I’d like to suggest a feature improvement for network configuration in DietPi.
I saw a lot of users on the net encountering connection problems when installing or configuring software, either using apt or the installer in DietPI-Software. The problem occurs when you have to configure a static IP address, for example when installing and configuring PiHole or similar tools that need a static IP.

**Issue**
When setting a static IP address in dietpi-config, DietPi allows users to enter any IP, gateway, and DNS values without checking whether they belong to the same subnet. This can lead to a misconfigured system where:

    The static IP is set in one subnet (e.g., 192.168.178.x),
    But the default gateway is from a different subnet (e.g., 192.168.0.1),
    Resulting in failed connectivity (e.g., ping, curl, etc.), while DNS might still work.

This scenario is very confusing and hard to debug, especially for users who aren't deeply familiar with networking.

**Feature Request**
Add a validation or warning in dietpi-config when:

    The configured static IP and gateway are not in the same subnet,
    The DNS server is outside of the IP/gateway subnet (optional, but helpful).

**Optional Enhancements**
    Auto-suggest the gateway and DNS based on the entered IP address (e.g., .1 of the subnet),
    Provide a brief explanation if validation fails.

**Benefit**
This would make static IP configuration more robust, reduce support cases on the net, and prevent connectivity issues caused by simple misconfiguration.

Thanks for all the great work on DietPi, I love it!

MichaIng, 2025-05-23:
Let's play a bit with AI: @GitHubCopilot please write a small bash function to check whether two IPv4 addresses are within the same subnet, based on a netmask as additional input.

MichaIng, 2025-05-23:
A function by Copilot. Not too bad:
```bash
# Usage: same_subnet <ip1> <ip2> <netmask>
# Returns 0 (true) if both IPs are in the same subnet, 1 (false) otherwise.

same_subnet() {
    local ip1="$1"
    local ip2="$2"
    local netmask="$3"

    # Convert IP to integer
    ip2int() {
        local IFS=.
        read -r i1 i2 i3 i4 <<< "$1"
        printf '%u\n' "$(( (i1 << 24) | (i2 << 16) | (i3 << 8) | i4 ))"
    }

    local ip1_int ip2_int netmask_int
    ip1_int=$(ip2int "$ip1")
    ip2_int=$(ip2int "$ip2")
    netmask_int=$(ip2int "$netmask")

    if (( (ip1_int & netmask_int) == (ip2_int & netmask_int) )); then
        return 0
    else
        return 1
    fi
}

# Example usage:
# same_subnet 192.168.1.10 192.168.1.20 255.255.255.0 && echo "Same subnet" || echo "Different subnet"
```

MichaIng, 2026-11-14:
Done with https://github.com/MichaIng/DietPi/commit/0c0edd3.

closed 2026-11-14 by MichaIng