MichaIng/DietPi #7544: Home Assistant Deprecating Core and Supervised installation methods, and 32-bit systems

whyisthisbroken, 2025-05-23:
Well – this is bad news...
We have time until Home Assistant 2025.12, and then it's over and out with updates :(

https://www.home-assistant.io/blog/2025/05/22/deprecating-core-and-supervised-installation-methods-and-32-bit-systems/

MichaIng, 2025-05-23:
The information has to be interpreted a little differently. Their own OS and container still uses Home Assistant core, so of course it can be still installed manually, updated etc. What the change actually means is that for a custom HA core installation, there is no support by the HA team anymore, and no documentation for it. This however practically does not change much: HA made it already somewhat difficult with very strict and edge requirements for Python and modules, which lead to:
- A lot of huge by default unnecessary dependencies, which pull a lot of build dependencies on 32-bit ARM and RISC-V, and require constant maintenance for a HA core integration like ours.
- Often breaking HA installation between DietPi updates, as of raised of changed dependencies.

The HA team pretty much ignored any raised topic in this regards, focusing on functional HA OS and HA container, which both come with additional integrations, features and dependencies, not required by/wanted for a minimal HA core implementation. It just makes sense to turn this practical limitation of custom installs into a formal one. For us/me, at this point, it changes literally nothing: Breakage was usually found by user reports, or own random install tests. And fixes were derived from error messages, and the HA core source code.

That 32-bit support is dropped however indeed changes things a bit, as there are still about 15% of such among reported DietPi systems, especially the Raspberry Pi Zero W is still widely used. But on the other hand, those are not powerful enough to run HA nicely, and the installation and updates takes ages, also as a lot of Python modules need to be additionally compiled. I'll try to query our survey database about how many 32-bit DietPi systems have HA installed.
- However, while the HA team does not provide binaries/releases for these architectures anymore, it also does not mean that they will stop function immediately. However, it is then possible that HA implements 64-bit only features or methods at some point, which can then break full functionality on 32-bit, or spinning up HA at all. But we can just keep it enabled until we actually see it breaking with some version. Then we can pin it to an older version for a while, before removing the option for 32-bit entirely.

whyisthisbroken, 2025-05-23:
Thanks for your feedback. I just wanted to post it here for information. We'll see where things go.

InnovoDeveloper, 2025-05-23:
would it be possible to provide and official container installation method along with the current python envelope method?  There are many good low cost hardware options that can run the extra load.  would DietPi be able to run this container option or are there conflicts like NetworkManager, etc.

MichaIng, 2025-05-23:
Of course you can run any container on DietPi with any Docker engine. The official Docker container works just fine. It is the light HA core installation with is now not only practically, but also formally our own maintenance efforts.

BernieV1977, 2025-05-24:
Thanks for the info. I'm running DietPi on a NUC with HA core and nothing else. I'ts a dedicated system for HA. I like this method and yes, sometimes things are broken after an update from HA. So I can just continue using the same update procedure within DietPi from version 2025.12?

MichaIng, 2025-05-24:
> So I can just continue using the same update procedure within DietPi from version 2025.12?

Yes

ceisserer, 2025-08-24:
For me this is also bad news - I am running homeassistant among many other services on a RaspberryPi3 with just 1GB of RAM. Unforntunatly because written in python, home-assistant is quite memory hungry - and I use the 32-bit armv7 containers to conserve memory despite running 64-bit raspberry pi os as python's inefficiencies seem to be less servere when using a 32-bit data model.