MichaIng/DietPi #7545: orange pi zero 3 - kernel oops when upgrading to 6.12.23-current-sunxi64

milkywade, 2025-05-24:
[ YES] I have searched the existing open and closed issues

Required Information
DietPi version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | echo $G_DISTRO_NAME $G_RASPBIAN
bookworm

Kernel version | uname -a
I don't know - dietpi crashed when updating to a last kernel

SBC model | echo $G_HW_MODEL_NAME or (EG: RPi3)
Orange Pi Zero 3 (aarch64)

Power supply used | (EG: 5V 1A RAVpower)
5V 3A

SD card used | (EG: SanDisk ultra)
Sandisk

Steps to reproduce
sudo apt upgrade

`sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  linux-dtb-current-sunxi64 linux-image-current-sunxi64 linux-libc-dev
3 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 41.0 MB of archives.
After this operation, 2659 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 https://deb.debian.org/debian-security bookworm-security/main arm64 linux-libc-dev arm64 6.1.140-1 [2120 kB]
Get:2 https://dietpi.com/apt all/orangepizero3 arm64 linux-dtb-current-sunxi64 arm64 25.08.0-trunk-dietpi1 [134 kB]
Get:3 https://dietpi.com/apt all/orangepizero3 arm64 linux-image-current-sunxi64 arm64 25.08.0-trunk-dietpi1 [38.8 MB]
Fetched 41.0 MB in 9s (4790 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 30263 files and directories currently installed.)
Preparing to unpack .../linux-dtb-current-sunxi64_25.08.0-trunk-dietpi1_arm64.deb ...
Armbian 'linux-dtb-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' starting.
Armbian 'linux-dtb-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' finishing.
Unpacking linux-dtb-current-sunxi64 (25.08.0-trunk-dietpi1) over (25.05.0-trunk-dietpi1) ...
Preparing to unpack .../linux-image-current-sunxi64_25.08.0-trunk-dietpi1_arm64.deb ...
Armbian 'linux-image-current-sunxi64' for '6.12.23-current-sunxi64': 'prerm' starting.
Armbian 'linux-image-current-sunxi64' for '6.12.23-current-sunxi64': 'prerm' finishing.
Armbian 'linux-image-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' starting.
Armbian 'linux-image-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' finishing.
Unpacking linux-image-current-sunxi64 (25.08.0-trunk-dietpi1) over (25.05.0-trunk-dietpi1) ...

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.026399] Internal error: Oops: 0000000096000021 [#1] SMP

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.026399] Internal error: Oops: 0000000096000021 [#1] SMP

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.028268] Code: 14000003 f9400021 b4000441 d1001020 (88dffc0a)

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.028268] Code: 14000003 f9400021 b4000441 d1001020 (88dffc0a)

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.029784] Internal error: Oops: 0000000096000004 [#2] SMP

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.029784] Internal error: Oops: 0000000096000004 [#2] SMP

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.031602] Code: 14000003 f9400021 b4000441 d1001020 (88dffc0a)

Message from syslogd@faro at May 24 19:23:02 ...
 kernel:[  744.031602] Code: 14000003 f9400021 b4000441 d1001020 (88dffc0a)

Message from syslogd@faro at May 24 19:23:03 ...
 kernel:[  745.002430] Internal error: Oops: 0000000096000005 [#3] SMP

Message from syslogd@faro at May 24 19:23:03 ...
 kernel:[  745.004244] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:23:03 ...
 kernel:[  745.002430] Internal error: Oops: 0000000096000005 [#3] SMP

Message from syslogd@faro at May 24 19:23:03 ...
 kernel:[  745.004244] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:23:11 ...
 kernel:[  752.838234] Internal error: Oops: 0000000096000005 [#4] SMP

Message from syslogd@faro at May 24 19:23:11 ...
 kernel:[  752.838234] Internal error: Oops: 0000000096000005 [#4] SMP

Message from syslogd@faro at May 24 19:23:11 ...
 kernel:[  752.840049] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:23:11 ...
 kernel:[  752.840049] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:24:16 ...
 kernel:[  817.860389] Internal error: Oops: 0000000096000005 [#5] SMP

Message from syslogd@faro at May 24 19:24:16 ...
 kernel:[  817.862200] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:24:16 ...
 kernel:[  817.860389] Internal error: Oops: 0000000096000005 [#5] SMP

Message from syslogd@faro at May 24 19:24:16 ...
 kernel:[  817.862200] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:24:29 ...
 kernel:[  830.568218] Internal error: Oops: 0000000096000004 [#6] SMP

Message from syslogd@faro at May 24 19:24:29 ...
 kernel:[  830.569962] Code: 14000003 f9400021 b4000441 d1001020 (88dffc0a)

Message from syslogd@faro at May 24 19:24:29 ...
 kernel:[  830.568218] Internal error: Oops: 0000000096000004 [#6] SMP

Message from syslogd@faro at May 24 19:24:29 ...
 kernel:[  830.569962] Code: 14000003 f9400021 b4000441 d1001020 (88dffc0a)

Message from syslogd@faro at May 24 19:25:21 ...
 kernel:[  882.844569] Internal error: Oops: 0000000096000005 [#7] SMP

Message from syslogd@faro at May 24 19:25:21 ...
 kernel:[  882.844569] Internal error: Oops: 0000000096000005 [#7] SMP

Message from syslogd@faro at May 24 19:25:21 ...
 kernel:[  882.846388] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:25:21 ...
 kernel:[  882.846388] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:26:26 ...
 kernel:[  947.845646] Internal error: Oops: 0000000096000005 [#8] SMP

Message from syslogd@faro at May 24 19:26:26 ...
 kernel:[  947.845646] Internal error: Oops: 0000000096000005 [#8] SMP

Message from syslogd@faro at May 24 19:26:26 ...
 kernel:[  947.847460] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:26:26 ...
 kernel:[  947.847460] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:27:31 ...
 kernel:[ 1012.841749] Internal error: Oops: 0000000096000005 [#9] SMP

Message from syslogd@faro at May 24 19:27:31 ...
 kernel:[ 1012.841749] Internal error: Oops: 0000000096000005 [#9] SMP

Message from syslogd@faro at May 24 19:27:31 ...
 kernel:[ 1012.843566] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:27:31 ...
 kernel:[ 1012.843566] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:28:36 ...
 kernel:[ 1077.845152] Internal error: Oops: 0000000096000005 [#10] SMP

Message from syslogd@faro at May 24 19:28:36 ...
 kernel:[ 1077.845152] Internal error: Oops: 0000000096000005 [#10] SMP

Message from syslogd@faro at May 24 19:28:36 ...
 kernel:[ 1077.846967] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Message from syslogd@faro at May 24 19:28:36 ...
 kernel:[ 1077.846967] Code: d2800015 14000003 f9400273 b4000233 (b9401a63)

Progress: [ 23%] [###########################################......................`

milkywade, 2025-05-24:
burned the last image to a new SD card, booted and logged in, the system began auto-update.

All went well until I arrive again at
`Unpacking linux-image-current-sunxi64 (25.08.0-trunk-dietpi1) over (24.11.0-trunk-dietpi1) ...`
... where the system freezes.

milkywade, 2025-05-24:
and again - this is the 3rd time below unpacking fails and renders the machine unusable.

I've tried by now with 3 different SD cards, 2 of which brand new.

I'm now going to swap the SBC and try to install just in case.

```
Armbian 'linux-dtb-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' starting.
Armbian 'linux-dtb-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' finishing.
Unpacking linux-dtb-current-sunxi64 (25.08.0-trunk-dietpi1) over (24.11.0-trunk-dietpi1) ...
Preparing to unpack .../09-linux-image-current-sunxi64_25.08.0-trunk-dietpi1_arm64.deb ...
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 'prerm' starting.
Armbian 'linux-image-current-sunxi64' for '6.6.44-current-sunxi64': 'prerm' finishing.
Armbian 'linux-image-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' starting.
Armbian 'linux-image-current-sunxi64' for '6.12.23-current-sunxi64': 'preinst' finishing.
Unpacking linux-image-current-sunxi64 (25.08.0-trunk-dietpi1) over (24.11.0-trunk-dietpi1) ...
Connection to testmachine closed by remote host.
Connection to testmachine closed.
```

milkywade, 2025-05-24:
As mentioned above, I have today used two other, brand new, SD cards, one of them from Verbatim themselves.
Yes, I'm using the raspi 4 PSU.
I'll try what you mention.

MichaIng, 2025-05-24:
Are these 1 GB RAM models, and in case did you check whether it does correctly detect those 1 GB instead of 2?
```sh
free -m
```

milkywade, 2025-05-24:
yes, these are 1GB

MichaIng, 2025-05-24:
As of #7242. Not sure how the errors would look like, it it tried to write to a non-existing memory address, but thinkable that a kernel oops happens. And possible that the kernel upgrade fulling fs cache and write buffer etc leads to attempts to write to addresses above 1 GB, if it thinks it is available.

milkywade, 2025-05-24:
it almost seems like the upgrade script for the last kernel is expecting 2GB and fails to upgrade if the system only shows 1

MichaIng, 2025-05-24:
Hmm, this patch actually looks related, as it was exactly the column or rows bit (sometimes one, sometimes he other) being detected one to high: https://github.com/armbian/build/blob/main/patch/u-boot/v2025-sunxi/0001-sunxi-h616-dram-Rework-size-detection.patch

That should be actually included in our build from Mai 2nd 🤔. Ah ... no it is not, as `sunxi64` by default uses another patch dir: https://github.com/armbian/build/blob/main/config/sources/families/include/sunxi64_common.inc
And a pretty old U-Boot version in general, whatever the reason for this is.

... how there was a pretty similar issue with the H6 chip already 😄: https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-sunxi/Allwinner-Fix-incorrect-ram-size-detection-for-H6-boards.patch

I'll move the Orange Pi Zero 3 and 2W over to the U-Boot 2025 sources and patches. Let's see whether this helps.

MichaIng, 2025-05-25:
... I spoke too soon. I boots fine on my 2 GB variant, but does not boot on my 1.5 GB one ... really, this board is a total mess, the kernel, the badly maintained 3rd party drivers, the never really solved RAM size detection issues ... one version/patch fixes one thing, breaks the other again, and the other way round. These two Zero SBCs really caused a magnitude of order of more serious and time consuming issues than every other SBC, and they keep me from implementing support for newer Orange Pi SBCs ...

MichaIng, 2025-05-25:
Did you also flash that U-Boot image?
```sh
sudo /boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```
I found the missing patch for 1.5 GB support in the meantime, trying to apply this as well. This is SUCH a huge mess in the meantime, with multiple patch dirs for multiple different U-Boot versions, all applying multiple patches on top of each other to fix different things, or fix each other. I wished there would be more efforts to upstream every finding immediately, and drop these patches ASAP.

milkywade, 2025-05-25:
just did so and still 2GB

MichaIng, 2025-05-25:
I took the package off the APT server again, so probably you are still on the `24.11.0` package now. I am currently rebuilding them with the 1.5 GB RAM size included.