MichaIng/DietPi #7559: Friendly ARM R5C reverses LAN and WAN ports after latest update.

InnovoDeveloper, 2025-05-27:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`\
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='not applied'
G_LIVE_PATCH_STATUS[2]='not applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bookworm
- 
- Kernel version | `uname -a`
- 6.12.25-current-rockchip64 #1 SMP PREEMPT Fri Apr 25 08:48:08 UTC 2025 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- NanoPi R5S/R5C (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
- 5V4A
- 
- SD card used | (EG: SanDisk ultra)
emmc

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
After the latest update after a reboot, my network port has switched form the LAN port to the WAN port.  I am not sure if this happened because of the update or just a coincidence as I haven't rebooted this device for a while.  However, I had my ethernet cable plugged into the LAN port and set to eth0 .  After reboot, I had to switch the cable to the WAN port for it to get access to the network.  The interesting part is that it's still eth0.

I do have a custom file for a static MAC address because these devices don't have an assigned MAC and it randomizes with every reboot.  the MAC address didn't change although the cable is not plugged into the WAN port.  

Seems that eth0 and eth1 were switched around and the WAN was assigned priority.  (On a side note, I am not sure if in the past I had forced eth0 to be the LAN port)

How do I switch that back.  I would like to continue to use the LAN port for my eth0.

#### Expected behaviour

LAN should maintain eth0

#### Actual behaviour

WAN port is taking over eth0

#### Extra details

`:~# lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
`

`lshw -C network
  *-network DISABLED
       description: Wireless interface
       product: RTL8822CE 802.11ac PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:01:00.0
       logical name: wlan0
       version: 00
       serial: 48:8f:4c:d4:76:14
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw_8822ce driverversion=6.12.25-current-rockchip64 firmware=N/A latency=0 link=no multicast=yes wireless=IEEE 802.11
       resources: irq:88 ioport:1000(size=256) memory:f4200000-f420ffff
  *-network
       description: Ethernet interface
       product: RTL8125 2.5GbE Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0002:01:00.0
       logical name: eth0
       version: 05
       serial: 00:70:3d:f0:e4:0e
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=6.12.25-current-rockchip64 duplex=full firmware=rtl8125b-2_0.0.2 07/13/20 ip=192.168.0.133 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:79 ioport:200000(size=256) memory:f0200000-f020ffff memory:f0210000-f0213fff
`

MichaIng, 2025-05-27:
> How did I implement the workaround? did I miss the steps?

You mean for the random MAC? Probably like this? https://github.com/MichaIng/DietPi/issues/6565#issuecomment-1694492893
We could actually do this automatically for the effected Ethernet adapters on these SBCs, just based on the MAC address they have on first boot session.

> so the second eth is not listed at all.

True it was not present in your first outputs above either 🤔. On my R5S, with the new kernel, all 3 adapters work, just with switched interface names. Can you show the kernel output from an affected device?
```sh
dmesg
```

InnovoDeveloper, 2025-05-28:
> > How did I implement the workaround? did I miss the steps?
> 

I meant if this was way to reswap LAN and WAN.  For the static mac i have a 10-staticmac in the /etc/network/interfaces.d/ directory that sets the MAC addresss

> You mean for the random MAC? Probably like this? [#6565 (comment)](https://github.com/MichaIng/DietPi/issues/6565#issuecomment-1694492893) We could actually do this automatically for the effected Ethernet adapters on these SBCs, just based on the MAC address they have on first boot session.
> 
> > so the second eth is not listed at all.
> 
> True it was not present in your first outputs above either 🤔. On my R5S, with the new kernel, all 3 adapters work, just with switched interface names. Can you show the kernel output from an affected device?
> 
> dmesg

```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.12.25-current-rockchip64 (build@armbian) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Fri Apr 25 08:48:08 UTC 2025
[    0.000000] KASLR enabled
[    0.000000] Machine model: FriendlyElec NanoPi R5C
[    0.000000] efi: UEFI not found.
[    0.000000] OF: reserved mem: Reserved memory: No reserved-memory node in the DT
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x1ff763f80-0x1ff7666ff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x00000001f0000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e3c00000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 34 pages/cpu s99352 r8192 d31720 u139264
[    0.000000] pcpu-alloc: s99352 r8192 d31720 u139264 alloc=34*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: root=UUID=ebd59027-0204-4357-8d68-38a506fe86a9 rootfstype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1048064
[    0.000000] Policy zone: Normal
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000dfc00000-0x00000000e3c00000] (64MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[... 592 lines cut ...]
[   14.420674] rk2-crypto fe380000.crypto: Register sha512 as rk2-sha512 7
[   14.420688] rk2-crypto fe380000.crypto: Register sm3 as rk2-sm3 8
[   14.427604] random: crng init done
[   14.453590] usbcore: registered new interface driver btusb
[   14.457568] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000c lmp_ver=0a lmp_subver=8822
[   14.459555] Bluetooth: hci0: RTL: rom_version status=0 version=3
[   14.459580] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_fw.bin
[   14.463183] rtw_8822ce 0000:01:00.0: enabling device (0000 -> 0003)
[   14.470099] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822cu_config.bin
[   14.473711] Bluetooth: hci0: RTL: cfg_sz 6, total sz 37018
[   14.473863] rtw_8822ce 0000:01:00.0: Firmware version 9.9.15, H2C version 15
[... cut ...]
[   16.988381] r8169 0002:01:00.0 eth0: Link is Down
[   17.213448] loop10: detected capacity change from 0 to 8
[   19.709200] r8169 0002:01:00.0 eth0: Link is Up - 1Gbps/Full - flow control off
[   59.362788] wlan0: authenticate with 24:5a:4c:91:a1:66 (local address=00:72:20:de:b7:f4)
[   59.618074] wlan0: send auth to 24:5a:4c:91:a1:66 (try 1/3)
[   59.650381] wlan0: send auth to 24:5a:4c:91:a1:66 (try 2/3)
[   59.675062] wlan0: send auth to 24:5a:4c:91:a1:66 (try 3/3)
[   59.706024] wlan0: authentication with 24:5a:4c:91:a1:66 timed out

```

MichaIng, 2025-05-28:
You can also add test files here. There is a button below the text field to do so:

[image]

InnovoDeveloper, 2025-05-28:
> You can also add test files here. There is a button below the text field to do so:
> 
> [image]

1.  I started with an image with and 9.8.1 and 9.12.1 that I know works with all adapters.
2.  On systems where lspci found both 2.5Gb ethernet adapters, everything updated well and kept eth0 and eth1
 ```
 0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
 
```
3.  On systems where lspci found only the second ethernet adapter, it treated that as WAN and make it eth0
```
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)

```
4. The WLAN card may not have bearing.
**5. The other thing I noticed is that system that was not working had Linux version 6.6.56-current-rockchip64 and the system that was working had Linux version 6.12.25-current-rockchip64.  I am not exactly sure how this happened as I have updated both systems to the latest Dietpi version.  I thougth the kernel version also got updated.**

[device-known-NOT-TO-WORK-From-98.txt](https://github.com/user-attachments/files/20496035/device-known-NOT-TO-WORK-From-98.txt)
[device-known-NOT-TO-WORK-From-98-913.txt](https://github.com/user-attachments/files/20496036/device-known-NOT-TO-WORK-From-98-913.txt)
[device-known-NOT-TO-WORK-From-912-913.txt](https://github.com/user-attachments/files/20496033/device-known-NOT-TO-WORK-From-912-913.txt)
[device-known-TO-WORK-Flashed-912.txt](https://github.com/user-attachments/files/20496037/device-known-TO-WORK-Flashed-912.txt)
[device-known-TO-WORK-From-912-913.txt](https://github.com/user-attachments/files/20496034/device-known-TO-WORK-From-912-913.txt)

InnovoDeveloper, 2025-05-29:
I am also happy to ship you a R5C to play around with.  In the meantime, would you say that it's safe to update to 9.13.2 for devices in the field?  Is the logic that if they are working so far, then there is no reason they should stop?  or is it a crap shoot?

InnovoDeveloper, 2025-05-29:
> Do the boards with the missing LAN port interface have this consequently every reboot, and those without the issue never have it?

Yes, it's either the board has it or doesn't.  Once it has it, even if I remove the PCI card, LAN is not detected on every reboot (at least the dozen or so reboots that I have tested with)

MichaIng, 2025-05-29:
> would you say that it's safe to update to 9.13.2 for devices in the field?

Yes, I took out the new kernel from our APT repo. So they will stay at Linux 6.6. But looking at your results, it really seems that the old kernel is causing the same issue, so maybe it would not even be bad or worsen things, if I move the kernel back in for the R5C. Maybe do the DietPi update first, which does not affect the issue as it does not do any kernel upgrade (or does it upgrade from an older 6.6 to a newer 6.6?). Test whether some of them are affected by the issue regardless, and in case upgrade the kernel for them manually, to see whether it changes something or not. I am still wondering why you face the missing Ethernet adapter now, while the previous kernel causes the same issue already.

... maybe before doing the DietPi update, run `apt update && apt upgrade` to see whether a kernel upgrade is offered. If this is older 6.6 to newer 6.6, and is causing the issue on some boards, then this narrows down the issue to a particular Linux patch version, or possibly patches/changes done at Armbian within the related Armbian version strings.

MichaIng, 2025-05-29:
First findings regarding the `busn_res` error: The kernel indeed seems to reserve the bus number range `00-0f` for some "null" device, probably the PCI bridges? So to avoid the conflict, other bus ranges need to start after `0f`, e.g. `10-1f`, as used in vendor kernel.

This overlay removes the "error" for all 3 busses, assigning an all dedicated set of bus numbers:
```
/dts-v1/;
/plugin/;
/ {
	fragment@0 {
		target = <&pcie2x1>;
		__overlay__ {
			bus-range = <0x10 0x1f>;
		};
	};
	fragment@1 {
		target = <&pcie3x1>;
		__overlay__ {
			bus-range = <0x20 0x2f>;
		};
	};
	fragment@2 {
		target = <&pcie3x2>;
		__overlay__ {
			bus-range = <0x30 0x3f>;
		};
	};
};
```
```console
root@NanoPiR5S:~# dmesg | grep busn_res
[    2.191014] pci_bus 0000:11: busn_res: [bus 11-1f] end is updated to 11
[    4.171278] pci_bus 0001:21: busn_res: [bus 21-2f] end is updated to 21
[    5.305127] pci_bus 0002:31: busn_res: [bus 31-3f] end is updated to 31
root@NanoPiR5S:~# lspci
0000:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:11:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:21:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
```
The ports are functional, so it does not cause any issues. But the changed bus numbers can theoretically cause breakage in userland, e.g. with "predictable interface names" changing, or other things which rely on the PCI identifier/bus number.

In the end, this is only an "info" message (kernel log severity), and all 3 devices can use the same bus number 1 without issue, as they use a different PCI domain (the first 4 digits, while I don't know yet what it technically means 😅). It does not seem to be a problem that they try to define the same bus range, as they can all use the same existing range just fine. So I don't think this is causing the issue. But it wouldn't hurt if you just test it:
```sh
mkdir /boot/overlay-user
cat << '_EOF_' > /boot/overlay-user/bus-range.dts
/dts-v1/;
/plugin/;
/ {
	fragment@0 {
		target = <&pcie2x1>;
		__overlay__ {
			bus-range = <0x10 0x1f>;
		};
	};
	fragment@1 {
		target = <&pcie3x1>;
		__overlay__ {
			bus-range = <0x20 0x2f>;
		};
	};
	fragment@2 {
		target = <&pcie3x2>;
		__overlay__ {
			bus-range = <0x30 0x3f>;
		};
	};
};
dtc -I dts  -O dtb -o /boot/overlay-user/lan.dtbo /boot/overlay-user/lan.dts
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=bus-range' /boot/dietpiEnv.txt
reboot
```

InnovoDeveloper, 2025-05-29:
In case I got the dtc conversion correct, this is the result (still no love)

[device-known-NOT-TO-WORK-after-applying-patch.txt](https://github.com/user-attachments/files/20509381/device-known-NOT-TO-WORK-after-applying-patch.txt)

InnovoDeveloper, 2025-05-29:
ok strange, after first apply, reboot, device actually connected and worked with eth0
 ```lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

On second and subsequent reboot, issue came back.
```
lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

so I am not sure if the first time was one of those random occurrences or part of a cure

MichaIng, 2025-05-29:
I guess it just shows that the issue indeed appears a bit randomly.

Okay I am one step further, and have a static IP address for the first 2.5 Gbit network adapter as well. Just for the second is does not work. I guess U-Boot assigns them just for the `ethernet0` and `ethernet1` aliases. ... yeah this seems to be the code: https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-rockchip/board.c#L352-L359
It sets `ethaddr` and `eth1addr` variables, which are then assigned to the `ethernet0` and `ethernet1` aliases respectively: https://github.com/u-boot/u-boot/blob/master/boot/fdt_support.c#L645-L662

However, for the R5C it is enough, for the R5S an enhancement. Please try this:
```sh
cat << '_EOF_' > /boot/overlay-user/mac.dts
/dts-v1/;
/plugin/;
/ {
	fragment@0 {
		target = <&pcie3x1>;
		__overlay__ {
			pcie@0,0 {
				reg = <0x00000000 0 0 0 0>;
				#address-cells = <3>;
				#size-cells = <2>;

				r8125_1: pcie@1,0 {
					reg = <0x000000 0 0 0 0>;
					local-mac-address = [ 00 00 00 00 00 00 ];
				};
			};
		};
	};
	fragment@1 {
		target = <&pcie3x2>;
		__overlay__ {
			pcie@0,0 {
				reg = <0x00000000 0 0 0 0>;
				#address-cells = <3>;
				#size-cells = <2>;

				r8125_2: pcie@1,0 {
					reg = <0x000000 0 0 0 0>;
					local-mac-address = [ 00 00 00 00 00 00 ];
				};
			};
		};
	};
	fragment@2 {
		target-path = "/aliases";
		__overlay__ {
			ethernet0 = "/pcie@fe270000/pcie@0,0/pcie@1,0";
			ethernet1 = "/pcie@fe280000/pcie@0,0/pcie@1,0";
		};
	};
};

_EOF_
dtc -I dts  -O dtb -o /boot/overlay-user/mac.dtbo -@ /boot/overlay-user/mac.dts
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=mac' /boot/dietpiEnv.txt
reboot
```
Btw, I think the delay init stuff does not work like that in mainline Linux. I couldn't find such attribute anywhere else in mainline device trees. So there was not delay. I guess that vendor kernel has a delay for the first bus `0001:11` is the reason why it is loaded _after_ the second one `0002:21`, and hence why the interfaces are reversed between mainline and vendor kernel. Also the vendor device tree hence shows `r8125_1` symbol for the second bus, and `r8125_2` for the first, to have those matching the order in which they are loaded and in which interfaces are assigned/named. The `bus-delay` overlay did not have this reversing effect, so we need to find a way to define such a delay in mainline Linux, to test whether this causes/fixes the failing device detection.

MichaIng, 2025-05-30:
Here to commit to make R5C MAC addresses static (without overlay): https://github.com/MichaIng/build/commit/f259b59
Build running: https://github.com/MichaIng/DietPi/actions/runs/15355356432

Once done, to test:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,images}-current-rockchip64.deb
dpkg -i linux-{dtb,images}-current-rockchip64.deb
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=' /boot/dietpiEnv.txt # in case ;)
reboot
```
It know it does not solve the actual issue of the missing network adapter, but something I could figure our meanwhile.

There is definitely some issue with PCI device detection and resource distribution, but so far I could not find it, looping through a lot of logs and source code and some Duck AI help which can explain a few thing pretty well, but sometimes hangs on logic errors as well 😄. Aside of the bus range conflict (which does not really seem to be an error), I was digging into kernel messages about resolved circular device tree node dependencies. The 3 PCIe parent nodes have a circular dependency with the legacy interrupt controllers, which are child nodes of them. But I wasn't able to find the actual dependency of the child node on the parent, aside of maybe one same interrupt number used by both. But it actually looks pretty intended as well, just like the bus range thing.

I was also checking a little further why Ethernet interfaces swapped on R5S from Linux 6.6 to 6.12: The PHY for the PCIe bus of the second 2.5 Gbit Ethernet adapter is somehow loaded 1-1.5 seconds later, hence the bus is as well, which is then after the WAN 1 Gbit adapter, which is is a non-PCIe GMAC one. But I have no idea why it initiates so much later now, much later than other nodes which have a higher address (this `@fe270000` part) and would hence normally be processed later. The related PCIe bus is btw the one which fails in your case, so it in fact does load later already than with Linux 6.6. You can see it also in your logs:
```
[    4.358654] phy phy-fe8c0000.phy.7: lane number 0, val 1
[    4.359189] phy phy-fe8c0000.phy.7: lane number 1, val 2
[    4.359683] phy phy-fe8c0000.phy.7: bifurcation enabled
[    4.360430] rockchip-dw-pcie 3c0400000.pcie: host bridge /pcie@fe270000 ranges:
[    4.361183] rockchip-dw-pcie 3c0400000.pcie:       IO 0x00f2100000..0x00f21fffff -> 0x00f2100000
[    4.363071] rockchip-dw-pcie 3c0400000.pcie:      MEM 0x00f2200000..0x00f3ffffff -> 0x00f2200000
[    4.363931] rockchip-dw-pcie 3c0400000.pcie:      MEM 0x0340000000..0x037fffffff -> 0x0040000000
```
That PHY with Linux 6.6 is loaded at 2.X seconds, and has number 3 instead of 7 (I assume this means it was the 3rd loaded PHY, not is the 7th), and the PCIe bus loaded right next is the parent node of the 1st 2.5 Gbit adapter for the R5C, and the 2nd 2.5 Gbit adapter on R5S (the two SBCs use different PCIe parent nodes for them). But there is no init delay or any such defined anywhere, and the PHY (dependency) is loaded later, not the PCIe bus itself, so this does not look like an attempt to fix problems, and as you showed it does not fix them. I think the init delay would need to be between loading the parent PCIe bus and the child bus used by the network adapter. But so far no idea how to achieve that.

Does anyone know a good place to discuss such questions and issues related to mainline kernel? For reasons, Armbian sadly is impossible, the Linux mailing list is pretty focused on actual patches, not questions or problems, and at vendor forums it is mostly about vendor kernel and OS images. I'll probably try my luck at https://lists.kernelnewbies.org/pipermail/kernelnewbies/, but would prefer a forum platform over a mailing list.

InnovoDeveloper, 2025-06-04:
I used the commands as below,  I think it's "image" not "images"
```cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
dpkg -i linux-{dtb,image}-current-rockchip64.deb
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=' /boot/dietpiEnv.txt # in case ;)
reboot```

In either case, my system got bricked after that, didn't boot up, no HDMI output either.   I reimaged the device a few times and same thing happened after every update.

MichaIng, 2025-06-05:
Yeah right, `image` instead of `images`.

> In either case, my system got bricked after that, didn't boot up, no HDMI output either.

Um, that is surprising. AFAIK it worked on an R5C for someone on the forum, let me check back. The change is pretty harmless as well, defining a node and an alias 🤔.

MichaIng, 2025-06-07:
Had another look into this, assured that changes were applied to the correct device nodes. Still do not see the error. @InnovoDeveloper on a functional system, can you apply this overlay please, which adds the change for one device node only, so we test one after another:
```sh
mkdir -p /boot/overlay-user
cat << '_EOF_' > /boot/overlay-user/mac.dts
/dts-v1/;
/plugin/;
/ {
	fragment@0 {
		target = <&pcie3x1>;
		__overlay__ {
			#address-cells = <3>;
			#size-cells = <2>;
			pcie@0,0 {
				reg = <0 0 0 0 0>;
				#address-cells = <3>;
				#size-cells = <2>;
				r8125_1: pcie@1,0 {
					reg = <0 0 0 0 0>;
					local-mac-address = [ 00 00 00 00 00 00 ];
				};
			};
		};
	};
};
_EOF_
dtc -I dts  -O dtb -o /boot/overlay-user/mac.dtbo -@ /boot/overlay-user/mac.dts
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=mac' /boot/dietpiEnv.txt
reboot
```
If it applied correctly, you should have an additional symbol:
```sh
cat /proc/device-tree/__symbols__/r8125_1
```

InnovoDeveloper, 2025-06-18:
> Had another look into this, assured that changes were applied to the correct device nodes. Still do not see the error. [@InnovoDeveloper](https://github.com/InnovoDeveloper) on a functional system, can you apply this overlay please, which adds the change for one device node only, so we test one after another:
> 
> apt install device-tree-compiler
> mkdir -p /boot/overlay-user
> cat << '_EOF_' > /boot/overlay-user/mac.dts
> /dts-v1/;
> /plugin/;
> / {
> 	fragment@0 {
> 		target = <&pcie3x1>;
> 		__overlay__ {
> 			#address-cells = <3>;
> 			#size-cells = <2>;
> 			pcie@0,0 {
> 				reg = <0 0 0 0 0>;
> 				#address-cells = <3>;
> 				#size-cells = <2>;
> 				r8125_1: pcie@1,0 {
> 					reg = <0 0 0 0 0>;
> 					local-mac-address = [ 00 00 00 00 00 00 ];
> 				};
> 			};
> 		};
> 	};
> };
> _EOF_
> dtc -I dts -O dtb -o /boot/overlay-user/mac.dtbo -@ /boot/overlay-user/mac.dts
> G_CONFIG_INJECT 'user_overlays=' 'user_overlays=mac' /boot/dietpiEnv.txt
> reboot
> If it applied correctly, you should have an additional symbol:
> 
> cat /proc/device-tree/__symbols__/r8125_1
> EDIT: I probably found something to delay PCIe bus detection: https://lkml.org/lkml/2023/5/1/506 This patch is not in mainline, but Armbian has another patch which adds (only) `bus-scan-delay-ms = <1000>;` to the ROCKPro64 device tree. Probably support for that is patched in elsewhere, or they followed a false assumption. At least something to test. EDIT2: Nope, sadly it has no effect. At first I thought it does: after applying it with an overlay, the my R6S (tinkering with it for another reason) refused to boot. Another time the two PCI busses were detected in reversed order. Not sure about the boot failure, as doing the very same later worked perfectly fine. The reversed detection order seems to have been something random (problematic as of swapped interface names). Repeating reboots, and adjusting the delay did not result in any change, both PCI busses and Ethernet devices were detected pretty much at the same time (there are ~10% = ~0.15 seconds variation possible), and above should add a full second delay, even `<5000>` on any of both busses did not change something.

did you still want me to try this overlay?

MichaIng, 2025-06-18:
> did you still want me to try this overlay?

Yes please. I want to finish switching all NanoPi R SBCs to static MAC addresses. Only the R5C is outstanding.

Btw, depending on time and mood you have to do more test iterations, I would accept your prior offer to send me an R5C, at best one where the first Ethernet port is often missing. That issue is also still outstanding. Not sure if I am able to solve it, but since we have a similar case/report with the R5S as well, I aim to try getting help at some Linux mailing list of forum, and would then be able to do test iterations more quickly. Does this really only happen for you if an M.2 WiFi module is attached, or also without? In that case, I would need that module as well. Might well depend on the exact PCIe device attached to the port.

InnovoDeveloper, 2025-06-25:
> Yes please. I want to finish switching all NanoPi R SBCs to static MAC addresses. Only the R5C is outstanding.
>
> Btw, depending on time and mood you have to do more test iterations, I would accept [...]

Sorry, have been swamped lately.  PM your shipping address and I'll get one in the mail for you.  I'll test this during the weekend, I should have some time.

MichaIng, 2025-06-25:
You are innovodev on our forum? Or send me an email, my address is public on the GitHub profile.

MichaIng, 2025-09-25:
Sorry for the delay, Debian Trixie release kept me distracted for a while.

Yes I have one here, and was able to replicate the issue. With that particular board, with an M.2 WiFi module attached as well:
- Most of the time, the device attached to PCI bus 1 is not detected, which is otherwise the `eth0`. In that case, the other Ethernet adapter which would be `eth1` becomes `eth0`.
- In rare cases, both Ethernet adapters are detected.
- In one case, the WiFi module was not detected, which matches some earlier reports: https://dietpi.com/forum/t/17823/59

There was an upstream commit which applied the different PCI bus ranges which caused those kernel message that we suspected as possibly related to the issue, where our tests splitting the ranges via device tree overlay however did not solve it: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?id=fbea35a
This commit has not been backported to Linux 6.12 but is in Linux 6.16 and hence our edge kernel packages. However, it does not change anything, even with this new kernel, above behavior is pretty much the same.

I'm verifying that the vendor kernel is not causing this issue, and then try to get some help via Linux mailing list.

MichaIng, 2026-04-19:
As of my tests, I am optimistic that this has been indeed solved, hence closing the issue. But please test and verify, when you find time.

closed 2026-04-19 by MichaIng