MichaIng/DietPi #7560: Latest DietPi does not install any desktop that functions!

oliverhbailey, 2025-05-28:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bookworm 0
- Kernel version | `uname -a`
- Linux DietPi-Pi5 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- RPi 5 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- 5V 27 Watt
- SD card used | (EG: SanDisk ultra)
- MicroSD (Microcenter Brand)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- DIETPI-SOFTWARE
- Was the software title installed freshly or updated/migrated?
- Freshly
- Can this issue be replicated on a fresh installation of DietPi?
- YES

- Bug report ID | `echo $G_HW_UUID`
d0f9d489-3d28-42c4-8294-3929d4164181
#### Steps to reproduce

1. ...Install Fresh DietPi for RASPBERRY PI 5 B
2. ...CONFIGURE WIFI SSID and PASSKEY
3. First Time Login using SSH
4. Change Passwords
5. dietpi-update
6. dietpi-software
7. Install lxde
8. install x11
9. confirm software to install
10. run install software option
11. run dietpi-config
12. check screen-resolution0.set screen resolution to 1080p
13. setup up autologin
14. set user to dietpi
15. set lightdm as graphical login manager
16. set user to dietpi
17. exit dietpi-config
18. reboot
19. ssh back into system as dietpi
23. exit ### Expected behaviour- ...#### Actual behaviourstartx
24. no xinit found
25. 
- ...
after startx should be able to login from display
nothing happens

#### Extra details

I tried uninstalling and reinstalling lxde and X11
I tried removing lxde and installing xfce
tried running startx and received error that no X11 system found
tried running startxfce4 and received error that no such program found
gave up
- ...

Joulinar, 2025-05-28:
The desktop should be shown on an external monitor or screen and not via SSH.

MichaIng, 2025-05-28:
Okay live patch merged. @oliverhbailey please run `dietpi-update`. It will offer you a new live patch, which will add the missing APT install call. Then reinstall X11:
```sh
dietpi-software reinstall 6
```

Alternatively, install the missing packages manually:
```sh
apt install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-user-session xfonts-base x11-xserver-utils x11-utils xserver-xorg-video-fbdev
```

ravenclaw900, 2025-05-28:
@MichaIng I read that comment to mean "I wrote earlier that nothing was displayed on the external monitor." Both it and the original were slightly unclear, but I think SSH was just used to run `startx`, and an external monitor was available to view the desktop.

oliverhbailey, 2025-05-29:
Correct.  The ssh was nust to start xserver.  Actually, it was going to use
Real VNC for remote client. But all that currently displays on yhe monitor
or RealVNC is a mouse cursor against  a black background.  I'll work on it
tomorrow.

Thank you.

Oliver Bailey