MichaIng/DietPi #7569: build unbound from source

z3huti, 2025-06-03:
is it possible that you guys can make an option when installing Unbound from dietpi-software we get options to install Unbound from

stable repo
testing repo
build from source (latest version)

z3huti, 2025-06-03:
> with Debian Trixie

testing now

systemctl status unbound gives

Jun 03 18:54:16 DietPi (unbound)[477]: unbound.service: Referenced but unset environment variable evaluates to an empty string: DAEMON_OPTS

did some changes to remove this warning, other than that it work ok

MichaIng, 2025-06-03:
No need to mute that warning. I mean you could declare the variable explicitly as empty string in the environment file, but it is not uncommon that services do not ship one at all, but do still load it, and supported variables.

z3huti, 2025-06-10:
if you are gonna provide own newer builds, please make it optional when installing from dietpi-software

btw dietpi on bookworm feels/runs snappier and faster than on trixie

MichaIng, 2025-06-10:
> if you are gonna provide own newer builds, please make it optional when installing from dietpi-software

I would like to avoid making it too complex. Like in some other cases, we would provide it via our APT repository. So our builds would then be installed automatically. It can be always prevented via `/etc/apt/preferences.d`.

> btw dietpi on bookworm feels/runs snappier and faster than on trixie

Which hardware, and which part/software exactly feels slower on Trixie? The DietPi version is not related to the kernel version, at least not on SBCs, so it would be some software component, and hence limited to a particular interface.

MichaIng, 2025-06-10:
On the Bookworm system, you use the new Linux 6.12 firmware stack already? If so, can you compare
```sh
dmesg
```
timestamps to see at which point they start to differ?

"Startup time" generally is a very vague term. Many things are initiated asynchronously, and there are different stages, bootloader, in case initramfs (not by default used with our RPi images), kernel, init system. If the kernel is the same, the init system should start at the very same point. There are some things which need to be initiated first, like the rootfs mount of course, and other "auto" mounts, which is `/boot/firmware` on RPi additionally, and some init system core components. Network and e.g. SSH server are brought up before local login prompts, so it can be possible (depending on how fast the network adapter initiates, and then e.g. DHCP server and other software components) to login via SSH earlier than locally. Local console login prompts appear after `systemd-user-sessions.service`, but other services can start well after login prompts. `systemd-analyze` shows the time it took for really all services scheduled to be started, which can include Unbound and Pi-hole, while both could hang unlimited without affecting any other part of the host system. What I am saying: it is complex, depends on what you define as a finished startup, what you look at, how you measure etc. The Debian version cannot make the hardware slower of course.

Also assure that you in case use the same overclocking in `/boot/config.txt`, i.e. that `cpu` shows the same governor and min/max frequencies. Also the SD card (or boot media) used can affect this, of course.

If hardware, kernel and performance settings are the same, Pi-hole web UI can only be affected by Pi-hole itself. So if it was the same version before and after, any different would be quite surprising: it does not even depends on Debian's PHP packages since Pi-hole v6, but is completely self-contained. Only left explanation could then be high activity or memory usage from other processes slowing down Pi-hole indirectly. Check via `htop`. Ah, also verify swap space is the same, on the same drive etc, if used at all.

z3huti, 2025-06-10:
on VM
Bookworm is on 6.1.0-37-amd64 
Trixie is on 6.12.30-amd64

on RPi 3b
Bookworm 6.12.25+rpt-rpi-v8

maybe in few days gonna test Trixie on RPI3b again

[image]
[image]

Dynamic5912, 2025-06-17:
Is this possible with Unbound?

Will it be reviewed and implemented at some point?

MichaIng, 2025-06-17:
It needs to be implemented differently, will update it after upcoming DietPi release.

Dynamic5912, 2025-07-26:
Does it install with the same switches and modules etc. as the current apt version?

_dietpi@dietpi:~$ sudo unbound -V
Version 1.17.1

Configure line: --build=aarch64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/aarch64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --with-pythonmodule --with-pyunbound --enable-subnet --enable-dnstap --enable-systemd --with-libnghttp2 --with-chroot-dir= --with-dnstap-socket-path=/run/dnstap.sock --disable-rpath --with-pidfile=/run/unbound.pid --with-libevent --enable-tfo-client --with-rootkey-file=/usr/share/dns/root.key --disable-flto --enable-tfo-server
Linked libs: libevent 2.1.12-stable (it uses epoll), OpenSSL 3.0.16 11 Feb 2025
Linked modules: dns64 python subnetcache respip validator iterator
TCP Fastopen feature available

BSD licensed, see LICENSE in source package for details.
Report bugs to unbound-bugs@nlnetlabs.nl or https://github.com/NLnetLabs/unbound/issues
_

z3huti, 2025-07-26:
> Does it install with the same switches and modules etc. as the current apt version?
>
> _dietpi@dietpi:~$ sudo unbound -V Version 1.17.1
>
> Configure line: --build=aarch64-linux-gnu --prefix=/usr [...]

 sudo unbound -V
Version 1.23.1

Configure line: --enable-checking --prefix=/usr --sysconfdir=/etc --localstatedir=/var --runstatedir=/run --without-pyunbound --without-pythonmodule --enable-systemd --enable-subnet --enable-cachedb --enable-dnstap --enable-tfo-client --enable-tfo-server --with-libhiredis --with-libnghttp2 --with-chroot-dir= --with-dnstap-socket-path=/run/dnstap.sock --with-libevent --with-pthreads --with-rootkey-file=/usr/share/dns/root.key --disable-rpath
Linked libs: libevent 2.1.12-stable (it uses epoll), OpenSSL 3.5.1 1 Jul 2025
Linked modules: dns64 cachedb subnetcache respip validator iterator
TCP Fastopen feature available

BSD licensed, see LICENSE in source package for details.
Report bugs to unbound-bugs@nlnetlabs.nl or https://github.com/NLnetLabs/unbound/issues

Dynamic5912, 2025-07-26:
What happened to your existing config file? Is it overwritten with a new one each time?

I also use unbound-control - no biggie if it's not included anymore

z3huti, 2025-07-26:
> What happened to your existing config file? Is it overwritten with a new one each time?
> 
> I also use unbound-control - no biggie if it's not included anymore

i think config file is not overwritten, it was just an update of package

MichaIng, 2025-07-28:
I am closing it here, let's keep discussion in #7632.

closed 2025-07-28 by MichaIng