MichaIng/DietPi #7580: Pine64 no sound aux port

GrimXer0, 2025-06-12:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.13
- Distro version | bookworm
- Kernel version | Linux DietPi 6.12.23-current-sunxi64 #1 SMP Thu Apr 10 12:39:41 UTC 2025 aarch64 GNU/Linux
- SBC model | PINE A64 (aarch64)
- SD card used | SanDisk

#### Steps to reproduce

1. select hw 0,0
2. no audio

#### Expected behaviour

- play audio via aux port

#### Actual behaviour

- nothing

MichaIng, 2025-06-12:
Are you sure the PINE A64 has an aux (Auxiliary) port? I cannot find it in any product page or documentation:
- https://pine64.org/devices/pine_a64/
- https://pine64.org/documentation/Pine_A64/_full/
- https://wiki.pine64.org/wiki/PINE_A64

Or do you mean the analogue audio jack? If so, does it produce noise with speaker test?
```sh
speaker-test -c 2
```
Does that command produce an error instead? If it shows info that it is producing noise, but you do not here something, please check whether you find some volume control that is too low:
```sh
alsamixer
```

GrimXer0, 2025-06-13:
Oh, now it works suddenly. Only thing happening now is a poping sound when playing audio. Any fix for that. Thanks for the help btw great support!

GrimXer0, 2025-06-13:
Wait, I said it wrong. There is a singular loud pop sound when playback starts. No weird noises during playback

MichaIng, 2025-06-13:
Does the same happen when you use HDMI audio?

And what do you mean "when playback starts". Does it happen when the player starts using the audio device, or also when you pause/resume playback? Should be still some kind of analogue signal distortion, since of course there is no blob added to the audio stream by any (other) software component. So it does not really change something about the things to test, trying to isolate the analogue signal as much as possible. Same for CPU usage etc: possible that starting playback, depending on player etc, causes a load peak for reading audio source, in case starting conversion, filling buffers and all such.

MichaIng, 2025-06-13:
I am not aware of power savings which would affect audio streams. Can you link where you read about it? Maybe I can derive where the same could be adjusted on our images.

GrimXer0, 2025-06-14:
it was this one [https://www.reddit.com/r/voidlinux/comments/nwi6rb/how_do_i_deactivate_the_sound_card_power_saving/](url)

MichaIng, 2025-06-15:
Can you show the output of this:
```sh
modinfo snd_hda_intel
```
It should be possible to apply this module option via `modprobe` config permanently.

MichaIng, 2025-09-21:
Closing this for now. Feel free to reopen if the issue persists.

closed 2025-09-21 by MichaIng