MichaIng/DietPi #7589: How to configure UART RX TX 115200 on Radxa Zero 3W

madhugoundla, 2025-06-18:
Hello, I am trying to connect to PIN 8 TX and PIN 10 RX as mentioned below.

https://docs.radxa.com/en/zero/zero3/hardware-design/hardware-interface

I have deployed Linux DietPi 6.1.115-vendor-rk35xx #1 SMP aarch64 GNU/Linux on Radxa Zero 3W.

In dietpi-config, i have disable FIQ0 and enable ttyS1.

in /boot/dietpiEnv.txt, i have added overlays=uart2

When i run minicom -D /dev/ttyS1 -b 115200, i get nothing.

Can someone please help, on how to read UART at 115200 on Radxa Zero 3w with DIETPi?

MichaIng, 2025-06-18:
`ttyFIQ0` is however correct for pins 6-8-10. No need to add any device tree overlay either. Ouput is however done at 1,500,000 baud rate. Please remove the overlay, disable the console on `ttyS1`, and re-enable the one on `ttyFIQ0`. Then try:
```sh
minicom -D /dev/ttyFIQ0 -b 1500000
```

madhugoundla, 2025-06-18:
Thank you let me try this, and yes i need 115200. Thank you for replying so fast.

It is fine it is loaded after kernel is loaded.

madhugoundla, 2025-06-18:
@MichaIng  I changed 1500000 to 115200 in dietpiEnv.tx and rebooted the system.

consoleargs=console=ttyS2, 115200 console=tty1 console=ttyFIQ0, 115200

But i am not able to get any data back from a barcode reader.

`minicom -D /dev/ttyFIQ0 -b 115200`

MichaIng, 2025-06-18:
Are these spaces really there, or is it just a copy&paste artifact? There must not be a space between the console device and the baud rate: `console=ttyFIQ0, 115200` => `console=ttyFIQ0,115200`

Let me test on my 3E whether it works there. Would be possible, but surprising, if it supports 1500000 only.

madhugoundla, 2025-06-18:
It was there, nothing from copy pasting. I reflashed it to keep it clean.

Thank you, if you can test. That would be great.

madhugoundla, 2025-06-18:
Looks like Rockchips are default to UART to 1500000 but Amlogic is 115200.

https://docs.radxa.com/en/zero/zero3/low-level-dev/serial

For Radxa products based on Rockchip chips, the default UART configuration is 1500000n8 without flow control.

Please check if your USB to TTL serial cable supports 1.5M baud rate:

Based on CP210X and PL2303x some products have baud rate limitations.
Based on FT232RL, some products have power issues.
The Flowing text uses a serial cable based on CH340.

https://docs.radxa.com/en/zero/zero2pro/low-level-dev/serial

For Radxa products based on Amlogic chips, the default UART configuration is 115200n8 without flow control.

The Flowing text uses a serial cable based on CH340.

MichaIng, 2025-06-18:
The console on `ttyS2` is btw unintentionally enabled. You can disable it like that:
```sh
/boot/dietpi/func/dietpi-set_hardware serialconsole 0 ttyS2
```
It fails anyway as the device does not exist.

madhugoundla, 2025-06-18:
Thank you for taking your time, we can go ahead and close this. Hopefully will help anyone running into this issue.

MichaIng, 2025-06-18:
However, there are [other UART pins](https://docs.radxa.com/en/zero/zero3/hardware-design/hardware-interface#gpio-interface), and overlays to enable it. But they have RK3568 prefix. But you can test those:
- Set `overlay_prefix=rk3568` in `/boot/dietpiEnv.txt`
- Set `overlays=uart3-m0` for the UART device at pins 3+5

Not sure whether it appears as `/dev/ttyS3` or other name after reboot, and whether it works. There are other overlays. See `/boot/dtb/rockchip/overlays`:

[image]

Set/add them to `overlays=` setting, separated by space is you add multiple ones at once. But better test only one by one: if one overlay fails to apply, none is applied.