MichaIng/DietPi #7590: reboot kernel panic - orange pi zero 3 1GB memory - DietPi v9.13.2

verniegithub, 2025-06-19:
Below it the reboot messages after first boot. DietPi v9.13.2
Linux DietPi 6.12.23-current-sunxi64 #1 SMP Thu Apr 10 12:39:41 UTC 2025 aarch64

root@DietPi:~# reboot
[ 1060.546362] systemd-shutdown[1]: Failed to set watchdog hardware timeout to 10min: Invalid argument
[ 1061.362038] Unable to handle kernel paging request at virtual address ffff80007a716658
[ 1061.370012] Mem abort info:
[ 1061.372812]   ESR = 0x0000000096000007
[ 1061.376582]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 1061.381908]   SET = 0, FnV = 0
[ 1061.384969]   EA = 0, S1PTW = 0
[ 1061.388127]   FSC = 0x07: level 3 translation fault
[ 1061.393023] Data abort info:
[ 1061.395922]   ISV = 0, ISS = 0x00000007, ISS2 = 0x00000000
[ 1061.401426]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[ 1061.406493]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[ 1061.411824] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000041db4000
[ 1061.418541] [ffff80007a716658] pgd=100000004273a003, p4d=100000004273a003, pud=1000000043673003, pmd=1000000044b7f003, pte
=0000000000000000
[ 1061.431089] Internal error: Oops: 0000000096000007 [#1] SMP
[ 1061.436672] Modules linked in: snd_soc_hdmi_codec sunxi_cedrus(C) polyval_ce polyval_generic snd_soc_sunxi_ahub v4l2_mem2m
em snd_soc_sunxi_machine sun50i_h6_prcm_ppu videobuf2_dma_contig dw_hdmi_i2s_audio dw_hdmi_cec videobuf2_memops videobuf2_v4l
2 snd_soc_sunxi_ahub_dam videodev sun50iw9_codec panfrost videobuf2_common gpu_sched mc drm_shmem_helper display_connector du
mp_reg cpufreq_dt nls_iso8859_1 rfkill fuse dm_mod motorcomm sun6i_rtc_ccu dwmac_sun8i mdio_mux [last unloaded: uwe5622_bsp_s
dio]
[ 1061.480697] CPU: 0 UID: 0 PID: 1 Comm: systemd-shutdow Tainted: G         C         6.12.23-current-sunxi64 #1
[ 1061.490702] Tainted: [C]=CRAP
[ 1061.493673] Hardware name: OrangePi Zero3 (DT)
[ 1061.498119] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1061.505081] pc : device_shutdown+0x184/0x298
[ 1061.509367] lr : device_shutdown+0xf4/0x298
[ 1061.513556] sp : ffff80008256bc30
[ 1061.516873] x29: ffff80008256bc30 x28: ffff000002a30000 x27: ffff0000049fac88
[ 1061.524018] x26: 0000000000000000 x25: ffff800082468030 x24: 0000000000000000
[ 1061.531161] x23: 0000000000000001 x22: ffff80008251fcb0 x21: ffff000004e4b808
[ 1061.538305] x20: ffff0000049fac08 x19: ffff0000049fac10 x18: ffffffffffffffff
[ 1061.545448] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[ 1061.552590] x14: 0000000000000001 x13: ffff8000821f9c78 x12: 0000000000000001
[ 1061.559733] x11: 000000f71e1896b1 x10: 0000000000000a10 x9 : ffff800080a6fcac
[ 1061.566877] x8 : ffff000002a30a70 x7 : 0000000000000000 x6 : 0000000000000001
[ 1061.574018] x5 : ffff0000049facec x4 : 0000000000000000 x3 : 0000000000000002
[ 1061.581160] x2 : 0000000000000001 x1 : 0000000000000000 x0 : ffff80007a716608
[ 1061.588304] Call trace:
[ 1061.590756]  device_shutdown+0x184/0x298
[ 1061.594688]  kernel_restart+0x48/0xb8
[ 1061.598362]  __do_sys_reboot+0x114/0x240
[ 1061.602291]  __arm64_sys_reboot+0x2c/0x40
[ 1061.606307]  invoke_syscall+0x50/0x120
[ 1061.610065]  el0_svc_common.constprop.0+0x48/0xf0
[ 1061.614775]  do_el0_svc+0x24/0x38
[ 1061.618097]  el0_svc+0x30/0xd0
[ 1061.621163]  el0t_64_sync_handler+0x120/0x130
[ 1061.625527]  el0t_64_sync+0x190/0x198
[ 1061.629202] Code: 941ab4f1 17ffffbb f9403680 b4fffda0 (f9402801) 
[ 1061.635297] ---[ end trace 0000000000000000 ]---
[ 1061.639970] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[ 1061.647646] Kernel Offset: disabled
[ 1061.651135] CPU features: 0x00,00000080,00200000,0200420b
[ 1061.656533] Memory Limit: none
[ 1061.659592] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

Second boot, login as root then reboot:

root@DietPi ~# reboot
Failed to connect to bus: No such file or directory
[  693.539820] systemd-shutdown[1]: Failed to set watchdog hardware timeout to 10min: Invalid argument
[  693.832181] reboot: Restarting system

U-Boot SPL 2025.04-armbian-2025.04-S3482-Pf089-H8869-V3d5b-Bb703-R448a-dirty (May 25 2025 - 14:46:18 +0000)
DRAM base address is defined as 0x40000000
[... 515 lines cut ...]
[    6.496767] systemd[1]: Starting systemd-remount-fs.service - Remount Root and Kernel File Systems...
[    6.503068] fuse: init (API version 7.41)
[    6.524839] systemd[1]: Starting systemd-udev-trigger.service - Coldplug All udev Devices...
[    6.541553] systemd[1]: Mounted dev-hugepages.mount - Huge Pages File System.
[    6.549837] systemd[1]: Mounted dev-mqueue.mount - POSIX Message Queue File System.
[    6.558345] systemd[1]: Mounted sys-kernel-debug.mount - Kernel Debug File System.
[    6.566938] systemd[1]: Mounted sys-kernel-tracing.mount - Kernel Trace File System.
[    6.576007] systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
[    6.585909] systemd[1]: Finished kmod-static-nodes.service - Create List of Static Device Nodes.
[    6.596427] systemd[1]: modprobe@configfs.service: Deactivated successfully.
[    6.603666] EXT4-fs (mmcblk0p1): re-mounted b05f15b5-3440-4999-ac23-320a121a2271 r/w. Quota mode: none.
[    6.604485] systemd[1]: Finished modprobe@configfs.service - Load Kernel Module configfs.
[    6.625914] systemd[1]: modprobe@dm_mod.service: Deactivated successfully.
[    6.634889] systemd[1]: Finished modprobe@dm_mod.service - Load Kernel Module dm_mod.
[    6.644742] systemd[1]: modprobe@efi_pstore.service: Deactivated successfully.
[    6.653169] systemd[1]: Finished modprobe@efi_pstore.service - Load Kernel Module efi_pstore.
[    6.663530] systemd[1]: modprobe@fuse.service: Deactivated successfully.
[    6.671872] systemd[1]: Finished modprobe@fuse.service - Load Kernel Module fuse.
[    6.680395] systemd[1]: Started systemd-journald.service - Journal Service.
[    6.775270] Adding 1102844k swap on /var/swap.  Priority:-2 extents:4 across:2085884k SS
[    6.831381] systemd-journald[219]: Received client request to flush runtime journal.

Debian GNU/Linux 12 DietPi ttyS0

DietPi login:

MichaIng, 2025-06-19:
Many thanks for your report.

Can you check whether the RAM size is consistently detected as 1 GB?

Also did this only happen once on very first reboot so far? I mean there was neither a kernel nor bootloader upgrade in the meantime. Current latest images ship with latest U-Boot, which aims to solve the prior RAM size detection. Or is the original image older than May 26th?

adshrc, 2025-07-08:
Last week, I installed DietPi on 15 OPi Zero 3 devices, each with 1GB of RAM. The kernel panic occurred on 12 of them, but as you said, only during the first reboot. After cutting the power, they all booted normally into the OS.