MichaIng/DietPi #7591: Unable to run tailscale on radxa zero 3E (CONFIG_TUN issue)

nilz-bilz, 2025-06-20:
### Creating a bug report/issues
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='applied'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm
- Kernel version | `uname -a`
```
Linux bombay 6.1.115-vendor-rk35xx #1 SMP Thu May  1 16:13:28 UTC 2025 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Radxa ZERO 3E (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
5V 2A USB adapter
- SD card used | (EG: SanDisk ultra)
WD Purple

#### Additional Information
- Software title: Tailscale
- Was the software title installed freshly or updated/migrated?
freshly installed
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. Install tailscale from the script on their website, or from dietpi-software [lnk](https://tailscale.com/download/linux)
3.  run `sudo tailscale up`
4. Produces error

#### Expected behaviour
Should be able to run tailscaled after installation and start tunneling traffic
 
#### Actual behaviour
```
failed to connect to local tailscaled (which appears to be running as tailscaled, pid 16007). Got error: 503 Service Unavailable: no backend
```

#### Extra details
$ journalctl -u tailscaled

```bash
Jun 20 07:16:41 bombay systemd[1]: Starting tailscaled.service - Tailscale node agent...
Jun 20 07:16:41 bombay tailscaled[15934]: logtail started
Jun 20 07:16:41 bombay tailscaled[15934]: Program starting: v1.84.0-t160c11f37-gbf6042458, Go 1.24.2: []string{"/usr/sbin/tailscaled", "--state=/var/lib/tailscale/tailscaled.state", "--socket=/run/tailscale/tailscaled.sock", "--port=41641"}
Jun 20 07:16:41 bombay tailscaled[15934]: LogID: 62f10e172cb0386f89c0ecb130df901b5f536ff277dfed093b01f2e87d3ef68b
Jun 20 07:16:41 bombay tailscaled[15934]: logpolicy: using $STATE_DIRECTORY, "/var/lib/tailscale"
Jun 20 07:16:41 bombay tailscaled[15934]: logpolicy.ConfigFromFile /var/lib/tailscale/tailscaled.log.conf: open /var/lib/tailscale/tailscaled.log.conf: no such file or directory
Jun 20 07:16:41 bombay tailscaled[15934]: logpolicy.Config.Validate for /var/lib/tailscale/tailscaled.log.conf: config is nil
Jun 20 07:16:41 bombay tailscaled[15934]: dns: [rc=unknown ret=direct]
Jun 20 07:16:41 bombay tailscaled[15934]: dns: using "direct" mode
Jun 20 07:16:41 bombay tailscaled[15934]: dns: using *dns.directManager
Jun 20 07:16:41 bombay tailscaled[15934]: dns: inotify: NewDirWatcher: context canceled
Jun 20 07:16:41 bombay tailscaled[15934]: wgengine.NewUserspaceEngine(tun "tailscale0") ...
Jun 20 07:16:41 bombay systemd[1]: Started tailscaled.service - Tailscale node agent.
Jun 20 07:16:42 bombay tailscaled[15934]: Linux kernel version: 6.1.115-vendor-rk35xx
Jun 20 07:16:42 bombay tailscaled[15934]: is CONFIG_TUN enabled in your kernel? `modprobe tun` failed with: modprobe: ERROR: could not insert 'tun': Invalid argument
Jun 20 07:16:42 bombay tailscaled[15934]: wgengine.NewUserspaceEngine(tun "tailscale0") error: tstun.New("tailscale0"): no such device
[... 52 lines cut ...]
Jun 20 07:16:45 bombay systemd[1]: Starting tailscaled.service - Tailscale node agent...
Jun 20 07:16:45 bombay tailscaled[16072]: logtail started
Jun 20 07:16:45 bombay tailscaled[16072]: Program starting: v1.84.0-t160c11f37-gbf6042458, Go 1.24.2: []string{"/usr/sbin/tailscaled", "--state=/var/lib/tailscale/tailscaled.state", "--socket=/run/tailscale/tailscaled.sock", "--port=41641"}
```

Seems to be an issue with the 'tun' option not being enabled in the kernel

$ zcat /proc/config.gz | grep CONFIG_TUN
```bash
CONFIG_TUN=m
# CONFIG_TUN_VNET_CROSS_LE is not set
```

$ sudo modprobe tun
```bash
modprobe: ERROR: could not insert 'tun': Invalid argument
```

To change the option to CONFIG_TUN=y requires a kernel recompile which i haven't tried yet. 

I expected tailscale to be able to run out of the box with this OS. Please let me know if this is expected behaviour and how i can get around this

Reference pastes:
https://pastebin.nilz.dev/?35e5b8095c12df46#hz8gB3hVNRrF89o2gGgQHSYoUEaMMyma76m6m8RFEAt
https://pastebin.nilz.dev/?3c8bc1999b5ae1d2=#JBt5tXQDwSzAc5kz1qwFiD8XFPA8Fgea1yBUP1jxRrh1

MichaIng, 2025-06-20:
`CONFIG_TUN=m` means that the driver is available as kernel module, otherwise `modprobe tun` would not be required in the first place. This is common and expected. However, of course loading this driver is supposed to succeed.

I will try to replicate on another SBC with this kernel.

nilz-bilz, 2025-06-21:
Thanks for the feedback. After looking into it, it seems to work after running the following:
> Install Tailscale via dietpi-software
1. sudo reboot
2. sudo depmod -a
3. sudo modprobe tun

I tried replicating this a few times on fresh installs, and sometimes it requires an extra reboot to get it working. I suppose the mismatched kernel version after the upgrade might be the reason, though the replication is still a bit inconsistent.

This workaround seems to be working for now. Thanks a lot :)

MichaIng, 2025-06-21:
For anything shipped with the kernel, `depmod` and any such should not be necessary. I am sure it is related to the mismatch then. We do already check for changed kernel versions, and in case suggest a reboot, but difficult to identify if the kernel version did not actually change, i.e. `/lib/modules/$(uname -r)` does still exist.

closed 2025-06-21 by MichaIng