MichaIng/DietPi #7594: Open Beta v9.14 | Orange Pi Zero 3 dmesg error

lonewolf-jp, 2025-06-21:
### Creating a bug report/issue
- [ok] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=1
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm

- Kernel version | `uname -a`
Linux DietPi 6.12.23-current-sunxi64 SMP Thu Apr 10 12:39:41 UTC 2025 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME`
Orange Pi Zero 3 (aarch64)

- Power supply used
Original power supply 5V 3A

- SD card use
KIOXA EXCERIA 32GB

#### Additional Information (if applicable)
- Software title
DietPi Open Beta v9.14
DietPi v9.13 shows same error message on dmesg.

#### Steps to reproduce

1. Turn on Orange Pi Zero 3
2. login.
3. dmesg.
4. Dmesg shows error message as below.
[    1.772502] dw-apb-uart 5000000.serial: **Error applying setting, reverse things back**
[    1.772759] sun6i-spi 5010000.spi: **Error applying setting, reverse things back**
[    1.774874] axp20x-i2c 1-0036: AXP20x variant AXP313a found
[    1.776556] axp20x-i2c 1-0036: AXP20X driver loaded
[    1.777943] sunxi-mmc 4020000.mmc: **Error applying setting, reverse things back**
[    1.778402] sunxi-mmc 4021000.mmc: **Error applying setting, reverse things back**
[    1.778589] dw-apb-uart 5000000.serial: **Error applying setting, reverse things back**
[    1.778840] sun6i-spi 5010000.spi: **Error applying setting, reverse things back**

[dmesg-log.txt](https://github.com/user-attachments/files/20844672/dmesg-log.txt)

#### Expected behaviour

It should show no error message.

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2025-06-21:
It seems to be originating from regulator code, similar reports for sunxi boards exist: https://lkml.org/lkml/2020/10/21/558

Might be related to device tree nodes which are disabled, or not in use. To debug, one can traverse the device tree to find the nodes from the log: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner/sun50i-h618-orangepi-zero3.dts?h=linux-6.12.y

And in case check back Armbian patches: https://github.com/armbian/build/tree/main/patch/kernel/archive/sunxi-6.12/patches.armbian

Support in Linux 6.12 is still somewhat new, so latest edge kernel might enhance things.

I will have a look into it when I find time. But low priority until there is an actual issue aside of error messages in the logs. I agree that there should be no "expected errors to ignore", but it is sadly reality for a lot of SBCs, and solving it potentially very time consuming.

lonewolf-jp, 2025-06-21:
@MichaIng 
OK. I understand that there is no harmful behavior with such messages from your explanation.
I will wait until kernel mainteiner or armbian maintainer fixes it.