MichaIng/DietPi #7600: cannot upgrade from 9.13.2 to 9.14.2

zoltanhans, 2025-06-23:
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.14.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v9.13.2
[ INFO ] DietPi-Update | Latest version  : v9.14.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=188187 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : home-assistant
[  OK  ] DietPi-Services | stop : minidlna
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Get:3 https://packages.sury.org/php bookworm InRelease [7551 B]
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Hit:6 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:7 https://deb.debian.org/debian bookworm InRelease
Hit:8 https://dietpi.com/apt bookworm InRelease
Hit:9 https://repo.mongodb.org/apt/debian bookworm/mongodb-org/7.0 InRelease
Hit:10 https://dl.ui.com/unifi/debian unifi-9.1 InRelease
Err:3 https://packages.sury.org/php bookworm InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Fetched 7551 B in 4s (1730 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php bookworm InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
E: Failed to fetch https://packages.sury.org/php/dists/bookworm/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
E: Some index files failed to download. They have been ignored, or old ones used instead.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -y -eany update
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...

### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2025-06-23:
Pls fill the troubleshooting template that you have ignored

MichaIng, 2025-06-23:
You have a mix of APT repositories for Bookworm and Bullseye, and the `packages.sury.org` PHP repository which we DietPi used only on Debian Buster, and which is causing the issue due to an expired APT key. Did you manually upgrade your system from Buster to Bullseye to Bookworm without following our instructions/using our script for it?

... checking your bug report upload, there is `/etc/apt/sources.list.d/glennr-install-script.list` adding the Bookworm repo, while the main `sources.list` is still on Bullseye. Seems to be this one: https://github.com/pforret/glennr-unifi/blob/main/scripts/video/unifi-video-3.10.13.sh
- I remember the problem that these Unifi controllers do/did require Java 8: https://github.com/MichaIng/DietPi/discussions/5057
- This script in turn adds the Debian Stretch repo on newer distros, to install Java 8 from there. Stretch is 3 versions older than Bookworm: Stretch (Java 8) > Buster (Java 11) > Bullseye (Java 17) > Bookworm.
- It generates `/etc/apt/sources.list.d/glennr-install-script.list` in any case, where is basically replicates the Debian main repo. This is done for other (compatible) Debian based distros without Debian main repo, like Kali Linux and such.
- But it makes sense that at some point packages from ancient Debian versions conflict and cannot be installed on recent Debian.
- The script hence says in the header it supports Debian up to Bullseye only, and it checks for and exits in this case.
- So I have no idea how this list file with Bookworm inside can exist even that the script should have exited before creating it. Maybe an older version of the script did not exit on too new Debian versions yet?
- Furthermore, how could i detect Bookworm as system? The main list file is at Bullseye, hence the checks done by the script - `lsb_release` or `/etc/os-release` - both should have returned Bullseye.
- Maybe the culprit is the PHP repo, which contains the Bookworm suite as well. But that repo should not contain packages which affect `lsb_release` and `/etc/os-release` outputs 🤔.

We can provide the steps to clean this up. But if you have any idea how it could have happened, please share.

MichaIng, 2025-06-24:
Okay, installation went well here, using Java 21, and also the update script did not cause a distro upgrade. No idea what happened in your case, but does not look like it would happen again.

One thing the script does wrong is, aside of that `/etc/apt/sources.list.d/glennr-install-script.list` only duplicates the already existing Debian repo entries, it adds
```
deb https://archive.debian.org/debian-security bullseye-security
```
which is invalid: `archive.debian.org` has no security repo. It either needs to be `deb.debian.org/debian-security` or `https://security.debian.org/debian-security`, otherwise `apt update` throws an error. In your case, it is `deb.debian.org/debian-security`. No idea why it is different here.

Anyway, lets get your system upgraded properly:
```sh
rm /etc/apt/sources.list.d/glennr-install-script.list # APT throws duplication warnings otherwise
bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-bookworm-upgrade')
```
The software migration part migrates from PHP 7.4 to PHP 8.2 and from Python 3.9 to Python 3.11, and reinstall related software titles. And Home Assistant is reinstalled, to move to the latest version, not supported on Bullseye but Bookworm. So this can take a bit.

closed 2025-06-25 by zoltanhans