MichaIng/DietPi #7602: WARNING: Concurrent execution of DietPi-Set_software detected

vellanix, 2025-06-24:
I run the dietpi-update (9.13) and when I was asked to change the default password for root and dietpi users I pressed ESC (suppose to leave the old password).

The update process get finished but when I try to SSH  I always get the warning:

WARNING: Concurrent execution of DietPi-Set_software detected

with the following screen:

[image]

I tried to kill the process, reboot, shutdown and more but I always get the warning when I ssh with root user.

Any hep?

thank you

### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 0

- Kernel version | `uname -a`
Linux blade00 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi (CM5) (aarch64)

- Power supply used | Original

- SD card used | eMMC

#### Steps to reproduce
run dietpi-update (9.13) and when ask for password change (user root and dietpi) press ESC

MichaIng, 2025-06-24:
First of all, do not press ESC but ENTER/RETURN to confirm whiptail dialogues in general. ESC aborts them, respectively has the same effect as hitting the right button (if there is a right one).

You do have some autostart option enabled so that autologin happens on the local console? I.e. check `dietpi-autostart` and set it to "0" for now (confirm with ENTER/RETURN (or Ok button) 😉).

Since you aborted the password change, the autologin on local console shows the same dialogue.

Hmm, I wonder how to enhance this. We want the default password to be changed in any case. But even if no screen and keyboard is attached, it is theoretically possible to enable autologin on the local console, and there it is treated as interactive login shell, no real way to know whether it is even accessible or not 🤔. I guess it is a rare case that the password was not changed in the first place, and that an autologin option was selected without any actual screen attached to the device (?), but looks like we should find a way to detect this, and prevent the whiptail dialogue. Or it needs to be able to kill and "take over" the concurrent process.

If for whatever reason the dialogue shows up on tty1 despite autologin disabled (needs to be investigated separately in case), kill the `dietpi-login` instance on tty1. Those whiptail dialogues themselves cannot be killed properly, which is very nasty indeed, but their parent shells need to be killed instead. Then hit "Retry" on the SSH dialogue, or if aborted, run `exec bash` or open a new SSH session to trigger a new dialogue.

zumbiepig, 2026-09-03:
> Hmm, I wonder how to enhance this. We want the default password to be changed in any case.

You really should just let people do what they want with their system. If I want my password to be `dietpi` (which i genuinely do), let me make my password `dietpi`. Maybe after an "Are you sure? This default password is insecure." But I should be able to choose the password I want.

MichaIng, 2026-09-03:
You can do anything you want with `passwd`, but within the responsibility of our tools, and the default password we set, you must change it.

The only other option I have in mind is to not set a default password at all, but ship the account locked (expired), forcing users to set a password (or create a new account) on first boot. But it might have an impact on automated first run setup, i.e. without regression, one would need to set a password via `dietpiml.txt` then.

closed 2026-10-03 by MichaIng