MichaIng/DietPi #7616: Unbound returns SERVFAIL, Can't Use internet.

lutfor-diu, 2025-07-06:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

DietPi version | '9.14'
Distro version | bookworm 0
Kernel version | Linux DietPi 6.12.34+rpt-rpi-v8
SBC model | RPi 4 Model B (aarch64) or (EG: RPi3)
Power supply used | RPi 3A
SD card used | SanDisk 32GB

Additional Information (if applicable)
Software title | Adguard home + Unbound
Was the software title installed freshly or updated/migrated? 
ans: freshly Install/update. same
Can this issue be replicated on a fresh installation of DietPi? 
Ans: yes. when freshly install.

Bug report ID | echo $G_HW_UUID
Steps to reproduce
fresh install
can't use internet. in adguard home,  dns request return SERVFAIL. 
server: 127.0.0.1:5335

Expected behaviour
should work and use internet without any problem. but after install with unbound return SERVFAIL and can't use internet.

Actual behaviour
Normal internet use.

MichaIng, 2025-07-06:
Does it work when using an upstream DNS provider instead of Unbound?

Please show the service logs:
```sh
journalctl -u adguardhome -u unbound
```

lutfor-diu, 2025-07-06:
> Does it work when using an upstream DNS provider instead of Unbound?
> 
> Please show the service logs:
> 
> journalctl -u adguardhome -u unbound

yes. it works when using only adguradhome. but when install with unbound it returns SERVFAIL. i will add journalctl -u adguardhome -u unbound output soon.

lutfor-diu, 2025-07-06:
journalctl -u adguardhome -u unbound
Jul 06 21:00:04 DietPi systemd[1]: Started adguardhome.service - AdGuard Home (DietPi).
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.654773 [info] AdGuard Home, version v0.107.63
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.667380 [info] tls_manager: using default ciphers
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.686882 [info] Initializing auth module: /mnt/dietpi_userdata/adguardhome/data/sessions.db
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.693066 [info] auth: initialized.  users:1  sessions:1
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.693490 [info] webapi: initializing
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.910514 [info] dnsproxy: upstream mode is set mode=parallel
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.910580 [info] dnsproxy: cache enabled size=4096
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.911778 [info] dnsproxy: max goroutines is set count=300
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.911964 [info] dnsproxy: cache ttl override is enabled min=2400 max=86400
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.911996 [info] dnsproxy: ratelimit is enabled rps=20 ipv4_subnet_mask_len=24 ipv6_subnet_mask_len=56
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.912013 [info] dnsproxy: server will refuse requests of type any
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.912037 [info] dnsproxy: upstream mode is set mode=parallel
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.912056 [info] dnsproxy: cache enabled size=4194304
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.912079 [info] dnsproxy: max goroutines is set count=300
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.919778 [info] addrproc: processing addresses
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.920434 [info] permcheck: warning: found unexpected permissions type=directory path=/mnt/dietpi_userdata/adguardhome perm=0777 want=0700
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.920628 [info] webapi: AdGuard Home is available at the following addresses:
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.924061 [info] go to http://127.0.0.1:8083
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.924106 [info] go to http://192.168.10.10:8083
Jul 06 21:00:06 DietPi AdGuardHome[458]: 2025/07/06 21:00:06.924247 [info] starting plain server server=plain addr=0.0.0.0:8083
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.921291 ERROR response received addr=192.168.10.1:53 proto=udp status="exchanging with 192.168.10.1:53 over udp: read udp 192.168.10.10:41227->192.168.10.1:53: i/o timeout"
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.921502 [info] dnsproxy: starting dns proxy server
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.921587 [info] dnsproxy: creating udp server socket addr=0.0.0.0:53
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.921801 [info] dnsproxy: listening to udp addr=[::]:53
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.921828 [info] dnsproxy: creating tcp server socket addr=0.0.0.0:53
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.932347 [info] dnsproxy: listening to tcp addr=[::]:53
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.932630 [info] dnsproxy: entering udp listener loop addr=[::]:53
Jul 06 21:00:08 DietPi AdGuardHome[458]: 2025/07/06 21:00:08.932687 [info] dnsproxy: entering listener loop proto=tcp addr=[::]:53
Jul 06 22:02:04 DietPi AdGuardHome[458]: 2025/07/06 22:02:04.783503 ERROR response received addr=192.168.10.1:53 proto=udp status="exchanging with 192.168.10.1:53 over udp: read udp 192.168.10.10:45805->192.168.10.1:53: i/o timeout"
Jul 06 23:20:24 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 06 23:20:24 DietPi unbound-helper[5146]: Updating /var/lib/unbound/root.key from /usr/share/dns/root.key
Jul 06 23:20:24 DietPi unbound[5151]: [5151:0] info: start of service (unbound 1.17.1).
Jul 06 23:20:24 DietPi systemd[1]: Started unbound.service - Unbound DNS server.
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.494085 [info] received signal signal=terminated
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.494131 [info] stopping AdGuard Home
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.494155 [info] webapi: stopping http server
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.494362 [info] webapi: stopped http server
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.494472 [info] dnsproxy: stopping server
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.495053 [info] dnsproxy: stopped dns proxy server
Jul 06 23:20:26 DietPi systemd[1]: Stopping adguardhome.service - AdGuard Home (DietPi)...
Jul 06 23:20:26 DietPi AdGuardHome[458]: 2025/07/06 23:20:26.518330 [info] stopped
Jul 06 23:20:26 DietPi systemd[1]: adguardhome.service: Deactivated successfully.
Jul 06 23:20:26 DietPi systemd[1]: Stopped adguardhome.service - AdGuard Home (DietPi).
Jul 06 23:20:26 DietPi systemd[1]: adguardhome.service: Consumed 27.451s CPU time.
Jul 06 23:20:26 DietPi systemd[1]: Started adguardhome.service - AdGuard Home (DietPi).
[... cut ...]
Jul 06 23:20:28 DietPi AdGuardHome[5261]: 2025/07/06 23:20:28.726551 [info] dnsproxy: creating tcp server socket addr=0.0.0.0:53
Jul 06 23:20:28 DietPi AdGuardHome[5261]: 2025/07/06 23:20:28.737964 [info] dnsproxy: listening to tcp addr=[::]:53
Jul 06 23:20:28 DietPi AdGuardHome[5261]: 2025/07/06 23:20:28.738254 [info] dnsproxy: entering udp listener loop addr=[::]:53
Jul 06 23:20:28 DietPi AdGuardHome[5261]: 2025/07/06 23:20:28.738272 [info] dnsproxy: entering listener loop proto=tcp addr=[::]:53

lutfor-diu, 2025-07-06:
installed unbound few mins ago to get "journalctl -u adguardhome -u unbound" output. when i was out off internet due to SERVFAIL

[image]

lutfor-diu, 2025-07-06:
> for testing, can you do following:
> 
> ```
> apt install dnsutils
> dig @127.0.0.1 -p 5335 dietpi.com
> ```
> 
> Pls execute both commands separately. First one is to install `dnsutils` tool. The other to check Unbound functionality using `dig` command.

dig @127.0.0.1 -p 5335 dietpi.com

; <<>> DiG 9.18.33-1~deb12u2-Debian <<>> @127.0.0.1 -p 5335 dietpi.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 47330
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;dietpi.com.                    IN      A

;; Query time: 1563 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1) (UDP)
;; WHEN: Mon Jul 07 00:15:04 +06 2025
;; MSG SIZE  rcvd: 39

lutfor-diu, 2025-07-06:
> Did the Unbound server throw an error message in return?
> 
> journalctl -u unbound
> Did you change any config of it?

nothing changed. as it is. 
facing this issue after "bookworm" released.

lutfor-diu, 2025-07-06:
> journalctl -u unbound

root@DietPi:~# journalctl -u unbound
Jul 06 23:20:24 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 06 23:20:24 DietPi unbound-helper[5146]: Updating /var/lib/unbound/root.key from /usr/share/dns/root.key
Jul 06 23:20:24 DietPi unbound[5151]: [5151:0] info: start of service (unbound 1.17.1).
Jul 06 23:20:24 DietPi systemd[1]: Started unbound.service - Unbound DNS server.
Jul 06 23:24:10 DietPi systemd[1]: Stopping unbound.service - Unbound DNS server...
Jul 06 23:24:10 DietPi unbound[5151]: [5151:0] info: service stopped (unbound 1.17.1).
Jul 06 23:24:11 DietPi systemd[1]: unbound.service: Deactivated successfully.
Jul 06 23:24:11 DietPi systemd[1]: Stopped unbound.service - Unbound DNS server.
Jul 07 00:14:04 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 07 00:14:04 DietPi unbound-helper[7375]: Updating /var/lib/unbound/root.key from /usr/share/dns/root.key
Jul 07 00:14:04 DietPi unbound[7380]: [7380:0] info: start of service (unbound 1.17.1).
Jul 07 00:14:04 DietPi systemd[1]: Started unbound.service - Unbound DNS server.

MichaIng, 2025-07-06:
> facing this issue after "bookworm" released.

You mean you used Unbound before on a Bullseye system, and after upgrading to Bookworm, it starts to fail? Or is it a fresh Unbound install?

lutfor-diu, 2025-07-06:
> > facing this issue after "bookworm" released.
> 
> You mean you used Unbound before on a Bullseye system, and after upgrading to Bookworm, it starts to fail? Or is it a fresh Unbound install?

yes, worked on bullseye. but after bookworm it failed. i searched on internet and found that this problem is because of resolvconf and unbound resolvconf

Joulinar, 2025-07-06:
we can trace the communication between Unbound and root DNS servers with `tcpdump`. Or increase the trace level of Unbound. This has nothing to do with AGH and we should put this aside. Challange is with Unbound, as we saw above in the test using `dig`

lutfor-diu, 2025-07-06:
> dig @127.0.0.1 -p 5335 dietpi.com

 dig @127.0.0.1 -p 5335 dietpi.com

; <<>> DiG 9.18.33-1~deb12u2-Debian <<>> @127.0.0.1 -p 5335 dietpi.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 1342
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;dietpi.com.                    IN      A

;; Query time: 1579 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1) (UDP)
;; WHEN: Mon Jul 07 00:33:53 +06 2025
;; MSG SIZE  rcvd: 39

lutfor-diu, 2025-07-06:
> we can trace the communication between Unbound and root DNS servers with `tcpdump`. Or increase the trace level of Unbound. This has nothing to do with AGH and we should put this aside. Challange is with Unbound, as we have seen above in the test using `dig`

yes. adgurad home no problem. problem with unbound config or resolvconf file. i tried so many ways to solve this problem but fails, so last 6-7 months i'm using adguardhome stand along.

Joulinar, 2025-07-06:
For the test with `tcpdump` we need 2 SSH windows next to each other. It would be helpful if there is no further communication via unbound or AGH is set to not use unbound.

SSH session 1:

```
apt install tcpdump
tcpdump -i any -c500 -nn port 53 or port 5335
```

switch to SSH Session 2 and submit the following command

```
dig @127.0.0.1 -p 5335 dietpi.com
```

Stop `tcpdum` within SSH session 1

lutfor-diu, 2025-07-06:
[image]
https://github.com/trinib/AdGuard-WireGuard-Unbound-DNScrypt?tab=readme-ov-file#install-unbound-

from this tutorial if i use "sudo apt-get install resolvconf -y && sudo systemctl restart unbound-resolvconf.service" cmd that unbound and adguard home works untill raspberry pi reboot. after rebooting same as before (unbound not work). also, this cmd won't work after reboot.

lutfor-diu, 2025-07-06:
> For the test with `tcpdump` we need 2 SSH windows next to each other. It would be helpful if there is no further communication via unbound or AGH is set to not use unbound.
>
> SSH session 1:
>
> ```
> apt [...]

doing, upload it soon.

lutfor-diu, 2025-07-06:
> Okay, please raise the Unbound log level:
> 
> G_CONFIG_INJECT 'verbosity:' '	verbosity: 3' /etc/unbound/unbound.conf.d/dietpi.conf
> systemctl restart unbound
> dig @127.0.0.1 -p 5335 dietpi.com
> journalctl -u unbound
> > problem with unbound config or resolvconf file
> 
> Which resolvconf file? `/etc/resolv.conf` or `/etc/resolvconf.conf`, on the Unbound host?
> 
> There was an issue with the `openresolv` package, which configured the Unbound host itself to use Unbound as its own DNS resolver automatically, on a regular basis, restarting Unbound each time. However, this did not affect whether Unbound is able to resolve queries or not, and `openresolv` was never preinstalled on DietPi. There is the less intrusive `resolvconf` package, but it is not preinstalled on DietPi either. Both of them provide ways to dynamically manage `/etc/resolv.conf` from other applications. But this is mostly required only when using a DHCP server with a dynamic IP address. And a DNS resolver host of course should have a static IP address, where `/etc/resolv.conf` is best to be static as well, and none of those two packages are hence needed or reasonable.
> 
> EDIT: Checking via `tcpdump` whether DNS requests are reaching and being sent upstream by Unbound is also a good idea. Raising the log level provides moreless the same information, logging each incoming query and its processing.

 G_CONFIG_INJECT 'verbosity:' '   verbosity: 3' /etc/unbound/unbound.conf.d/dietpi.conf
systemctl restart unbound
dig @127.0.0.1 -p 5335 dietpi.com
journalctl -u unbound
[  OK  ] G_CONFIG_INJECT | Setting in /etc/unbound/unbound.conf.d/dietpi.conf adjusted:       verbosity: 3

; <<>> DiG 9.18.33-1~deb12u2-Debian <<>> @127.0.0.1 -p 5335 dietpi.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 10969
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;dietpi.com.                    IN      A

;; Query time: 2839 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1) (UDP)
;; WHEN: Mon Jul 07 00:48:27 +06 2025
;; MSG SIZE  rcvd: 39

Jul 06 23:20:24 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 06 23:20:24 DietPi unbound-helper[5146]: Updating /var/lib/unbound/root.key from /usr/share/dns/root.key
Jul 06 23:20:24 DietPi unbound[5151]: [5151:0] info: start of service (unbound 1.17.1).
Jul 06 23:20:24 DietPi systemd[1]: Started unbound.service - Unbound DNS server.
Jul 06 23:24:10 DietPi systemd[1]: Stopping unbound.service - Unbound DNS server...
Jul 06 23:24:10 DietPi unbound[5151]: [5151:0] info: service stopped (unbound 1.17.1).
Jul 06 23:24:11 DietPi systemd[1]: unbound.service: Deactivated successfully.
Jul 06 23:24:11 DietPi systemd[1]: Stopped unbound.service - Unbound DNS server.
Jul 07 00:14:04 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 07 00:14:04 DietPi unbound-helper[7375]: Updating /var/lib/unbound/root.key from /usr/share/dns/root.key
Jul 07 00:14:04 DietPi unbound[7380]: [7380:0] info: start of service (unbound 1.17.1).
Jul 07 00:14:04 DietPi systemd[1]: Started unbound.service - Unbound DNS server.
Jul 07 00:33:45 DietPi unbound[7380]: [7380:0] info: service stopped (unbound 1.17.1).
Jul 07 00:33:45 DietPi systemd[1]: Stopping unbound.service - Unbound DNS server...
Jul 07 00:33:45 DietPi systemd[1]: unbound.service: Deactivated successfully.
Jul 07 00:33:45 DietPi systemd[1]: Stopped unbound.service - Unbound DNS server.
Jul 07 00:33:45 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 07 00:33:46 DietPi unbound[7748]: [7748:0] info: start of service (unbound 1.17.1).
Jul 07 00:33:46 DietPi systemd[1]: Started unbound.service - Unbound DNS server.
Jul 07 00:48:24 DietPi unbound[7748]: [7748:0] info: service stopped (unbound 1.17.1).
Jul 07 00:48:24 DietPi systemd[1]: Stopping unbound.service - Unbound DNS server...
Jul 07 00:48:24 DietPi systemd[1]: unbound.service: Deactivated successfully.
Jul 07 00:48:24 DietPi systemd[1]: Stopped unbound.service - Unbound DNS server.
Jul 07 00:48:24 DietPi systemd[1]: Starting unbound.service - Unbound DNS server...
Jul 07 00:48:24 DietPi unbound[7828]: [1751827704] unbound[7828:0] debug: switching log to syslog
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] debug: chdir to /etc/unbound
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] debug: drop user privileges, run as unbound
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] debug: module config: "validator iterator"
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] notice: init module 0: validator
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] notice: init module 1: iterator
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] debug: target fetch policy for level 0 is 3
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] debug: target fetch policy for level 1 is 2
Jul 07 00:48:24 DietPi unbound[7828]: [7828:0] debug: target fetch policy for level 2 is 1
[... 311 lines cut ...]
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] debug: iterator[module 1] operate: extstate:module_wait_reply event:module_event_reply
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] info: iterator operate: query . NS IN
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] info: response for . NS IN
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] info: reply from <.> 192.36.148.17#53
[... cut ...]
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] debug: Failed to get a delegation, giving up
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] debug: return error response SERVFAIL
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] debug: validator[module 0] operate: extstate:module_wait_module event:module_event_moddone
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] info: validator operate: query . DNSKEY IN
Jul 07 00:48:27 DietPi unbound[7828]: [7828:0] debug: cache memory msg=66072 rrset=66072 infra=11517 val=66368 subnet=0

Joulinar, 2025-07-06:
I strongly recommend not to enter 127.0.0.1 in `/etc/resolv.conf`. Our experience shows that it is best to enter a Global Upstream DNS provider in the local `/etc/resolv.conf`. On the DietPi system there is nothing at all that needs to be blocked by an AdBlocker. (Unless you use a browser on the system itself). This has no effect on the AdBlocking of clients in your own network. It leads more to problems if you do it this way.

MichaIng, 2025-07-06:
WTF, ignore that. This "important" hint does not make sense. There is no reason to make the Unbound host use Unbound for DNS resolving, especially not if it is about ad blocking, which is relevant for a browser, but not for a server. The recommendation is usually the other way round, to explicitly NOT use the DNS resolver hosted by a system for this system itself, as it only causes issues, as maintenance can lead to the server breaking its own DNS. Please reverse that:
```sh
apt autopurge resolvconf
systemctl disable --now unbound-resolvconf
```
Be careful when following 3rd party info, as this was neither an official AdGuard Home nor Unbound nor DietPi repository.

lutfor-diu, 2025-07-06:
> doing, upload it soon.

tcpdump -i any -c500 -nn port 53 or port 5335
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
01:01:01.938705 lo    In  IP 127.0.0.1.58403 > 127.0.0.1.5335: UDP, length 51
01:01:01.939230 eth0  Out IP 192.168.10.10.7303 > 192.112.36.4.53: 27882% [1au] NS? . (28)
01:01:01.989197 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.7303: 27882 ServFail 0/0/1 (28)
01:01:01.989716 eth0  Out IP 192.168.10.10.39238 > 192.36.148.17.53: 6673% [1au] NS? . (28)
01:01:02.041387 eth0  In  IP 192.36.148.17.53 > 192.168.10.10.39238: 6673 ServFail 0/0/1 (28)
01:01:02.041733 eth0  Out IP 192.168.10.10.41274 > 192.203.230.10.53: 6497% [1au] NS? . (28)
01:01:02.086069 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.41274: 6497 ServFail 0/0/1 (28)
01:01:02.086553 eth0  Out IP 192.168.10.10.12772 > 192.112.36.4.53: 10382% [1au] NS? . (28)
01:01:02.131896 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.12772: 10382 ServFail 0/0/1 (28)
01:01:02.132221 eth0  Out IP 192.168.10.10.34196 > 199.7.91.13.53: 26353% [1au] NS? . (28)
01:01:02.178364 eth0  In  IP 199.7.91.13.53 > 192.168.10.10.34196: 26353 ServFail 0/0/1 (28)
01:01:02.178793 eth0  Out IP 192.168.10.10.12668 > 192.203.230.10.53: 14785% [1au] NS? . (28)
01:01:02.227357 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.12668: 14785 ServFail 0/0/1 (28)
01:01:02.227680 eth0  Out IP 192.168.10.10.27095 > 199.7.83.42.53: 16869% [1au] NS? . (28)
01:01:02.270623 eth0  In  IP 199.7.83.42.53 > 192.168.10.10.27095: 16869 ServFail 0/0/1 (28)
01:01:02.271009 eth0  Out IP 192.168.10.10.33977 > 192.36.148.17.53: 8641% [1au] NS? . (28)
01:01:02.313676 eth0  In  IP 192.36.148.17.53 > 192.168.10.10.33977: 8641 ServFail 0/0/1 (28)
01:01:02.314005 eth0  Out IP 192.168.10.10.61020 > 193.0.14.129.53: 34538% [1au] NS? . (28)
01:01:02.360033 eth0  In  IP 193.0.14.129.53 > 192.168.10.10.61020: 34538 ServFail 0/0/1 (28)
01:01:02.360366 eth0  Out IP 192.168.10.10.56914 > 198.97.190.53.53: 59504% [1au] NS? . (28)
01:01:02.408350 eth0  In  IP 198.97.190.53.53 > 192.168.10.10.56914: 59504 ServFail 0/0/1 (28)
01:01:02.408761 eth0  Out IP 192.168.10.10.7105 > 198.41.0.4.53: 57085% [1au] NS? . (28)
01:01:02.458672 eth0  In  IP 198.41.0.4.53 > 192.168.10.10.7105: 57085 ServFail 0/0/1 (28)
01:01:02.459139 eth0  Out IP 192.168.10.10.50109 > 192.112.36.4.53: 23514% [1au] NS? . (28)
01:01:02.510581 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.50109: 23514 ServFail 0/0/1 (28)
01:01:02.511001 eth0  Out IP 192.168.10.10.49686 > 192.5.5.241.53: 45599% [1au] NS? . (28)
01:01:02.558182 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.49686: 45599 ServFail 0/0/1 (28)
01:01:02.558566 eth0  Out IP 192.168.10.10.48459 > 199.7.83.42.53: 17259% [1au] NS? . (28)
01:01:02.602249 eth0  In  IP 199.7.83.42.53 > 192.168.10.10.48459: 17259 ServFail 0/0/1 (28)
01:01:02.602657 eth0  Out IP 192.168.10.10.32213 > 198.97.190.53.53: 40522% [1au] NS? . (28)
01:01:02.645347 eth0  In  IP 198.97.190.53.53 > 192.168.10.10.32213: 40522 ServFail 0/0/1 (28)
01:01:02.645659 eth0  Out IP 192.168.10.10.12400 > 202.12.27.33.53: 64413% [1au] NS? . (28)
01:01:02.694845 eth0  In  IP 202.12.27.33.53 > 192.168.10.10.12400: 64413 ServFail 0/0/1 (28)
01:01:02.695252 eth0  Out IP 192.168.10.10.37316 > 192.5.5.241.53: 47771% [1au] NS? . (28)
01:01:02.739315 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.37316: 47771 ServFail 0/0/1 (28)
01:01:02.739704 eth0  Out IP 192.168.10.10.18286 > 192.5.5.241.53: 36021% [1au] NS? . (28)
01:01:02.791295 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.18286: 36021 ServFail 0/0/1 (28)
01:01:02.791754 eth0  Out IP 192.168.10.10.9577 > 170.247.170.2.53: 40360% [1au] NS? . (28)
01:01:02.837723 eth0  In  IP 170.247.170.2.53 > 192.168.10.10.9577: 40360 ServFail 0/0/1 (28)
01:01:02.838051 eth0  Out IP 192.168.10.10.19036 > 202.12.27.33.53: 24459% [1au] NS? . (28)
01:01:02.880877 eth0  In  IP 202.12.27.33.53 > 192.168.10.10.19036: 24459 ServFail 0/0/1 (28)
01:01:02.881168 eth0  Out IP 192.168.10.10.65475 > 202.12.27.33.53: 49323% [1au] NS? . (28)
01:01:02.931769 eth0  In  IP 202.12.27.33.53 > 192.168.10.10.65475: 49323 ServFail 0/0/1 (28)
01:01:02.932169 eth0  Out IP 192.168.10.10.54561 > 198.41.0.4.53: 1209% [1au] NS? . (28)
01:01:02.982448 eth0  In  IP 198.41.0.4.53 > 192.168.10.10.54561: 1209 ServFail 0/0/1 (28)
01:01:02.982782 eth0  Out IP 192.168.10.10.16238 > 192.36.148.17.53: 51824% [1au] NS? . (28)
01:01:03.037611 eth0  In  IP 192.36.148.17.53 > 192.168.10.10.16238: 51824 ServFail 0/0/1 (28)
01:01:03.038023 eth0  Out IP 192.168.10.10.12395 > 192.5.5.241.53: 1921% [1au] NS? . (28)
01:01:03.086432 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.12395: 1921 ServFail 0/0/1 (28)
01:01:03.086806 eth0  Out IP 192.168.10.10.30695 > 198.41.0.4.53: 7184% [1au] NS? . (28)
01:01:03.140028 eth0  In  IP 198.41.0.4.53 > 192.168.10.10.30695: 7184 ServFail 0/0/1 (28)
01:01:03.140474 eth0  Out IP 192.168.10.10.11990 > 198.41.0.4.53: 58787% [1au] NS? . (28)
01:01:03.185935 eth0  In  IP 198.41.0.4.53 > 192.168.10.10.11990: 58787 ServFail 0/0/1 (28)
01:01:03.186294 eth0  Out IP 192.168.10.10.19334 > 193.0.14.129.53: 35567% [1au] NS? . (28)
01:01:03.233141 eth0  In  IP 193.0.14.129.53 > 192.168.10.10.19334: 35567 ServFail 0/0/1 (28)
01:01:03.233556 eth0  Out IP 192.168.10.10.16088 > 198.41.0.4.53: 39106% [1au] NS? . (28)
[... cut ...]
01:01:03.426222 eth0  In  IP 192.36.148.17.53 > 192.168.10.10.45437: 37676 ServFail 0/0/1 (28)
01:01:03.426881 eth0  Out IP 192.168.10.10.7709 > 198.97.190.53.53: 56405% [1au] NS? . (28)
01:01:03.470953 eth0  In  IP 198.97.190.53.53 > 192.168.10.10.7709: 56405 ServFail 0/0/1 (28)
01:01:03.471415 eth0  Out IP 192.168.10.10.49971 > 170.247.170.2.53: 65072% [1au] NS? . (28)
01:01:03.514470 eth0  In  IP 170.247.170.2.53 > 192.168.10.10.49971: 65072 ServFail 0/0/1 (28)
01:01:03.514838 lo    In  IP 127.0.0.1.5335 > 127.0.0.1.58403: UDP, length 39

lutfor-diu, 2025-07-06:
[image]

lutfor-diu, 2025-07-06:
> I would consider this hint to be extremely dangerous. Our install script automatically configures AGH/PiHole together with Unbound so that no user input or configuration is required at all.

i know brother. thats why i love dietpi! easy and user level config free. 

but last 6-7 month i was unable to use adguardhome with unbound.

Joulinar, 2025-07-06:
did you followed more steps on this stupid guide?

lutfor-diu, 2025-07-06:
> did you followed more steps on this stupid guide?

no brother. i tried flashing new dietpi image several times. new start, but no luck.

Joulinar, 2025-07-06:
can you test following as well

```
delv dnssec.works @127.0.0.1 -p 5335
```

MichaIng, 2025-07-06:
However, `resolvconf` and `unbound-resolvconf` should not be responsible for the failing queries. Hmm, Unbound seems to send out the query to 33 root servers, getting `THROWAWAY`, then giving up and returning a `SERVFAIL` instead.

... although, maybe it cause the issue: https://www.reddit.com/r/pihole/comments/yvd2lw/very_strange_or_very_stupid_unbound_problem_not/
But I do not understand the relationship, in case there is one.

@lutfor-diu 
Installing `resolvconf` and enabling `unbound-resolvconf.service` was the only thing you followed from this guide?
EDIT: I see, answered already.

Here the same issue reported by some (or same) on Pi-hole forum, more competently handled, but also no solution: https://discourse.pi-hole.net/t/very-strange-or-very-stupid-unbound-problem-not-resolvconf-resolvers/59219
There were some pointers to DNSSEC. I need to have dinner, will check the debug steps done at the Pi-hole forum later.

lutfor-diu, 2025-07-06:
> can you pls avoid doing screen prints. thx
> 
> ```
> cat /etc/fstab/
> ```

tcpdump -i any -c500 -nn port 53 or port 5335
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
01:10:14.743668 lo    In  IP 127.0.0.1.43536 > 127.0.0.1.5335: UDP, length 53
01:10:14.745178 eth0  Out IP 192.168.10.10.57976 > 192.203.230.10.53: 33813% [1au] NS? . (28)
01:10:14.792335 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.57976: 33813 ServFail 0/0/1 (28)
01:10:14.792850 eth0  Out IP 192.168.10.10.59686 > 199.7.83.42.53: 18115% [1au] NS? . (28)
01:10:14.836842 eth0  In  IP 199.7.83.42.53 > 192.168.10.10.59686: 18115 ServFail 0/0/1 (28)
01:10:14.837232 eth0  Out IP 192.168.10.10.22782 > 202.12.27.33.53: 55609% [1au] NS? . (28)
01:10:14.878870 eth0  In  IP 202.12.27.33.53 > 192.168.10.10.22782: 55609 ServFail 0/0/1 (28)
01:10:14.879233 eth0  Out IP 192.168.10.10.37342 > 192.112.36.4.53: 44353% [1au] NS? . (28)
01:10:14.927478 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.37342: 44353 ServFail 0/0/1 (28)
01:10:14.927967 eth0  Out IP 192.168.10.10.12869 > 192.203.230.10.53: 11054% [1au] NS? . (28)
01:10:14.974139 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.12869: 11054 ServFail 0/0/1 (28)
01:10:14.974548 eth0  Out IP 192.168.10.10.16761 > 192.36.148.17.53: 51065% [1au] NS? . (28)
01:10:15.025792 eth0  In  IP 192.36.148.17.53 > 192.168.10.10.16761: 51065 ServFail 0/0/1 (28)
01:10:15.026303 eth0  Out IP 192.168.10.10.47166 > 192.203.230.10.53: 63860% [1au] NS? . (28)
01:10:15.074661 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.47166: 63860 ServFail 0/0/1 (28)
01:10:15.075571 eth0  Out IP 192.168.10.10.43042 > 192.5.5.241.53: 51848% [1au] NS? . (28)
01:10:15.118401 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.43042: 51848 ServFail 0/0/1 (28)
01:10:15.118879 eth0  Out IP 192.168.10.10.12366 > 192.112.36.4.53: 21613% [1au] NS? . (28)
01:10:15.160137 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.12366: 21613 ServFail 0/0/1 (28)
01:10:15.160514 eth0  Out IP 192.168.10.10.18421 > 192.203.230.10.53: 18461% [1au] NS? . (28)
01:10:15.204449 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.18421: 18461 ServFail 0/0/1 (28)
01:10:15.204939 eth0  Out IP 192.168.10.10.30986 > 198.41.0.4.53: 40907% [1au] NS? . (28)
01:10:15.251694 eth0  In  IP 198.41.0.4.53 > 192.168.10.10.30986: 40907 ServFail 0/0/1 (28)
01:10:15.252115 eth0  Out IP 192.168.10.10.32231 > 192.36.148.17.53: 1063% [1au] NS? . (28)
01:10:15.295406 eth0  In  IP 192.36.148.17.53 > 192.168.10.10.32231: 1063 ServFail 0/0/1 (28)
01:10:15.295783 eth0  Out IP 192.168.10.10.35539 > 192.112.36.4.53: 43127% [1au] NS? . (28)
01:10:15.342785 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.35539: 43127 ServFail 0/0/1 (28)
01:10:15.343275 eth0  Out IP 192.168.10.10.33217 > 193.0.14.129.53: 16669% [1au] NS? . (28)
01:10:15.390800 eth0  In  IP 193.0.14.129.53 > 192.168.10.10.33217: 16669 ServFail 0/0/1 (28)
01:10:15.391183 eth0  Out IP 192.168.10.10.19289 > 199.7.91.13.53: 61971% [1au] NS? . (28)
01:10:15.444478 eth0  In  IP 199.7.91.13.53 > 192.168.10.10.19289: 61971 ServFail 0/0/1 (28)
01:10:15.445046 eth0  Out IP 192.168.10.10.51686 > 198.41.0.4.53: 62370% [1au] NS? . (28)
01:10:15.490894 eth0  In  IP 198.41.0.4.53 > 192.168.10.10.51686: 62370 ServFail 0/0/1 (28)
01:10:15.491288 eth0  Out IP 192.168.10.10.49506 > 192.5.5.241.53: 41712% [1au] NS? . (28)
01:10:15.537066 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.49506: 41712 ServFail 0/0/1 (28)
01:10:15.537563 eth0  Out IP 192.168.10.10.56382 > 192.203.230.10.53: 30907% [1au] NS? . (28)
01:10:15.589669 eth0  In  IP 192.203.230.10.53 > 192.168.10.10.56382: 30907 ServFail 0/0/1 (28)
01:10:15.590068 eth0  Out IP 192.168.10.10.37373 > 199.7.83.42.53: 37717% [1au] NS? . (28)
01:10:15.638170 eth0  In  IP 199.7.83.42.53 > 192.168.10.10.37373: 37717 ServFail 0/0/1 (28)
01:10:15.638658 eth0  Out IP 192.168.10.10.65395 > 170.247.170.2.53: 41584% [1au] NS? . (28)
01:10:15.682214 eth0  In  IP 170.247.170.2.53 > 192.168.10.10.65395: 41584 ServFail 0/0/1 (28)
01:10:15.682630 eth0  Out IP 192.168.10.10.15342 > 199.7.91.13.53: 246% [1au] NS? . (28)
01:10:15.725300 eth0  In  IP 199.7.91.13.53 > 192.168.10.10.15342: 246 ServFail 0/0/1 (28)
01:10:15.725836 eth0  Out IP 192.168.10.10.29339 > 170.247.170.2.53: 37075% [1au] NS? . (28)
01:10:15.768393 eth0  In  IP 170.247.170.2.53 > 192.168.10.10.29339: 37075 ServFail 0/0/1 (28)
01:10:15.768834 eth0  Out IP 192.168.10.10.15354 > 192.5.5.241.53: 55516% [1au] NS? . (28)
01:10:15.811535 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.15354: 55516 ServFail 0/0/1 (28)
01:10:15.811962 eth0  Out IP 192.168.10.10.62042 > 192.33.4.12.53: 60454% [1au] NS? . (28)
01:10:15.859400 eth0  In  IP 192.33.4.12.53 > 192.168.10.10.62042: 60454 ServFail 0/0/1 (28)
01:10:15.859878 eth0  Out IP 192.168.10.10.28735 > 192.33.4.12.53: 7827% [1au] NS? . (28)
01:10:15.910289 eth0  In  IP 192.33.4.12.53 > 192.168.10.10.28735: 7827 ServFail 0/0/1 (28)
01:10:15.910698 eth0  Out IP 192.168.10.10.59087 > 199.7.91.13.53: 35836% [1au] NS? . (28)
01:10:15.960610 eth0  In  IP 199.7.91.13.53 > 192.168.10.10.59087: 35836 ServFail 0/0/1 (28)
01:10:15.961078 eth0  Out IP 192.168.10.10.11255 > 192.112.36.4.53: 24565% [1au] NS? . (28)
01:10:16.012331 eth0  In  IP 192.112.36.4.53 > 192.168.10.10.11255: 24565 ServFail 0/0/1 (28)
[... cut ...]
01:10:16.200659 eth0  In  IP 170.247.170.2.53 > 192.168.10.10.14005: 40509 ServFail 0/0/1 (28)
01:10:16.201033 eth0  Out IP 192.168.10.10.51110 > 192.58.128.30.53: 50962% [1au] NS? . (28)
01:10:16.245317 eth0  In  IP 192.58.128.30.53 > 192.168.10.10.51110: 50962 ServFail 0/0/1 (28)
01:10:16.245846 eth0  Out IP 192.168.10.10.40503 > 192.5.5.241.53: 41161% [1au] NS? . (28)
01:10:16.290950 eth0  In  IP 192.5.5.241.53 > 192.168.10.10.40503: 41161 ServFail 0/0/1 (28)
01:10:16.291308 lo    In  IP 127.0.0.1.5335 > 127.0.0.1.43536: UDP, length 41

MichaIng, 2025-07-06:
`THROWAWAY` seems to happen if the DNS response is somehow not trusted at some point. But weird is that the queried root servers seem to give this response, and not Unbound, but let's check whether the related files are readable by the Unbound service:
```sh
ls -l /var/lib/unbound/root.hints
ls -l /var/lib/unbound/root.key /usr/share/dns/root.key
ls -l /usr/share/dns/root.ds
```
I checked the script which prints `Updating /var/lib/unbound/root.key from /usr/share/dns/root.key`, and it should do that only if the first file is newer than the second, hence it happens only once. In your case, the same was printed however on every Unbound restart, so something must fail there, and the script indeed does not seem to error out, if one of its commands fails. But it should print an error message, which is not seen in your logs. A bit weird.

The trust anchor also contains info about when it was and will be probed, whether it as successful etc. Let's check that:
```sh
cat /var/lib/unbound/root.key
```
And please also show your Unbound config:
```sh
cat /etc/unbound/unbound.conf.d/dietpi.conf
```
Ah, and since DNSSEC requires somewhat precise system time, can you force a time sync and check whether this changes something?
```sh
/boot/dietpi/func/run_ntpd 1
dig dnssec.works @127.0.0.1 -p 5335 +dnssec
```

lutfor-diu, 2025-07-07:
```
 ls -l /var/lib/unbound/root.hints
ls -l /var/lib/unbound/root.key /usr/share/dns/root.key
ls -l /usr/share/dns/root.ds
-rw-r--r-- 1 root root 3310 Jul  7 00:13 /var/lib/unbound/root.hints
-rw-r--r-- 1 root    root    768 Jan  8 06:09 /usr/share/dns/root.key
-rw-r--r-- 1 unbound unbound 768 Jan  8 06:09 /var/lib/unbound/root.key
-rw-r--r-- 1 root root 166 Jan  8 06:09 /usr/share/dns/root.ds
```

> cat /var/lib/unbound/root.key

cat /var/lib/unbound/root.key
. IN DNSKEY 257 3 8 AwEAAaz/tAm8yTn4Mfeh5eyI96WSVexTBAvkMgJzkKTOiW1vkIbzxeF3+/4RgWOq7HrxRixHlFlExOLAJr5emLvN7SWXgnLh4+B5xQlNVz8Og8kvArMtNROxVQuCaSnIDdD5LKyWbRd2n9WGe2R8PzgCmr3EgVLrjyBxWezF0jLHwVN8efS3rCj/EWgvIWgb9tarpVUDK/b58Da+sqqls3eNbuv7pr+eoZG+SrDK6nWeL3c6H5Apxz7LjVc1uTIdsIXxuOLYA4/ilBmSVIzuDWfdRUfhHdY6+cn8HFRm+2hM8AnXGXws9555KrUB5qihylGa8subX2Nn6UwNR1AkUTV74bU= ; keytag 20326
. IN DNSKEY 257 3 8 AwEAAa96jeuknZlaeSrvyAJj6ZHv28hhOKkx3rLGXVaC6rXTsDc449/cidltpkyGwCJNnOAlFNKF2jBosZBU5eeHspaQWOmOElZsjICMQMC3aeHbGiShvZsx4wMYSjH8e7Vrhbu6irwCzVBApESjbUdpWWmEnhathWu1jo+siFUiRAAxm9qyJNg/wOZqqzL/dL/q8PkcRU5oUKEpUge71M3ej2/7CPqpdVwuMoTvoB+ZOT4YeGyxMvHmbrxlFzGOHOijtzN+u1TQNatX2XBuzZNQ1K+s2CXkPIZo7s6JgZyvaBevYtxPvYLw4z9mR7K2vaF18UYH9Z9GNUUeayffKC73PYc= ; keytag 38696

> cat /etc/unbound/unbound.conf.d/dietpi.conf

```
cat /etc/unbound/unbound.conf.d/dietpi.conf
# https://nlnetlabs.nl/documentation/unbound/unbound.conf/
server:
        # Do not daemonize, to allow proper systemd service control and status estimation.
        do-daemonize: no

        # A single thread is pretty sufficient for home or small office instances.
        num-threads: 1

        # Logging: For the sake of privacy and performance, keep logging at a minimum!
        # - Verbosity 2 and up practically contains query and reply logs.
        verbosity: 3
        log-queries: no
        log-replies: no
        # - If required, uncomment to log to a file, else logs are available via "journalctl -u unbound".
        #logfile: "/var/log/unbound.log"

        # Set interface to "0.0.0.0" to make Unbound listen on all network interfaces.
        # Set it to "127.0.0.1" to listen on requests from the same machine only, useful in combination with Pi-hole.
        interface: 127.0.0.1
        # Default DNS port is "53". When used with Pi-hole, set this to e.g. "5335", since "5353" is used by mDNS already.
        port: 5335

        # Control IP ranges which should be able to use this Unbound instance.
        # The DietPi defaults permit access from official local network IP ranges only, hence requests from www are denied.
        access-control: 0.0.0.0/0 refuse
        access-control: 10.0.0.0/8 allow
        access-control: 127.0.0.1/8 allow
        access-control: 172.16.0.0/12 allow
        access-control: 192.168.0.0/16 allow
        access-control: ::/0 refuse
        access-control: ::1/128 allow
        access-control: fd00::/8 allow
        access-control: fe80::/10 allow

        # Private IP ranges, which shall never be returned or forwarded as public DNS response.
        # NB: 127.0.0.1/8 is sometimes used by adblock lists, hence DietPi by default allows those as response.
        private-address: 10.0.0.0/8
        private-address: 172.16.0.0/12
        private-address: 192.168.0.0/16
        private-address: 169.254.0.0/16
        private-address: fd00::/8
        private-address: fe80::/10

        # Define protocols for connections to and from Unbound.
        # NB: Disabling IPv6 does not disable IPv6 IP resolving, which depends on the clients request.
        do-udp: yes
        do-tcp: yes
        do-ip4: yes
        do-ip6: no

[... 58 lines cut ...]
[  OK  ] DietPi-TimeSync | mkdir -p /run/systemd/timesync

; <<>> DiG 9.18.33-1~deb12u2-Debian <<>> dnssec.works @127.0.0.1 -p 5335 +dnssec
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 65165
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags: do; udp: 1232
;; QUESTION SECTION:
;dnssec.works.                  IN      A

;; Query time: 1701 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1) (UDP)
;; WHEN: Mon Jul 07 06:31:12 +06 2025
;; MSG SIZE  rcvd: 41

```

> cat /etc/resolv.conf

cat /etc/resolv.conf

nameserver 9.9.9.9

MichaIng, 2025-07-07:
> unbound DoT and DoH works. but without forwarding, facing that wired error.

What do you mean "without forwarding"? I remember AdGuard Home supports downstream DoH and DoT, but this is indeed not what I mean, and not related to Unbound. Unbound can resolve via upstream DoH or DoT providers, instead of doing recursive DNS resolving via root servers, like so for DoT: https://dietpi.com/docs/software/dns_servers/#unbound-activating-dns-over-tls-dot

Cloudflare was generally blocked in Shenzhen when I was there, IIRC, but Quad9 might work. However, this is what failed in your case? At least for DoT, I would not wonder, since it runs on an own port and hence can be filtered easily. DoH is more difficult to block, reminds me to add this to our docs. But in the end, the question is what you want to achieve with Unbound. Recursive DNS resolving via root servers is to avoid depending on a 3rd party provider. But DoT and DoH implicitly requires a 3rd party provider. Recursive resolving is for privacy towards 3rd party organizations, DoT/DoH for privacy towards everything between you and the DNS provider: local network, ISP, backbone, ... If the first does not work and the latter is not needed, then there is no point to use Unbound.

MichaIng, 2025-07-08:
For further analysis, I think we would need to get a particular DNS response from a particular DNS root server, capture it via `tcpdump`, compare the content to what someone from a different location (where it works) get, in case making additional use of `tshark`.

In theory it should reveal a modification of the response, so that Unbound cannot verify it with the key anymore. The only open question regarding this theory is why there is not clear error message of Unbound about this, but only a debug message, otherwise failing silently.

quickgrooves, 2025-08-02:
I'm had this exact same issue since June and have had to abandon Unbound/recursive DNS altogether for Pihole to function. Tried fresh installs of Pihole and Unbound several times and numerous troubleshooting, including what's been suggested above. 

Some notes from my testing:
    dig @1.1.1.1 google.com worked consistently so my Pi had no network connectivity issues, and my ISP wasn't blocking standard DNS traffic.

    dig @a.root-servers.net google.com +trace worked so my Pi/network could successfully perform a full recursive lookup, sending requests to the root servers and receiving the responses back.

So the problem wasn't my network, my firewall, or my ISP. Seems like a very specific, low-level issue with the Unbound service itself that prevented it from doing the exact same thing my dig command was able to do. After one last clean install of Unbound and applying a simple and known-good config, I still would get SERVFAIL. Even running it as the root user outside of its sandbox wasn't successful. As soon as I set my Pihole's upstream DNS to a public DNS server, everything worked fine. Have had to give up on Unbound for now, but really want to get back to a functioning recursive DNS.

lutfor-diu, 2025-08-02:
> I've had this exact same issue since June and have had to abandon Unbound/recursive DNS altogether for Pihole to function. Tried fresh installs of Pihole and Unbound several times and numerous troubleshooting, including what's been suggested above.
> 
> Some notes from my testing: dig @1.1.1.1 google.com worked consistently so my Pi had no network connectivity issues, and my ISP wasn't blocking standard DNS traffic.
> 
> ```
> dig @a.root-servers.net google.com +trace worked so my Pi/network could successfully perform a full recursive lookup, sending requests to the root servers and receiving the responses back.
> ```
> 
> So the problem wasn't my network, my firewall, or my ISP. Seems like a very specific, low-level issue with the Unbound service itself that prevented it from doing the exact same thing my dig command was able to do. After one last clean install of Unbound and applying a simple and known-good config, I still would get SERVFAIL. Even running it as the root user outside of its sandbox wasn't successful. As soon as I set my Pihole's upstream DNS to a public DNS server, everything worked fine. Have had to give up on Unbound for now, but really want to get back to a functioning recursive DNS.

In my case, exactly same issue.

MichaIng, 2025-08-02:
Does our own up-to-date Unbound build change something about that?
```sh
cd /tmp
wget "https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/unbound_$G_HW_ARCH_NAME.deb"
apt install ./"unbound_$G_HW_ARCH_NAME.deb"
```

Babbalucio, 2025-08-11:
May I suggest an alternative way to set up adguard home + unbound?
Instead of running everything on a VM, pull a recent 1.23 unbound docker image.
Keep adguard running on the host (listening on 53), and unbound docker (listening on 5335)

MichaIng, 2025-08-11:
What would Unbound in a Docker container change about the fact that AdGuard Home needs to use `127.0.0.1:5335` as only upstream?

If the Unbound build for whatever reason is the culprit, I suggest to test our own build first. If this does not change something, `tcpdump` and `tshark` can be used to compare the exact packages received when doing the same query from a system where the same setup works. I still guess it is some ISP intervention, but the packet inspection will reveal.

Docker just adds another layer of potential failures.

Babbalucio, 2025-08-11:
I say docker container cause the unbound version is up to date in the docker container
Yes, it's another layer, but I have used unbound this way in the past without problems...
Now that I'm trying on this new updated dietpi, Im getting this SERVFAIL problem similar to the reported issue. But maybe i'm just breaking something else, just ignore. Anyway, pulling a docker image is easy to do.

MichaIng, 2025-08-11:
> I say docker container cause the unbound version is up to date there.

Our package is also the very latest Unbound version.

> I say docker container cause the unbound version is up to date there.

You mean installing Unbound via `dietpi-software`, or you mean installing the package I linked above? Currently, we do not ship our own builds with `dietpi-software` or APT yet, but for testing, since I need to do some choices about how many differences Debian's package has compared to upstream I apply to our package as well. So it is a balance between keeping an upgrade from Debian's package as minimal change, or staying closer to how upstream does.

In any case, our package ships the same `/etc/unbound/unbound.conf.d/dietpi.conf` that our `dietpi-software` option installs on top of the Debian package. So if you previously used an instance in a Docker container, check and in case adjust the settings in `/etc/unbound/unbound.conf.d/dietpi.conf`. E.g. it sets the port to 53, and in case of Unbound, the setting defined first has priority, so changing the port in some `unbound.conf.d/local.conf` alphanumerically sorted after ours, it won't have any effect.

Babbalucio, 2025-08-11:
> What I said. My suggestion was to test our dedicated builds, which are currently **not** installed automatically by `dietpi-software`: [#7616 (comment)](https://github.com/MichaIng/DietPi/issues/7616#issuecomment-3146743912)

Ok, sorry, I was missing that specific post

I was focused on dietpi-software, that prompts for unbound after installing adguardhome
and the version you get there it's 1.17.x, or similar

will test this newest one in the future

My hybrid approach is still valid if you want to exclude both the provided unbound versions and run just unbound in a sandbox environment.

Babbalucio, 2025-08-11:
that's why Im waiting to test later, I have also noticed some weird routing in europe in the past few days.

Will try the next week with the provided 1.23.1 package.

Redis is a must if you update/reboot the system regularly
In case you run unbound as a docker container, it's practically mandatory... 
(for example, dietpi-backup will stop docker temporarily, and that will reset unbound cache)

MichaIng, 2025-08-11:
> Redis is a must if you update/reboot the system regularly

Why that? Note that most DNS records have a TTL of 1-5 minutes. So unless you do multiple reboots (oder Docker or service restarts) in this short period, you won't benefit from the persistent cache. As said, this "sounds" like it would be a benefit, but practically I need to try hard with a very special constructed scenario too see any real benefit in a real use case. Just read the discussion in the PR where we checked some individual more or less related settings, their behavior, and how their defaults (upstream) changed recently.

MichaIng, 2025-08-11:
You can configure Unbound to serve the expired cache entry first, yes. `prefetch:` is actually an argument against the need to do this, since it updates the cache before its entries expire. Both has not much to do with the question whether a dedicated/persistent cache has any noticeable benefit or not, neither regarding the number of upstream queries (network strain), nor end user performance.

> The difference between opening a site with a warm cache and a fresh cache on the first request is pretty big for me, I have to check my router leds to see if it's still up or not xD

WTF, how long does a `dig 1.1.1.1 dietpi.com` or such take in your case? 122 ms for me, I guess it can be 500 ms in bad cases. But again this affects a single request, and modern browsers prefetch DNS queries in various circumstances before you actually access the page.

Babbalucio, 2025-08-12:
ik, basically there are 2 different ways to use serve-expired
one conservative and one much more immediate
each one needs a specific "settings combo" to work properly
it's always well documented as you see

redis is needed for specific user cases, has its clear advantages but only if you have a decent hard disk

Will update if I find out something interesting with the provided 1.23.1 deb file.

MichaIng, 2025-08-12:
> redis is needed for specific user cases, has its clear advantages but only if you have a decent hard disk

I wouldn't call it "needed", and I still believe that most users who rate it useful now would not even recognize if the persistent cache was gone over night. Of course I believe you when you say you do recognize first website access delay with cold cache, but you probably do also pay attention then. E.g. I have my desktop browse cache on a RAM disk, hence start with cleared cache each day. Sure there is a very notable delay when loading any website the first time, needing to pull dozens of assets. But in practice, I start my browser, I meanwhile check emails (which triggers DietPi, GitHub and many other queries while syncing, filling the cache), start maybe GitHub desktop, which pulls repos in the background (thereby queries GitHub DNS as well), and my home server would have loaded `raw.githubusercontent.com` and `dietpi.com` into the cache, if an Unbound package scheduled a service restart, with DietPi update check and MOTB running afterwards. So even if there is theoretically a one-time delay at some point, I would just not recognize it at all if not paying particular attention, and it has zero effect on my schedule. In any case, we are talking about ~122 ms.

And Unbound is the last one in a series of DNS caching instances:
- Browsers have their own instance. In Chromium-based browsers it is cleared on restart, and strictly sticks to TTL, but you hence won't recognize a sudden Unbound restart.
- OSes often have another DNS cache layer, for Windows the same applies as for Chromium-based browsers: lost on reboot, strictly sticking to TTL. However, means if you restart your browser and Unbound, the OS would cover it.
- AdGuard Home also has its own DNS cache, like Pi-hole. It is not persistent, but you can raise minimum TTL and let it respond with expired entries (before trying to update them, just like `serve-expired: yes` + `serve-expired-client-timeout: 0` in Unbound): https://github.com/AdguardTeam/AdGuardHome/wiki/Configuration#configuration-file
  So even if OS + Unbound is restarted and loose their cache at the same time, AdGuard Home can cover it.

So for Unbound persistent cache to even become relevant (in combination with `serve-expired: yes` + `serve-expired-client-timeout: 0`), the whole server hosting Unbound + AdGuard Home would need to be restarted (both services individually is very unlikely to ever happen), and at the same time your client system, or relevant DNS entries would need expire + there must be no other client fetching DNS records after server restart, like a mobile or desktop app, email (like notification mails fetching assets, thereby resolving hostnames), etc, and no other person which might use the same network and AdGuard Home + Unbound instance must take the decasecond order delay for you.

However, if despite all these factors, you really feel a notable difference in daily browsing schedule when setting up using a persistent Unbound cache, at least the disk storage speed does not matter: Redis is an in-memory database with optional sporadic sync to disk to allow it survive restarts, not an on-disk database with memory cache like MariaDB. So even with a 10 MB/s SD card, it won't slow down your DNS queries, even the SD card wear is limited, thanks to Linux write buffer and the relatively small size of the database, being synced only every now and then 🙂.

Babbalucio, 2025-08-12:
Yes and no, I agree partially about the redis pro and cons.

You don’t need a fast disk for redis, but one of the main goal of dietpi is minimizing disk writes, to prevent sd cards wearing.
Even if it’s a minumal wearing, in this case i would avoid redis.

Having a small/medium home network (25/30 devices), of different types, the impact of an empty cache is evident, with pretty noticeable hiccups on normal daily activities.

Adguardhome cache is good, and it’s getting better and better (I also use a forced min ttl to maximize cache hits). But it’s not persistent either.

I’m n case you use unbound as adguardhome forwarder, you can have a second layer of cache. 

But you reboot the system and both caches are lost.

browsers cache/prefetchers, OS dns cache, will mitigate a bit the problem of an empty cache

If the system never needs to be rebooted, ok, redis becomes pretty much irrelevant.

But if the system needs regular reboots, I find more reasonable to make at least one cache persistent.
And unbound+redis is the most reliable way to achieve that.

MichaIng, 2025-08-12:
So far so good.

Note that DNSSEC is enabled via dedicated config `/etc/unbound/unbound.conf.d/root-auto-trust-anchor-file.conf`. This is taken over from the Debian package, to keep the change of config file structure minimal.

And I would generally recommend to keep it enabled, especially for recursive DNS resolving. Since all is plain text without any other authentication mechanism, there is nothing else which could assure you a genuine response. But of course this exactly could cause SERVFAIL responses (I guess, if Unbound does not log DNSSEC failures differently) if e.g. ISP really manipulates DNS requests to enforce local locks. I would then think about switching to upstream DoT, or DoH/oDoH via `dnscrypt-proxy`. In the latter case without Unbound, as long as either persistent caching is not needed, or `dnscrypt-proxy` supports it. _AdGuard Home does not at time of writing._ This is, as long as it does not trigger any local law enforcement. I am not aware that bypassing the ISPs DNS + in case DNS inceptions would be treated like a crime anywhere, but better be sure.

> I have already tuned my system buffer/connections in a `/etc/sysctl.d/99-adguard-dns.conf` file

Oh right, there was a new warning to raise `net.core.rmem_max` and `net.core.wmem_max` or so? To which value? I will add this to `dietpi-software`.

> and change `use-caps-for-id` to no, cause it seems sometimes it can cause issues

Did you face actual issues with this set to `yes`? I mean the number of DNS root servers is limited, this is enabled by default not only in our config, but also in Pi-holes suggested config and in many other cases. So I expect DNS root servers meanwhile all support and handle random casing correctly, otherwise we should see related reports more often, considering the ... above 20,000 Unbound instances on DietPi systems in the wild 🤔. So if this causes issues, that might be another hint that those queries are manipulated at some point.

Babbalucio, 2025-08-12:
I will leave the adguard home DNSSEC option disabled, cause we are already sure unbound recursive is doing the checks internally (dig commands confirm that)

as you say... less is more

having another option enabled just to see the DNSSEC icon in adguard logs, it's another potential risk to break something.

MichaIng, 2025-08-12:
Best would be to use the AdGuard Home cache only, which is closer to clients. But since it cannot be configured as good, I would also stick with the one in Unbound. As long as it does not cause memory strains, I would also just leave the one in AdGuard Home enabled, as you say. It reduces requests between AGH and Unbound and can hence serve clients a little faster with a little less overhead.

But ... hmm, `prefetch: yes` functionality could be actually counter-acted with an AGH cache, as long as it does not support DNS prefetch as well 🤔. If AGH does not query Unbound before the records are expired, then they will expire in Unbound as well, hence `prefetch: yes` won't have any effect. And if AGH and Unbound are on the same system, the requests between them should cause only very minimal delay and overhead, far below those added by an upstream query.

And then again, when configuring Unbound to serve expired cache entries right away, before updating those afterwards, `prefetch: yes` is basically irrelevant anyway, or do I miss something? Should not play a role whether Unbound updates the cache entry on incoming query for a cache entry which reached 90% of its TTL, or whether it does so when the cache entry expired, as long as it does serve the cache entry right in both cases. In that case, it could even make sense to disable `prefetch: yes` to reduce upstream queries a bid 🤔. So either use prefetch, then better disable AGH cache for prefetch in Unbound to work, or serve expired cache entries directly (as you do) in which case AGH cache should not have any negative effect, but might reduce delay and processing marginally, for a little added memory usage.

MichaIng, 2025-08-12:
Interesting that Unbound did in fact prefetch quite often. I would have expected the following per entry:
- Client queries AGH => no cache entry
- AGH queries Unbound => no ache entry
- Unbound queries upstream => answer with TTL
- Unbound stores cache entry and adjusts TTL according to `cache-min-ttl` and `cache-max-ttl`.
- Unbound forwards the answer with adjusted TTL to AGH
- AGH stores cache entry and adjusts TTL according to its `cache_ttl_min` and `cache_ttl_max`, which I assume are not narrower than the ones from Unbound? Hence same TTL.
- Clients continue to query AGH which will answer from cache until TTL has been reached.
- Once entry expired, on next request, AGH queries Unbound. Since both cache entries with same TTL, the one in Unbound has expired as well, slightly earlier by the time it took for Unbound to forward the answer to AGH. Hence Unbound should not be able to ever "pre"fetch, since all cache entries are assured to be expired before AGH queries Unbound for each respectively.

The only way I can think of how Unbound should be ever in a situation to prefetch, is if `cache_ttl_max` in AGH has been configured to be significantly lower than `cache-max-ttl` in Unbound. That way it would be possible that an entry has expired in AGH while it is still valid in Unbound. But it needs to be very low in AGH to cause a significant share of prefetches, since most TTLs are very low, like 1 minute in case of GitHub and 5 minutes hardcoded for Cloudflare proxied domains. So most cache entries will never reach any close to max, unless reduced to like 10 minutes or so.

So unless you really reduced the max TTL in AGH, either I am missing something, or Unbound counts something as a prefetch differently than I assume: only in case of an incoming request done while the cache entry reached between 90% and <100% of its TTL.

Babbalucio, 2025-08-12:
nope, AGH has the max size cache available.
TTL are untouched in AGH, only changes are in dietpi.conf
 cache-min-ttl: 300
 serve-expired: yes
 serve-expired-ttl: 86400
 serve-expired-client-timeout: 0

Maybe it's the AGH optimistic cache algorithm?

Babbalucio, 2025-08-13:
> So upstream DoH with let's say Cloudflare DNS is way faster for you than recursive resolving via Unbound? That is interesting, I would not expect such a significant difference 

I have consistent 25-40ms resolving on the most popular DoH servers: clodflare, google, opendns, controld
Quad9 just 10ms slower. 
I was using the four above in parallel and quad9 as fallback. 

Typical unbound recursion is around 200ms, problematic ones well.. 500/1000

If the cache works well I cant see difference ofc

closed 2025-09-21 by MichaIng

Babbalucio, 2025-09-21:
Yes, it's resolved on my end, and I've been using Unbound + AdGuard Home for a while now.
What was the solution?

I'm not 100% sure, but it was likely a combination of factors:

- Upgrading to a much newer Unbound build (probably the main factor).
- Disabling DNSSEC in AdGuard Home, which also means you shouldn’t mix Unbound with other public DNS servers in your forwarders.
- Be careful with EDNS client subnet, (you have that option both in AGH/unbound): better don't play with that (I think it also uses a different unbound cache/module, it's another failure point)

Thank you

MrBuyakah, 2026-01-30:
Well, it is the government, after all.  Not sure how compliant they may be in their other sites too.  I may turn back on this setting as my default and use a workaround for sites that are not compliant.  Thanks for the explanation.