MichaIng/DietPi #7618: No leds after upgrade to 9.14 (6.12 kernel) on rock64

sanlupkim, 2025-07-09:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm
- Kernel version | `uname -a`
Linux rock64-60 6.12.35-current-rockchip64 #1 SMP PREEMPT Fri Jun 27 10:11:46 UTC 2025 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
ROCK64 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
Any
- SD card used | (EG: SanDisk ultra)
SanDisk 32G

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
headless
- Was the software title installed freshly or updated/migrated?
upgrade from 9.12 to 9.14.2
- Can this issue be replicated on a fresh installation of DietPi?
Yes, it's kernel related (?)

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
Run kernel 6.12. On 6.6.5 leds (red and white) working.
krernel 6.6.5:
 ```
ls -l /sys/class/leds/
 total 0
 lrwxrwxrwx 1 root root 0 Jan 21  2016 led-0 -> ../../devices/platform/leds/leds/led-0
 lrwxrwxrwx 1 root root 0 Jan 21  2016 led-1 -> ../../devices/platform/leds/leds/led-1
```
krernel 6.12.35: directory empty.
No dir /sys/devices/platform/leds/leds/

#### Expected behaviour

during kernel boot red blinking (heartbeat), white led - blinking when read/write on SD card.

#### Actual behaviour
red and white led constantly on.
directory /sys/class/leds/ is empty
No directory /sys/devices/platform/leds/leds/

#### Extra details

dmesg error:
[   20.703474] platform leds: deferred probe pending: leds-gpio: Failed to get GPIO '/leds/led-0'

MichaIng, 2025-07-28:
Many thanks for your reports. I will have a look next dev cycle. Sources to look at:
- Mainline Linux 6.12 device tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3328-rock64.dts?h=linux-6.12.y (the LEDs are defined there)
- Mainline Linux 6.6 device tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3328-rock64.dts?h=linux-6.6.y
- Armbian patches for Linux 6.12: https://github.com/armbian/build/tree/main/patch/kernel/archive/rockchip64-6.12
- Armbian patches for Linux 6.6: https://github.com/armbian/build/tree/main/patch/kernel/archive/rockchip64-6.6

sanlupkim, 2025-09-13:
I've check latest armbian distro (trixie) with edge-kernels. The conclusion is that bug appeared in 6.12 kernel:

- `Linux rock64 6.11.0-edge-rockchip64 #1 SMP PREEMPT Sun Sep 15 14:57:56 UTC 2024 aarch64 GNU/Linux `-  leds **OK**
- `Linux rock64 6.12.11-current-rockchip64 #1 SMP PREEMPT Thu Jan 23 16:23:05 UTC 2025` - leds **NOT** working
- latest edge kernel - `6.16.4-edge-rockchip64` - sitll **NOT** working with message `kernel: platform leds: deferred probe pending: leds-gpio: Failed to get GPIO '/leds/led-0'`
HTH

MichaIng, 2025-09-13:
Since I expect LED nodes at a different path, is this empty as well?
```sh
ls -l /sys/class/leds/
```

sanlupkim, 2025-09-13:
On kernel > 6.11 `/sys/class/leds/` is empty.
`/proc/device-tree/leds/` exist and looks like no difference between 6.11 and 6.12, i.e.: 
```
 cat /proc/device-tree/leds/led-1/linux,default-trigger
 heartbeat
```

MichaIng, 2025-09-13:
So the device tree nodes exist, associated with a driver via `compatible = "gpio-leds";` the same way like on most other SBCs, but the driver fails to set up the sysfs nodes. You pasted a related kernel error:
```
[ 20.703474] platform leds: deferred probe pending: leds-gpio: Failed to get GPIO '/leds/led-0'
```
It menions `led-0` while you showed the device tree node `led-1` only. Does `/proc/device-tree/leds/led-0` exist as well and have the expected `mmc0` default trigger?

sanlupkim, 2025-09-13:
Oh ,sorry led-1 was a example. led-0 also exist and has `mmc0` trigger.
```
# ls -l /proc/device-tree/leds/
total 0
-r--r--r-- 1 root root 10 Sep 13 16:14 compatible
drwxr-xr-x 2 root root  0 Sep 13 16:14 led-0
drwxr-xr-x 2 root root  0 Sep 13 16:14 led-1
-r--r--r-- 1 root root  5 Sep 13 16:14 name

```

MichaIng, 2025-09-13:
Let's see what happens if we disabled `led-0`, since there was no error for `led-1`:
```sh
sudo apt install device-tree-overlay
sudo mkdir -p /boot/overlay-user
cat << '_EOF_' | sudo tee /boot/overlay-user/disable-led-1.dtso > /dev/null
/dts-v1/;
/plugin/;
/ {
	fragment@0 {
		target = <&power_led>;
		__overlay__ {
			status = "disabled";
		};
	};
};
_EOF_
sudo dtc -I dts -O dtb -o /boot/overlay-user/disable-led-1.dtbo /boot/overlay-user/disable-led-1.dtso
G_SUDO G_CONFIG_INJECT 'user_overlays=' 'user_overlays=disable-led-1' /boot/dietpiEnv.txt
```

MichaIng, 2025-09-13:
Does the rk805 node exist, actually?
```sh
ls -l /proc/device-tree/i2c@ff160000/pmic@18
```

sanlupkim, 2025-09-13:
Yes, it is
```
# ls -l /proc/device-tree/i2c@ff160000/pmic@18
total 0
-r--r--r-- 1 root root  4 Sep 13 21:44 '#clock-cells'
-r--r--r-- 1 root root  4 Sep 13 21:44 '#gpio-cells'
-r--r--r-- 1 root root 21 Sep 13 21:44  clock-output-names
-r--r--r-- 1 root root 15 Sep 13 21:44  compatible
-r--r--r-- 1 root root  0 Sep 13 21:44  gpio-controller
-r--r--r-- 1 root root  4 Sep 13 21:44  interrupt-parent
-r--r--r-- 1 root root  8 Sep 13 21:44  interrupts
-r--r--r-- 1 root root  5 Sep 13 21:44  name
-r--r--r-- 1 root root  4 Sep 13 21:44  phandle
-r--r--r-- 1 root root  4 Sep 13 21:44  pinctrl-0
-r--r--r-- 1 root root  8 Sep 13 21:44  pinctrl-names
-r--r--r-- 1 root root  4 Sep 13 21:44  reg
drwxr-xr-x 9 root root  0 Sep 13 21:44  regulators
-r--r--r-- 1 root root  0 Sep 13 21:44  rockchip,system-power-controller
-r--r--r-- 1 root root  4 Sep 13 21:44  vcc1-supply
-r--r--r-- 1 root root  4 Sep 13 21:44  vcc2-supply
-r--r--r-- 1 root root  4 Sep 13 21:44  vcc3-supply
-r--r--r-- 1 root root  4 Sep 13 21:44  vcc4-supply
-r--r--r-- 1 root root  4 Sep 13 21:44  vcc5-supply
-r--r--r-- 1 root root  4 Sep 13 21:44  vcc6-supply
-r--r--r-- 1 root root  0 Sep 13 21:44  wakeup-source

```

sanlupkim, 2026-02-17:
Loading module pinctrl-rk805 solves the problem.
Since 6.12 module pinctrl-rk805 was externalized and can't be loaded automatically. It has to be loaded "manually" (at boot).
See:
https://forum.armbian.com/topic/58035-no-led-device-nodes-since-612-rock64/?do=findComment&comment=233313

torte71, 2026-02-18:
Great! That fixes the cause instead of dealing with the symptoms.
I've applied this patch to an armbian/build checkout for the mksklipad50 board and now the pinctrl-rk805 module gets loaded automatically, without an entry in /etc/modules (checked filesystem and initrd to ensure that).
(I've noticed you've already fixed the missing semicolon in https://github.com/MichaIng/build/commit/f1ab7f1d61171a0d96c677ab71f69546dd2ef723)

Kernel: 6.18.12
Build settings: BOARD=mksklipad50; BRANCH=current; RELEASE=trixie; BUILD_MINIMAL=yes; BUILD_DESKTOP=no; KERNEL_CONFIGURE=no;
Result: Positive, though a) for a different board, but with the same issue and similar setup (rk3328 with status LEDs connected to rk805 gpio bank) and b) for "plain" armbian instead of DietPi.

sanlupkim, 2026-02-18:
I confirm that works
on Dietpi 6.18.12-current-rockchip64 module pinctrl_rk805 is loaded automatically at boot (test on rock64)

closed 2026-02-21 by MichaIng