MichaIng/DietPi #7619: Halt not shutting down Native PC DietPi (no real power drop)

mdr92, 2025-07-13:
### Creating a bug report/issue
- [Y] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.14.2` although has been an issue on multiple versions for a few years
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 6.1.0-37-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)` - Repeated on a number of PC's, see comments
- Power supply used | Multiple including stock and various USB PD
- SD card used | N/A, Various Sata & NVME SSD's

#### Steps to reproduce

1. Boot PC with DietPI
2. Run Halt Command
3. Power use barely drops when halted, drives keep spinning

#### Expected behaviour

- Power used by Mini PC should drop to Standby, approx ~0.5w

#### Actual behaviour

- Carries on using Power, drops from idle power of ~8.5w to ~7.5w, external drives carry on spinning too, but SSH disconnects and can no longer be reached

#### Extra details

- Suspected this for a while as my drives kept spinning and the power LEDs stayed on but finally got a power meter to confirm. 
- Also tested with Libreelec and that correctly shuts down with power LEDS going off and power dropping to ~0.5w
- I've tested on clean installs with no peripherals/drives and the following NUCs/Mini PC's

- Intel NUC6CAYS (J3455)
- Gigabyte BRIX GB-BXBT-1900 (J1900)
- GMKTec G2 (N100)
- Fodenn F10 (N100)

MichaIng, 2025-07-17:
Try `poweroff`. If it does not work, try to enable ACPI functionality, via `systemd-logind`:
```sh
systemctl unmask systemd-logind
apt install dbus
systemct start dbus systemd-logind
```

mdr92, 2025-07-18:
> Try `poweroff`. If it does not work, try to enable ACPI functionality, via `systemd-logind`:
> 
> systemctl unmask systemd-logind
> apt install dbus
> systemct start dbus systemd-logind

Poweroff sorted it, sorry for wasting your time, I feel pretty silly ha but coming from mostly RPi's where Halt would always turn them off I didn't think to use a different command.

closed 2025-07-18 by mdr92