MichaIng/DietPi #7630: APT Update doesn’t work

ILPlais, 2025-07-23:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version: 
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=15
G_DIETPI_VERSION_RC=-1
G_GITBRANCH='dev'
G_GITOWNER='MichaIng'
```
- Distro version: `bullseye 0`
- Kernel version: `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model: `RPi 4 Model B (aarch64)`

#### Additional Information (if applicable)
- Software title `apt`

#### Steps to reproduce
1. `sudo apt update`

#### Expected behaviour
- Update the packets list from the repositories.

#### Actual behaviour
```shell
dietpi@DietPi:~$ sudo apt update
```
```
Réception de :1 https://deb.debian.org/debian-security bullseye-security InRelease [27,2 kB]
Atteint :2 https://archive.raspberrypi.com/debian bullseye InRelease
Atteint :3 https://archive.heckel.io/apt debian InRelease
Atteint :4 https://packages.sury.org/php bullseye InRelease
Atteint :5 https://dietpi.com/apt bullseye InRelease
Atteint :6 https://dietpi.com/apt all InRelease
Err :7 https://ftp.fr.debian.org/debian bullseye InRelease
  Connexion à ftp.fr.debian.org: 443 (212.27.32.66) impossible. - connect (111: Connexion refusée) Impossible d'initialiser la connexion à ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Le réseau n'est pas accessible)
Err :8 https://ftp.fr.debian.org/debian bullseye-updates InRelease
  Impossible d'initialiser la connexion à ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Le réseau n'est pas accessible)
Err :9 https://ftp.fr.debian.org/debian bullseye-backports InRelease
  Impossible d'initialiser la connexion à ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Le réseau n'est pas accessible)
27,2 ko réceptionnés en 2s (11,2 ko/s)
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait
Tous les paquets sont à jour.
W: Impossible de récupérer https://ftp.fr.debian.org/debian/dists/bullseye/InRelease  Connexion à ftp.fr.debian.org: 443 (212.27.32.66) impossible. - connect (111: Connexion refusée) Impossible d'initialiser la connexion à ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Le réseau n'est pas accessible)
W: Impossible de récupérer https://ftp.fr.debian.org/debian/dists/bullseye-updates/InRelease  Impossible d'initialiser la connexion à ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Le réseau n'est pas accessible)
W: Impossible de récupérer https://ftp.fr.debian.org/debian/dists/bullseye-backports/InRelease  Impossible d'initialiser la connexion à ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Le réseau n'est pas accessible)
W: Le téléchargement de quelques fichiers d'index a échoué, ils ont été ignorés, ou les anciens ont été utilisés à la place.
```
In English, the errors are:
```Unable to initialize connection to ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: The network is not accessible)
W: Unable to retrieve https://ftp.fr.debian.org/debian/dists/bullseye/InRelease Connection to ftp.fr.debian.org: 443 (212.27.32.66) impossible. - connect (111: Connection refused) Unable to initialize connection to ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: Network is not accessible)
W: Unable to retrieve https://ftp.fr.debian.org/debian/dists/bullseye-updates/InRelease Unable to initialize connection to ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: The network is not accessible)
W: Unable to retrieve https://ftp.fr.debian.org/debian/dists/bullseye-backports/InRelease Unable to initialize connection to ftp.fr.debian.org: 443 (2a01:e0c:1:1598::2). - connect (101: The network is not accessible)
W: Downloading some index files failed, they were ignored, or the old ones were used instead.
```

MichaIng, 2025-07-23:
There is two issues:
- The `bullseye-backports` suite has been removed from the main Debian repo recently. It can be either replace with the suite from `archive.debian.org` or just removed. For simplicity for now:
  ```sh
  sudo sed -i '/bullseye-backports/d' /etc/apt/sources.list
  ```
  This will be fixed with upcoming DietPi release as well.
- In your case however, the your APT mirror itself cannot be connected to, since it does not support HTTPS, or does not anymore. Did you change the mirror recently? However, I suggest to switch to the global default mirror, which is a mirror director/CDN that pulls from a close mirror automatically, and is available via HTTPS:
  ```sh
  sudo sed -i 's/ftp.fr.debian.org/deb.debian.org/' /etc/apt/sources.list
  ```

There is actually no real reason to use the per-country `ftp.*.debian.org` mirrors anymore.

Btw, do you use the `packages.sury.org` repository intentional? We used that on Debian 9 Stretch, but do not do that anymore and recommend to remove it on recent DietPi, if not needed for something specific. In case you upgraded to Buster and/or Bullseye manually, you might want to redo some migration steps:
```sh
rm -fv /etc/apt/sources.list.d/dietpi-{php,wireguard}.list
rm -fv /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm -fv /etc/apt/preferences.d/dietpi-{php,openssl,xrdp,wireguard,kodi,openhab}
```

ILPlais, 2025-07-23:
Thanks for your help. I deleted the `bullseye-backports` repository from my `/etc/apt/sources.list` file and it works.

I changed the domain name `ftp.fr.debian.org` to `deb.debian.org`, since the first one does not support HTTPS connections.

I deleted the references to the `packages.sury.org` repository. I don't know why I still had it 🤔.
However, the files that were deleted are:
```
/etc/apt/sources.list.d/php.list
/etc/apt/trusted.gpg.d/php.gpg
```

I will leave the issue open until the migration to [%v9.15](https://github.com/MichaIng/DietPi/milestone/151) is complete.

Have a nice day 😉