MichaIng/DietPi #7636: Unable to boot after recent update on RPI4

basav-raj, 2025-07-26:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

After updating using inbuilt dietpi updater, I rebooted my RPI 4 but it never came back online. I was running dietpi on a SSD connected through USB. 

I only see a single green light blink (long) and no activity there after. I check the SSD, it's working fine (i.e can access the data, etc). 

Was something related to boot changed in the latest update?

MichaIng, 2025-07-26:
From which DietPi version did you upgrade? And which kernel version was previously used, i.e. was it upgraded, or did you migrate to the new kernel stack, as offered by the update, in case it was a Bookworm system with old kernel stack?

And how is the SSD powered? If via USB only, please attach it via case/station with dedicated PSU, as RPi USB ports are not designed to reliably power external 2.5" drives: it might work, but it may not work anymore at any time, like after an otherwise perfectly functional kernel upgrade.

Joulinar, 2025-07-26:
> I only see a single green light blink (long)

You are sure in the LED code? At least 1 long flash is not documented on RPI docs.

https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes

basav-raj, 2025-07-26:
I was using this [SSD](https://shop.sandisk.com/products/ssd/portable-ssd/portable-ssd-sandisk-extreme-usb-3-2?sku=SDSSDE61-500G-G25), if it's relevant.

Joulinar, 2025-07-26:
You can try to use a powered USB hub.

MichaIng, 2025-07-26:
> Is there any other way for me to confirm this?

Since HDMI does not output anything, I guess, a UART adapter would be needed to check bootloader output via serial console.

There was no dialogue with
```
[ INFO ] Raspberry Pi kernel/firmware migration available
```
after which, when confirming, you could further choose to install additional kernel builds for other RPi models, right? Otherwise, given that the last kernel update was released on July 2nd, unlikely that it was applied now, which would also be done via `apt upgrade` outside of DietPi updates and triggering a related login banner info. Ah, was a reboot suggested after the update, or did you just do it by yourself? Because it is only suggested by the updater if the kernel was indeed upgraded.

It could be also the EEPROM bootloader, which is managed by the `rpi-eeprom` package and updated automatically on (every) boot, if the `rpi-eeprom` ships and marks a new version to be flashed. Not sure in how far this can have an effect on USB drive power peaks, and their timing/effect on a successful boot procedure. The latest package v28.3-1 was released just 4 days ago, so that would fit from the timing. One significant change of the latest build is this: https://github.com/raspberrypi/rpi-eeprom/commit/dd984a2
Overpower check should not have an effect if no SD card is used, but the skipped probe in that case, leading to up to a second faster boot time. On rotating disks this alone could lead to failing USB boot, if the drives come up too slowly, but with an SSD unlikely?
However, that is in `latest` build, while `default` is still from 2 months ago, so the EEPROM bootloader would not be updated, unless `FIRMWARE_RELEASE_STATUS` in `/etc/default/rpi-eeprom-update` was changed manually. I guess we would see on consecutive reports, if this bootloader change really causes USB boot issues for more users. One could try to add some `boot_delay` to `config.txt` in that case.

The SD card overpower check btw added a new LED signal, missing in the documentation: "1 long, 2 short"

> I was using this [SSD](https://shop.sandisk.com/products/ssd/portable-ssd/portable-ssd-sandisk-extreme-usb-3-2?sku=SDSSDE61-500G-G25), if it's relevant.

I remember this (or a very similar one) was used in another report, where we found out that, despite not being a 2.5" format drive, it indeed draws similar power. So yeah, in this case a powered USB hub sadly makes sense for stability. Generally, I would recommend to not power any block device larger than a pen drive via USB on any Raspberry Pi. _Exception: With 5A PSU, the Raspberry Pi 5 seems to provide max 1.6A per USB port, which should be enough for almost all 2.5" drives, including startup peaks: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#maximum-power-output_

basav-raj, 2025-07-28:
> One could try to add some boot_delay to config.txt in that case.

This worked! I just set `boot_delay=5` and it indeed worked. I believe this should be added to the docs, perhaps as a caveat for those using external drives. 

Thank you for your detailed instructions.

closed 2025-07-28 by basav-raj

MichaIng, 2025-07-30:
So we understand whether the bootloader update really triggered this issue:
- Can you check the current version via `rpi-eeprom-update`?
- And as according to the commit text, boot speedup is just "over a second", does `boot_delay=1` or `boot_delay=2` work as well?

basav-raj, 2025-07-30:
> Can you check the current version via rpi-eeprom-update?

```bash
dietpi@DietPi:~$ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Thu May  8 15:21:35 UTC 2025 (1746717695)
    LATEST: Thu May  8 15:21:35 UTC 2025 (1746717695)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2711/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

> And as according to the commit text, boot speedup is just "over a second", does boot_delay=1 or boot_delay=2 work as well?

unfortunately I won't be able to test it anytime soon as I'm away from the board for a while.