MichaIng/DietPi #7637: Open Beta v9.15 | Please help testing and hardening our upcoming release

MichaIng, 2025-07-26:
RC version | v9.15.0
----------- | -------
v9.14.0 PR | #7606
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2025-06-27**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av9.15
____
### Beta v9.15.0
_(2025-07-19)_

#### New images
- **Orange Pi 3** | Support for the non-LTS version of this board was added.

#### Enhancements
- **NanoPi R5C** | Its onboard Ethernet adapters are now assigned static MAC addresses as well, like done earlier for the other NanoPi mini routers last DietPi release.
- **DietPi-Display** | Kernel command-line config detection has been widened to support more setups: GRUB is supported independent of the architecture, and settings are now changed in all found (and supported) config files, instead of just in the first. Many thanks to @adshrc for pointing out the too limited support for GRUB: https://github.com/MichaIng/DietPi/issues/7390#issuecomment-3012510974
- **DietPi-Software** | **Moonlight (GUI)**: The game streaming client has been unlocked for all ARM and RISC-V systems, excluding only ARMv6 RPi models. Functionality may however depend on whether KMS and V4L2 drivers are available, and how well they work.
- **DietPi-Software** | **Unbound**: In favour of Unbound builtin data, the monthly cron job to download DNS root hints from https://www.internic.net/domain/named.root is removed, together with the `root-hints:` setting in our default config. DNS root servers do by far not change that regularly, and Debian defaults do not make use of dedicated root hints either. This also closes one potential attack vector introduced by the monthly download, which relied on HTTPS authenticity only.
- **DietPi-Software** | **Unbound**: We do now provide our own up-to-date Unbound package, though for now not automatically via our APT server. You can find the packages on our download server, in the respective Debian version sub directory, e.g.: https://dietpi.com/downloads/binaries/bookworm/

#### Bug fixes
- **Debian Bullseye** | Resolved an issue where APT updates failed because the `bullseye-backports` suite is not available on the main Debian repo anymore. It will be pulled from archive.debian.org now. Many thanks to @ILPlais for reporting this issue: https://github.com/MichaIng/DietPi/issues/7630
- **DietPi-Dashboard** | Resolved an issue where the installing the nightly version, enforced on Raspberry Pi 5 and RISC-V systems, failed if no `unzip` was installed. Reason was a resulting `HEAD` request, which is always failing with 404 on the `nightly.link` URL we use to store dashboard nightly builds.
- **DietPi-Software** | **microblog.pub**: Resolved an issue where the installation failed because of outdated dependencies. Since the project is not currently maintained, its Python dependencies are not updated. This is now done with Poetry within our installation code.
- **DietPi-Software** | **Ampache**: Resolved an issue where the install on Debian Bullseye systems always throw an error that the latest version could not be detected, hence the fallback URL was used. Many thanks to @jerin04 for reporting this issue: https://github.com/MichaIng/DietPi/issues/7611
- **DietPi-Software** | **File Browser**: Resolved an issue where the installation failed if the default software password had a length below 12 characters. The minimum password length requirement for File Browser is reduced automatically, to match the default software password if needed. However, we do encourage everyone to assign a long and strong default software password. 12 characters is not an unreasonable minimum. Many thanks to @lutfor-diu for reporting this issue: https://github.com/MichaIng/DietPi/issues/7612
- **DietPi-Software** | **OctoPrint**: Resolved a regression where the installation failed on RISC-V systems because of false `runuser` syntax.

Joulinar, 2025-07-27:
one issue occurred for a user running Apache + NC https://dietpi.com/forum/t/open-beta-v9-15-please-help-testing-and-hardening-our-upcoming-release/23878/3?u=joulinar

As far as I can see, you added a patch for this

MichaIng, 2025-07-27:
A typo, there is a "t" missing in "nexcloud".

MichaIng, 2025-07-27:
Beta v9.15.1 has been merged: #7642

MichaIng, 2025-07-28:
DietPi v9.15 has been released: #7638

Many thanks to all testers! ❤️

closed 2025-07-28 by MichaIng