MichaIng/DietPi #7639: Installation DietPi & Raspberry Pi5 // doesnt work!

RaspClarry, 2025-07-27:
Dear community,
I tried everything to fix this issue, but Im not able to resolve.

Trying to install diepi on my RaspPi5 comes to this: 

#### Details:
- Date           | 2025-07-27 07:03:04
- Program name   | DietPi-Software
- Command        | `apt-get -y dist-upgrade`
- Exit code      | 100
- DietPi version | v9.14.2 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | `Linux DietPi 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux`
- Architecture   | `arm64`
- Hardware model | RPi 5 Model B (aarch64) (ID=5)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following package was automatically installed and is no longer required:
  linux-image-6.12.25+rpt-rpi-2712
Use 'apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up libc-bin (2.36-9+rpt2+deb12u12) ...
Segmentation fault
Segmentation fault
dpkg: error processing package libc-bin (--configure):
 installed libc-bin package post-installation script subprocess returned error exit status 139
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)

THANK YOU!

RaspClarry, 2025-07-27:
[image]

Joulinar, 2025-07-27:
> ```
> Segmentation fault
> ```

Seems to be some kind of data corruption. Maybe you can try a different SD card.

RaspClarry, 2025-07-27:
Thank you :)
I tried two different... I can try a third one.
BTW, is it normal that the sd card lost the original size after flashing? - Sorry Im a greenhorn :)
Or did I made something wrong?

RaspClarry, 2025-07-27:
SDcard had 128GB and after flashing with balena it shows 128MB

Joulinar, 2025-07-27:
ok on Windows, you see the vfat partition only. This is the bootFS. There is a 2nd partition Windows is not able to show. The BootFS is a ext4 partition, which is a linux file system format.

Joulinar, 2025-07-27:
Yes, if that still doesn't work, we need a few system logs. When you see the blue error message, you need to exit the blue error menu and return to the command line. Then I can tell you how to move the logs to the vfat partition so that you can read them under Windows.

RaspClarry, 2025-07-27:
@Joulinar first of all.. THANK YOU!
It seems to be that the third sdcard works...

closed 2025-07-27 by RaspClarry