MichaIng/DietPi #7643: dietpi 9.14.2 for Radxa Zero3W Kernel 6.1.115 built without support for either of the common I2S DAC drivers (`MAX98357a`, `PCM5102A`)

drahnreb, 2025-07-27:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm

- Kernel version | `uname -a`
Linux radxazero3w 6.1.115-vendor-rk35xx #1 SMP Tue Jun  3 13:10:51 UTC 2025 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Radxa ZERO 3 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
5V

- SD card used | (EG: SanDisk ultra)
eMMC

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Want to install [Pimoroni Audio Amp Shield DAC](https://shop.pimoroni.com/products/audio-amp-shim-3w-mono-amp?variant=32341591064659). https://pinout.xyz/pinout/audio_amp_shim

- Was the software title installed freshly or updated/migrated?
Fresh.

- Can this issue be replicated on a fresh installation of DietPi?
Yes.

#### Steps to reproduce

1. Connected Shim via soldering.
2. Compiled custom device tree overlay.
3. Build codec driver `MAX98357a`.
4. Added as `user_overlay` to dietpiEnv.txt
(Details [here](https://gist.github.com/drahnreb/8403466d4884b50ff6d9c1f181741ff7))

6. Not working - initial pop of the speaker but no sound.
#### Expected behaviour
- Sound

#### Actual behaviour

```bash
zcat /proc/config.gz | grep MAX98357
# CONFIG_SND_SOC_MAX98357A is not set
```

we could try with another codec (should be ok, since pimoroni is referencing `hifiberry`), but:
```bash
zcat /proc/config.gz | grep PCM5102A
# CONFIG_SND_SOC_PCM5102A is not set
```

using a dts with `dummy_codec` also does not work.

MichaIng, 2025-07-27:
We can enable the respective drivers, but no guarantee that it will work, at least not without fiddling with the device tree as well (we are currently trying to get a PCM51xxA running on the Orange Pi Zero 3, with mainline kernel), and I can only guess that Armbian does not enable them intentionally.

Generally the config is here: https://github.com/armbian/build/blob/main/config/kernel/linux-rk35xx-vendor.config#L2036
We have a workflow on our repo which builds the kernel from our fork of the Armbain build system:
- https://github.com/MichaIng/DietPi/blob/master/.github/workflows/armbian.yml
- https://github.com/MichaIng/build

I am currently busy with a lot of other things, so would take a while before I can apply those changes and generate test builds myself. I do have a Radxa ZERO 3E, but no such HAT to test it myself either.

drahnreb, 2025-07-27:
@MichaIng thanks for the swift reply. 

Yes, I succeeded compiling the driver. The dtoverlay seems to load fine.

```bash
root@ radxazero3w:/boot# lsmod | grep max98357a
snd_soc_max98357a      16384  1

root@ radxazero3w:/boot# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PimoroniAmpSHIM [Pimoroni-Amp-SHIM], device 0: fe430000.i2s-HiFi HiFi-0 [fe430000.i2s-HiFi HiFi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I could test a test build.

MichaIng, 2025-11-24:
I am thinking about migrating this SBC to mainline Linux, which has the driver included as module. The issues like missing HDMI output seem to be solved, at least works well on my Radxa ZERO 3E.

Just in case you want to test:
```sh
apt install linux-{image,dtb}-current-rockchip64
```