MichaIng/DietPi #7644: Debian Trixie | Testing and migration script

MichaIng, 2025-07-27:
Similar to how we did for Debian Bullseye => Bookworm upgrades, we created a script to upgrade from Debian Bookworm to Trixie: https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-trixie-upgrade

To test it:
```sh
bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/refs/heads/dev/.meta/dietpi-trixie-upgrade')
```

Debian Trixie has planned to be released on August 9th, hence it basically finished, so surprises to be expected from now until release. From DietPi side, whenever testing software installations, they were tested on Trixie as well, but a final test run for all software options is outstanding, and will be done the next days instead.

MichaIng, 2025-07-29:
Thanks for testing.

Right, `<(command)` as well as `< <(command)` and similar do not work with `sudo`, since the implied temporary file is lost before `bash` can use it. A `command | sudo bash` pipe breaks TUI dialogues. Only variant would be `sudo bash -c "$(command)", not sure why we did not suggest that one in our Bookworm release blog post (I just took the command from there). I adjusted it and added `sudo`, should work fine like that.

Regarding SABnzbd:
- A venv seems resonable indeed, given the large number of specific dependencies and versions: https://github.com/sabnzbd/sabnzbd/blob/develop/requirements.txt#L13
  Though I wonder why it tries to uninstall the dist module instead of just overriding it with the site module. But that again, in theory, could break another dist module or package which depends on the exact/not older `jaraco.functools` the Debian package ships.
- But in fact, this is the very first report which a clear dist vs `pip` module conflict I ever faced, while we pretty consequently avoid venv for all software installs and override the `externally-managed` flag present since Debian Bookworm.
- A question here is why `python3-setuptools` is actually installed. We install `pip` `get-pip.py` and recent `setuptools`, and there was only a single case where the Debian package was needed temporary due to a bug in another Debian package. Since PyPI provides pre-compiled modules in nearly all cases, and the Python ecosystem develops fast, IMO the Debian packaged Python modules are mostly irrelevant.
- But withr the "externally managed" things more commonly used, even that I am a fan of efficient resource sharing, I guess we will need to move more and more Python software installs into venv. That this becomes more common across distros, as well as `pipx` etc, is inditectly leading to stricter Python dependencies and more conflicts remaining unnoticed or ignored.

hrfried, 2025-07-30:
I'll have to respond better when I have more time, but yeah it's really a nightmare understanding what python is going to try to do, and distributions have been (rightfully so) moving more and more toward disabling the ability to even mess with python out of a venv. I've had weird issues like python sourcing random forgotten dependencies in `~/.local` left behind from old install scripts at work, and  e.g. inadvertently grabbing an ancient copy of libssl with it, but leaving no indication that it was actually pulling in that dependency without serious `lsof` digging (pyside2 has always been a nightmare in itself honestly)...

I've been incredibly happy with `uv` lately and it's brought my insane workflow of python/pip/pipx/pyenv/pyenv<insertwrapperhere>/poetry crap all into one single tool. Basically create a pyproject.toml (easy to convert from a `requirements.txt`), `uv sync` to bootstrap a venv with the dependencies, then just source it and you're good to go. Upgrading major python version? `uv python list` to find available versions, `uv python pin cpython-<version>` (it'll drop a .python-version similar to pyenv), and then `uv sync` and you're good to go again. Want to upgrade a single dependency? Change a number/comparator in the pyproject.toml. Also creates lockfiles of dependencies by hash upon syncing.

It's insanely fast compared to anything else I've ever used too...It does do a lot of caching, which might need to be handled specially for dietpi since there's the SD card constraint for many users, although I've used UV a bit on my rpis and haven't seen any issues.

I always sound like a zealot when I talk about it, but its really been a gamechanger.

I think there may have been some other issues with a couple services failing after install, but I'll have to come back to this when I have more time.

MichaIng, 2025-07-31:
Maybe I can replicate it on a test system when installing `syncplay` and then run the distro upgrade.

Quite possible that there was some kind of dependency quirk the solver could not solve, without removing the package once. Generally, if anyone faces a similar issue, please copy&paste the related output here. We might be able to work around the issue in the script, if we see the package installed, or even solve it at Debian.

lukaszsobala, 2025-07-31:
My `syncplay-server` package was downloaded from the project's github - it was not the distro package.

MichaIng, 2025-07-31:
Ah I see they provide DEB packages: https://github.com/Syncplay/syncplay/releases
Indeed their `syncplay-server` package is all-in-one, while the (newer, by version string) one from Debian Trixie splits out parts into `syncplay-common`. If I try to install their server package on Trixie, there is hence a file conflict:
```console
dpkg: error processing archive /tmp/syncplay-server_1.7.4.deb (--unpack):
 trying to overwrite '/usr/share/icons/hicolor/256x256/apps/syncplay.png', which is also in package syncplay-common (1.7.4+repack1-1)
```

lukaszsobala, 2025-07-31:
To be honest, debian's packages seem more sensible... Maybe there is a way to force this installation?

Thanks for looking into it, there may be 0 other persons with this issue 😂

hrfried, 2025-08-01:
> `uv` however, as Python module itself, cannot be bootstrapped, can it? What makes it better than `pyenv`? That you do not need `pip` within the `pyenv`?

It's like every single thing I used to do with 5 tools, but in one, with nothing ever conflicting with each other. Oh, and it's like 20 times faster for all those operations. The below took maybe 30-60 seconds to build ab-initio on a raspberry pi zero 2 W. 

There are some caveats with it preferring the standalone python builds, (which you can override and use host/other python), but overall I find myself using their standalone builds whenever possible. And a lot of projects (including sabnzbd, coincidentally) have been working with uv devs on this normally, when there are problems. 

See:
https://github.com/astral-sh/python-build-standalone/issues/548
https://github.com/sabnzbd/sabctools/issues/132
(fixed with `sabctools==8.2.6`—although there might be some kinks to work out still, it built and seemed to run fine just sedding it in the requirements.txt from the latest release tarball).

So overall, the process would be like this:
```
# install uv. Lot of other ways you can install as well
curl -fsSL https://astral.sh/uv/install.sh | sh

# if you have a need for a specific python, you can check out available versions here
# and then `uv python install` it. if you don't choose one, it'll choose automatically when
# you `uv sync`, within the confines of your pyproject.toml. pretty intuitive.

dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ uv python list

# let's install the latest cpython arm build from 3.13 major release, just to better illustrate.
# you're a big pyenv junky like I used to be, so I'm trying to show the kinds of things that were
# gamechangers for me. :)

dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ uv python install cpython-3.13.5-linux-aarch64-gnu

# again, you don't have to pin manually—if you don't,  it'll just use
# the latest within the confines of what's defined in your pyproject.toml
# if you do pin, you'll see a `.python-version` similar to pyenv in your working dir

dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ uv python pin cpython-3.13.5-linux-aarch64-gnu

dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ uv init --bare
Initialized project `sabnzbd-master`

# only because of the workaround necessary above. probably by the time you wanna change things, 8.2.6 will be in the release tarball lol
dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ sed -i 's|sabctools==8.2.5|sabctools==8.2.6|g' requirements.txt

dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ uv add -r requirements.txt
Resolved 58 packages in 16.06s
      Built paho-mqtt==1.6.1
      Built sgmllib3k==1.0.0
Prepared 46 packages in 7.30s
Installed 46 packages in 605ms
 + apprise==1.9.3
 + babelfish==0.6.1
 + blinker==1.9.0
# ...
# removed package list for clarity, but just lists all of them out as you install with nice ansi shell coloration
# ...

# bootstrap and init a .venv from the pyproject.toml/uv.lock generated by your `uv add`
# you can similarly just add/remove packages/versions to the pyproject.toml with uv add/uv remove
dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ uv sync
Resolved 58 packages in 8ms
Audited 46 packages in 0.20ms

# source the venv and run it with the same config set by current dietpi package. Web UI runs and looks fine, but I have all my arrs in proxmox, so I won't really be able to test it thoroughly
dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ source .venv/bin/activate
(sabnzbd-master) dietpi@DietPiZero1:~/sabnzbd-uv/sabnzbd-master$ python3 -OO /etc/sabnzbd/SABnzbd.py -f sabnzbd.ini -n --disable-file-log
...
```

Overall the entire init/build process is maybe a minute or so. Took longer to type this up than to set up, even including figuring out the necessity of using 2.4.6 sabctools. Notably, I did get this warning while running from sabnzbd:
`2025-07-31 22:35:14,344::WARNING::[SABnzbd:429] Could not link to OpenSSL library, please report here: https://github.com/sabnzbd/sabnzbd/issues/2421`
which seems pretty minor, but don't really have time to look into it at the moment. Notably I just grabbed the latest python and sent it, so I didn't really think too much about it, lol.

Note, you can also throw no-cache option in a `uv.toml`, e.g. `~/.config/uv/uv.toml` but haven't messed with it personally yet since I'm normally building stuff on beefy supermicro workstations where I'll cache anything and everything to speed up my workflow :P:
```
$ uv cache --help
Manage uv's cache

Usage: uv-3.13 cache [OPTIONS] <COMMAND>

Commands:
  clean  Clear the cache, removing all entries or those linked to specific packages
  prune  Prune all unreachable objects from the cache
  dir    Show the cache directory

Cache options:
  -n, --no-cache               Avoid reading from or writing to the cache, instead using a temporary
                               directory for the duration of the operation [env: UV_NO_CACHE=]
...
```

Apologies in advance for the absolutely horrible formatting and stream-of-consciousness text. Super busy and just been trying to late-night followup with the remaining energy of the day. :) But I hope this gives a bit of an overview of why I've been really enjoying uv, and why I think it'd potentially make your life a lot easier. Been a long time since I've been excited about python packaging... 

You'd be much quicker than me to implement, since I'm not very familiar with how you organize everything in `dietpi-software`/global functions and such, but hopefully that's at least a good starting place if you wanted to try building with uv. There's potential I could put in a PR for this but it'd probably be a while—super busy and a thousand work/personal projects pulling my attention in a hundred directions. I'd be more than happy to assist if you have any troubles with the uv portion—just don't know if I have the mental bandwidth left to dig into a new set of multi-thousand LOC bash scripts at the moment. 😂

hrfried, 2025-08-01:
Just to provide a version that's a little bit easier to test, these are the basic commands you could run to bootstrap and run sabnzbd with uv, assuming you're in the unwrapped tarball with the relevant `sabnzbd.ini`.

```
curl -fsSL https://astral.sh/uv/install.sh | sh
uv python install cpython-3.13.5-linux-aarch64-gnu
uv python pin cpython-3.13.5-linux-aarch64-gnu
uv init --bare
sed -i 's|sabctools==8.2.5|sabctools==8.2.6|g' requirements.txt
uv add -r requirements.txt
uv sync
source .venv/bin/activate
python3 -OO /etc/sabnzbd/SABnzbd.py -f sabnzbd.ini -n --disable-file-log
```

Adding on another thing I like about uv. It's simple to define dependency groups like local sources—e.g. prebuilt source wheels/sdists/raw source dirs you grabbed and modified—right in your pyproject.toml...which is amazing when you need to patch dependencies and don't want to deal with the nightmare of doing that with...literally every other python tool lol. Same workflow here. `uv init --bare`, `uv sync` `source .venv/bin/activate` and that's it. `uv.lock` keeps a history of package hashes throughout git history. Super nice.

```
[project]
name = "REDACTED"
version = "5.3.1"
description = "REDACTED"
readme = "README.md"
requires-python = ">=3.10"
dependencies = [
    "altgraph==0.17.4",
    "bidict==0.23.1",
    "certifi==2024.2.2",
    "charset-normalizer==3.3.2",
    "fbs",
    "h11==0.14.0",
    "idna==3.6",
    "numpy==1.26.4",
    "packaging==23.2",
    "pillow>=11.2.0",
    "pyinstaller==6.4.0",
    "pyinstaller-hooks-contrib==2024.2",
    "pyside2>=5.15.2.1",
    "python-dotenv==1.0.1",
    "python-engineio==4.11.2",
    "python-socketio==5.12.1",
    "requests==2.32.3",
    "simple-websocket==1.0.0",
    "urllib3==2.2.1",
    "websocket-client>=1.8.0",
    "wsproto==1.2.0",
]

[tool.uv.sources]
fbs = { path = "./external/wheels/fbs-1.2.5-py3-none-any.whl" }
```

MichaIng, 2025-08-01:
So `uv` pulls pre-compiled Python binaries from here: https://github.com/astral-sh/python-build-standalone/releases

Sounds like a great replacement for `pyenv` indeed, to speed up our Home Assistant installs/upgrades. Now I am additionally happy that we split HA data and the `pyenv` instance, including module site consequently. And HA uses `uv` as well for its internal dependency management, when installing integrations via web UI etc. So shouldn't be too hard to replace `/home/homeassistant/.pyenv` (which is purged on any HA reinstall anyway) with a `uv` Python instance, and adjust `/mnt/dietpi_userdata/homeassistant/deps` to point to the `uv` package site.

Same theoretically for `microblog.pub`, though that project seems sadly dead, would be wasted time at this point.

MichaIng, 2025-08-06:
> Are they safe to remove?

Yes, all but the Vim one are expected, and you most likely edited `vimrc` intentionally.

`apt autopurge` really does not purge left config files from removed packages. I'll add this cleanup step to our upgrade scripts.

aszurnasirpal, 2025-08-10:
Any advice on how to proceed? Should I restore the backup, or is there an easy fix?

MichaIng, 2025-08-10:
Just repeat the script, it will now purge the conflicting `docker-compose` package before attempting any package upgrades.

aszurnasirpal, 2025-08-10:
I did, however new error shows up.

```
root@sedruga:/home/dietpi# bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-trixie-upgrade')"
[ INFO ] dietpi-trixie-upgrade | Purge obsolete docker-compose package from Debian repo, since docker-compose-plugin from Docker repo is installed already, and both conflict on Trixie
[ INFO ] dietpi-trixie-upgrade | APT purge docker-compose, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 alsa-utils : Depends: libasound2t64 (>= 1.2.14) but it is not going to be installed
 bluez : Depends: libasound2t64 (>= 1.0.17) but it is not going to be installed
         Depends: libglib2.0-0t64 (>= 2.76.0) but it is not going to be installed
         Depends: libreadline8t64 (>= 7.0~rc1) but it is not going to be installed
 chromium : Depends: libasound2t64 (>= 1.0.17) but it is not going to be installed
            Depends: libatk1.0-0t64 (>= 2.32.0) but it is not going to be installed
            Depends: libcups2t64 (>= 1.7.0) but it is not going to be installed
            Depends: libglib2.0-0t64 (>= 2.40.0) but it is not going to be installed
            Depends: libgtk-3-0t64 but it is not going to be installed or
                     xdg-desktop-portal-backend
 girepository-tools : Depends: libglib2.0-0t64 (= 2.84.3-1) but it is not going to be installed
 gnupg : Depends: gnupg-l10n (= 2.2.40-1.1) but 2.4.7-21 is to be installed
 gvfs : Depends: libglib2.0-0t64 (>= 2.83.0) but it is not going to be installed
 gvfs-backends : Depends: gvfs (= 1.50.3-1) but 1.57.2-2 is to be installed
                 Depends: gvfs-common (= 1.50.3-1) but 1.57.2-2 is to be installed
                 Depends: gvfs-daemons (= 1.50.3-1) but 1.57.2-2 is to be installed
                 Depends: gvfs-libs (= 1.50.3-1) but 1.57.2-2 is to be installed
                 Depends: libimobiledevice6 (>= 1.3.0) but it is not installable
 gvfs-daemons : Depends: libglib2.0-0t64 (>= 2.83.0) but it is not going to be installed
 gvfs-libs : Depends: libglib2.0-0t64 (>= 2.83.0) but it is not going to be installed
 iproute2 : Depends: libtirpc3t64 (>= 1.0.2) but it is not going to be installed
 libatk-bridge2.0-0t64 : Depends: libatk1.0-0t64 (= 2.56.2-1) but it is not going to be installed
                         Depends: libglib2.0-0t64 (>= 2.80.0) but it is not going to be installed
 libatopology2t64 : Depends: libasound2t64 (>= 1.2.7.1) but it is not going to be installed
 libatspi2.0-0t64 : Depends: libglib2.0-0t64 (>= 2.80.0) but it is not going to be installed
 libbabeltrace1 : Depends: libglib2.0-0t64 (>= 2.76.0) but it is not going to be installed
 libblockdev-crypto3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-fs3 : Depends: libext2fs2t64 (>= 1.42.11) but it is not going to be installed
                   Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-loop3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-mdraid3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-nvme3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-part3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-swap3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libblockdev-utils3 : Depends: libglib2.0-0t64 (>= 2.76.0) but it is not going to be installed
 libblockdev3 : Depends: libglib2.0-0t64 (>= 2.42.2) but it is not going to be installed
 libcloudproviders0 : Depends: libglib2.0-0t64 (>= 2.56.0) but it is not going to be installed
 libcurl3t64-gnutls : Depends: libgnutls30t64 (>= 3.8.6) but it is not going to be installed
                      Depends: libnettle8t64 but it is not going to be installed
 libfm-modules : Depends: libfm-extra4t64 (>= 1.2.0) but it is not going to be installed
                 Depends: libfm-gtk3-4t64 (>= 1.3.2-2~) but it is not going to be installed
                 Depends: libfm4t64 (>= 1.3.0) but it is not going to be installed
                 Depends: libglib2.0-0t64 (>= 2.80.0) but it is not going to be installed
                 Depends: libgtk-3-0t64 (>= 3.0.0) but it is not going to be installed
 libgck-2-2 : Depends: libglib2.0-0t64 (>= 2.74.0) but it is not going to be installed
 libgcr-4-4 : Depends: libglib2.0-0t64 (>= 2.74.0) but it is not going to be installed
 libgio-2.0-dev : Depends: libglib2.0-0t64 (= 2.84.3-1) but it is not going to be installed
 libgio-2.0-dev-bin : Depends: libglib2.0-0t64 (= 2.84.3-1) but it is not going to be installed
 libgirepository-2.0-0 : Depends: libglib2.0-0t64 (= 2.84.3-1) but it is not going to be installed
 libglib2.0-bin : Depends: libglib2.0-0t64 (= 2.84.3-1) but it is not going to be installed
 libgoa-1.0-0b : Depends: libglib2.0-0t64 (>= 2.83.0) but it is not going to be installed
 libgstreamer1.0-0 : Depends: libglib2.0-0t64 (>= 2.81.1) but it is not going to be installed
 libmenu-cache-bin : Depends: libfm-extra4t64 (>= 1.2.0) but it is not going to be installed
                     Depends: libglib2.0-0t64 (>= 2.76.0) but it is not going to be installed
 libmsgraph-1-1 : Depends: libglib2.0-0t64 (>= 2.80.0) but it is not going to be installed
[... cut ...]
 - Command: apt-get -y --allow-change-held-packages autopurge docker-compose
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
```
- Reference code: ce545ef0-1b89-4461-b92e-b3818a73fb25

MichaIng, 2025-08-10:
Run `apt -f install`, then repeat.

StephanStS, 2025-08-10:
Maybe, also a `dpkg --remove docker-compose` helps (so I think I did it).

aszurnasirpal, 2025-08-10:
ok, I finished despite this error, and rebooted.
Most of my services were up and running ok. 
Some noticed problems:
Mylar3
```
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE] Some modules are missing and may need to be installed via pip before proceeding. PySocks is required only if using proxies.
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE] Reinstall each of the listed module(s) below or reinstall the included requirements.txt file
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE] The following modules are missing:
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  APScheduler >=3.6.3
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  beautifulsoup4 >=4.8.2
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  cfscrape >=2.0.8
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  cheroot >=8.2.1
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  CherryPy >=18.5.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  feedparser >=5.2.1
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  Pillow >=10.1
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  portend >=2.6
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  pycryptodome >=3.19.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  pystun >=0.1.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  PySocks >=2.22.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  simplejson >=3.17.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  six >=1.13.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  tenacity >=8.2.3
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  tzlocal >=2.0.0
Aug 10 17:13:20 sedruga python3[1201]: [MODULES UNAVAILABLE]  user_agent2 >=2021.12.11
Aug 10 17:13:21 sedruga systemd[1]: mylar.service: Deactivated successfully.
Aug 10 17:46:20 sedruga python3[36482]:   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
Aug 10 17:46:20 sedruga python3[36482]:   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
Aug 10 17:46:20 sedruga python3[36482]:   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
Aug 10 17:46:20 sedruga python3[36482]:   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
Aug 10 17:46:20 sedruga python3[36482]:   File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
Aug 10 17:46:20 sedruga python3[36482]:   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
Aug 10 17:46:20 sedruga python3[36482]:   File "/home/dietpi/.local/lib/python3.13/site-packages/cfscrape/__init__.py", line 19, in <module>
Aug 10 17:46:20 sedruga python3[36482]:     from urllib3.util.ssl_ import create_urllib3_context, DEFAULT_CIPHERS
Aug 10 17:46:20 sedruga python3[36482]: ImportError: cannot import name 'DEFAULT_CIPHERS' from 'urllib3.util.ssl_' (/home/dietpi/.local/lib/python3.13/site-packages/urllib3/util/ssl_.py)
Aug 10 17:46:20 sedruga systemd[1]: mylar.service: Control process exited, code=exited, status=1/FAILURE
```
Another one that is down is sabnzbd, also it seems that It has problem with python package

MichaIng, 2025-08-10:
You most likely have the package manager `pip` installed. Try this, but check that no end user software you need is uninstalled:
```sh
apt autopurge python3-pip python3-setuptools python3-distutils
```
And if you aborted the `dietpi-software` run, I recommend to rerun it to get Python 3 with `pip` etc, so you can install the modules for Mylar3:
```sh
G_PROMPT_BACKUP_DISABLED=1 /boot/dietpi/dietpi-software reinstall 38 40 89 114 118 125 130 136 139 141 143 153 155 180 181 210
# Remove left config files from removed packages
mapfile -t packages < <(dpkg -l | mawk '$1=="rc"{print $2}')
G_AGP "${packages[@]}"
```

MichaIng, 2025-08-10:
<s>Okay looks like you previously installed all Python modules via APT instead of `pip`. Older `pip` still handled it gracefully, ignoring and overriding the system packages, but recent `pip` fails in this case. Let's check it a bit differently, so we can more selectively uninstall some.</s>

Okay, you found it already.

> Ok, after removing and installing again the python3-jaraco package

At best install it via `pip` to avoid further conflicts:
```sh
apt autopurge python3-jaraco
pip3 install jaraco
```
Funnily it is about the very same Python package @hrfried faced the same issue with. I get it, we need to move our Python software options into `venv`'s consequently. It does not fully solve these issues, since `pip` itself, `setuptools` and `wheel` are needed on the system, and we need to pull those via `get-pip.py`, as APT ships too old versions on older Debian. But it is about these 3 only then. Maybe I find a way to convince `get-pip.py` or `pip` to just ignore those 3 system packages and naturally override them via `/usr/local` vs `/usr` bin and site locations, like it used to do with earlier versions. That it complains about not being able to uninstall the system package instance (which it should not even try to), is a somewhat new behaviour.

MichaIng, 2025-08-10:
Glad everything works again. Yeah the Python upgrade is somewhat a critical point. Everything single module needs to be reinstalled, in cases re-compiled, `pip` vs `apt` conflicts can theoretically appear, and now additionally that `pip` attempts to uninstall the `apt`-installed modules.

I am still not aware of a case where `pip`-installed module versions conflict with `apt`-installed ones, like the a module that was installed only via `apt` and pulled in a dependency via `apt` would fail to work, due to a different version of the same dependency installed via `pip`. This scenario is the reason why distros can now place a flag which instructs `pip` to not install any module on system level. We override this with a setting since 3-4 years, and I literally cannot remember any report of such conflict. But since this has become common across distros, hence the use of `venv` or other sandboxed environment for every Python project, I guess the previously graceful handling of `pip` with `apt`-installed modules, not trying to uninstall them, was removed at some point, as it might not have looked like a necessary extra code anymore. So now we are running into new issues, which are the result of preventing potential conflicts, which we never saw happening in practice 😅. But well, forces us to finally go with the flow. If it would just not cause so much extra work, while I have enough to do with true issues.

FPSUsername, 2025-08-11:
This is what's shown by running the commands
```
ii  base-files                     25.5.2-12.4+deb12u11-bookworm  arm64        Armbian base system miscellaneous files
```

``` 
base-files:
  Installed: 25.5.2-12.4+deb12u11-bookworm
  Candidate: 25.5.2-12.4+deb12u11-bookworm
  Version table:
 *** 25.5.2-12.4+deb12u11-bookworm 100
        100 /var/lib/dpkg/status
     24.8.3-13.5-trixie 500
        500 http://apt.armbian.com trixie/trixie-utils arm64 Packages
     24.8.2-13.5-trixie 500
        500 http://apt.armbian.com trixie/trixie-utils arm64 Packages
     24.8.1-13.5-trixie 500
        500 http://apt.armbian.com trixie/trixie-utils arm64 Packages
     24.5.1-13.2-trixie 500
        500 http://apt.armbian.com trixie/trixie-utils arm64 Packages
     24.2.1-13-trixie 500
        500 http://apt.armbian.com trixie/trixie-utils arm64 Packages
     13.8 500
        500 http://ftp.de.debian.org/debian trixie/main arm64 Packages
Warning: Could not execute pager - PagerSetup (2: No such file or directory)
```

``` 
deb http://ftp.de.debian.org/debian/ trixie main contrib non-free-firmware
deb http://ftp.de.debian.org/debian/ trixie-updates main contrib non-free
deb https://deb.debian.org/debian-security/ trixie-security main contrib non-free
deb http://ftp.de.debian.org/debian/ trixie-backports main contrib non-free'
```

MichaIng, 2025-08-11:
Hmm, DietPi v8.22 had a pre-patch to rename `armbian.list` to `dietpi-armbian.list`, and `dietpi-installer` calls on unsupported Armbian images (which hence left the Armbian repo in place) rename it as well from that point on. So I wonder why there is still a `armbian.list`. Looks like we need to repeat the patch which removes the repo and downgrades `base-files`, as well if `dietpi-armbian.list` does not exist, but `armbian.list` instead.

```sh
[[ -f '/etc/apt/sources.list.d/armbian.list' ]] && G_EXEC mv /etc/apt/sources.list.d/{,dietpi-}armbian.list

if [[ -f '/etc/apt/sources.list.d/dietpi-armbian.list' ]]
then
	cat << '_EOF_' > /etc/apt/preferences.d/dietpi-armbian || exit 1
Package: *
Pin: origin apt.armbian.com
Pin-Priority: -1

Package: armbian-firmware* linux-*
Pin: origin apt.armbian.com
Pin-Priority: 500
_EOF_
fi

if dpkg --compare-versions "$(dpkg-query -Wf '${Version}' base-files)" gt 20
then
	G_DIETPI-NOTIFY 2 'Enforcing downgrade of Armbian'\''s base-files package to Debian'\''s'
	cat << '_EOF_' > /etc/apt/preferences.d/dietpi-armbian-tmp
Package: base-files
Pin: release o=Debian
Pin-Priority: 1000
_EOF_
	G_EXEC eval 'echo '\''APT::Get::Allow-Downgrades "1";'\'' > /etc/apt/apt.conf.d/dietpi-armbian'
	G_AGUG
	G_EXEC rm /etc/apt/preferences.d/dietpi-armbian-tmp
	G_EXEC rm /etc/apt/apt.conf.d/dietpi-armbian
fi

if [[ -f '/etc/apt/sources.list.d/dietpi-armbian.list' && -f '/etc/apt/sources.list.d/dietpi.list' ]] && grep -q 'dietpi.com/apt all' /etc/apt/sources.list.d/dietpi.list
then
	G_DIETPI-NOTIFY 2 'Removing obsolete Armbian APT repository'
	G_EXEC rm /etc/apt/sources.list.d/dietpi-armbian.list
	[[ -f '/etc/apt/preferences.d/dietpi-armbian' ]] && G_EXEC rm /etc/apt/preferences.d/dietpi-armbian
	[[ -f '/etc/apt/trusted.gpg.d/dietpi-armbian.gpg' ]] && G_EXEC rm /etc/apt/trusted.gpg.d/dietpi-armbian.gpg
fi
```

FPSUsername, 2025-08-11:
Thank you for the very quick work!
```
OS: Debian GNU/Linux 13 (trixie) aarch64

> cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=15
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

> cat /etc/debian_version
13.0
```

MichaIng, 2025-08-11:
So, apart of the fact that some patch was not applied on your system a longer time ago, the problem is this:
- Armbian ships their own `base-files` package, to ship their own `/etc/os-release` (and related) with `PRETTY_NAME="Armbian 24.8.3 trixie"` as example.
- Their package has a version string which matches the Armbian version, not one which matches the Debian version. So e.g. above package is v24.8.3, while Debian Trixie's is v13.8. Hence APT never upgrades to the Debian version of the package, which is of course also intended by Armbian.
- However, Armbian does not assure that the package version on newer distro versions is above that one of older distro version, something which is implied by Debian's package naming scheme, but not by Armbian. So at time of writing, `base-files` on Armbian's Trixie suite was updated last year, hence still has this v24.x.y version string. The one from the Bookworm suite however was updated this year, hence has a v25.5.2 version string.
- So doing a distro upgrade on Armbian from Bookworm to Trixie implicitly results in an outdated `base-files` package, with `/etc/os-release` and `/etc/debian_version` showing false versions.
- I remember the exact same happened when Bookworm was released, or maybe it was Bullseye, and I reported this. Looks like they did not learn 🤷‍♂️.

MichaIng, 2025-08-11:
I updated the distro testing page on our wiki with all tests finished in CI: https://github.com/MichaIng/DietPi/wiki/Debian-Trixie-testing

There is a small number of software options which cannot be installed non-interactively, and I need to go through them, either enable unattended installs for them, hack it inside our testing script, or test it locally.

Also the results include a bunch of fixes done in DietPi v9.16 `dev` branch. I'll maybe release it earlier than planned. The upgrade script warns in case any yet incompatible software option is currently installed, based on the test results and fixes so far, and allows you to abort the upgrade in case.

afritsch, 2025-08-11:
The dist-upgrade with the script went fine for my setup (opi3b), just now "who" and "w" do not show any active sessions. I did some googling and only found hints regarding a file named /run/utmp, but I have no idea.

Another thing I found is that apparently dpkg overwrote my zram config by renaming "/etc/systemd/zram-generator.conf" (which can be undone quickly).

MichaIng, 2025-08-11:
I didn't even know the commands `who` and `w` 😅, used `last`, `lastb`, and `lastlog` to query those logs. However, they are gone due to year 2038 problem: https://wiki.debian.org/pam_lastlog2%20and%20wtmpdb

MichaIng, 2025-08-11:
> Reboot not required, just re-login

Yes I recognized and edited it above. If dbus and logind were not running already, they just need to be stared and it works right away 👍.

> `dbus-broker`

Lennart told me about the new DBus broker already, and his reaction seemed like he was wondering that Debian is still distributing the "legacy" `dbus-daemon` as default implementation. And this has not changed with Trixie either. I never had a closer look into it, just recognized that the new one starts 2 processes, and consumes a little more memory overall. In how far is the broker better than the old daemon? logind is one of the few dependents of DBus I know at all, and it just works with the old daemon 😅.

MichaIng, 2025-08-11:
The "old" one is however not abandoned or so, but it is actively developed. So some advertisement buzz words like "new" and "high performance" won't convince me, as long as everything does work well with the "old" one, and all I do see with the new one is one more process and higher memory usage 😅. I guess Debian maintainers are thinking the same, and hence did not change their default. Guess we need to wait for someone going into technical depth, to explain what really makes it practically better.

DonPablo1010, 2025-08-11:
I was wondering whether the traditional .list files were deliberately left out from being migrated to the new APT deb822 .sources format.

Wouldn’t it be feasible to include the conversion from .list to .sources files directly within the upgrade script?

I came across this when following the Proxmox-documented upgrade process for Debian 13 (Trixie) and noticed the discrepancy. Integrating the conversion step during the upgrade could streamline the process and reduce potential user confusion or manual adjustments afterward.

Was this omission intentional, or could adding such a step into the upgrade script be considered?

Thanks for your work and for any insights!

MichaIng, 2025-08-11:
> I was wondering whether the traditional .list files were deliberately left out from being migrated to the new APT deb822 .sources format.

Until some weeks ago, any `apt` call was warning and offering an automatic migration of the old `.list` files to new `.sources` ones, but also broke individual cases when doing so. But at some point it stopped the warnings and offer. And I was quite happy, since implementing this would cost too much time to do all other Trixie related changes and fixes. It is on the ToDo, but no particular reason to rush.

There are no functional differences between old and new format. The new one is certainly better to read and understand for people new to Debian.

numanair, 2025-08-14:
Just wanted to report a success with the upgrade.

hecatae, 2025-08-15:
updated a physical Oneprovider dedicated server Intel C2350, 4GB RAM 1TB HDD without issue using the script.

MichaIng, 2025-08-15:
Thanks for testing and your feedback guys! I'm implementing further fixes and enhancements for software install options on Trixie, some of which are relevant for earlier Debian versions as well.

Also the upgrade scripts keeps getting better in dealing with certain possible issues one might run into. Not all but some of the Python 3 package related issues are fixed this way: https://github.com/MichaIng/DietPi/commit/61779ed
Though we are sort or careful, not actively purging the APT packages, just in case anything needed really depends on them, DEB package-wise. And found this `mime-support` dummy package remaining, even if not needed by anything, which keeps to other unnecessary packages as well in case. Safe to purge on Trixie, where it does not exist anymore in the repo, hence cannot have any depedency:
```sh
apt autopurge mime-support
```

I will push a DietPi v9.16 beta tomorrow, with release on next Saturday, following the aimed schedule. With this release, we will also link Trixie-based images on our website and publish a blog post with some general info about Debian Trixie and release cycle, promoting our upgrade script.

famewolf, 2025-08-16:
The script fails processing the package "preload".

famewolf, 2025-08-16:
> Can you paste the complete error message please?
>
> For reference: https://packages.debian.org/trixie/preload

Errors were encountered while processing:
preload
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] dietpi-trixie-upgrade | APT dist-upgrade
- Command: apt-get -y dist-upgrade
[FAILED] dietpi-trixie-upgrade | Unable to continue, dietpi-trixie-upgrade
will now terminate.

famewolf, 2025-08-16:
> Are there other messages between
>
> Setting up preload (x.y.z) ...
>
> and
>
> Errors were encountered while processing: preload
>
> ?
>
> The postinst script of this package fails, but your paste starts after
> potential output of this script.

There is no "setting up preload message.  To clarify I've pasted a bit
above the error.

etting up python3-docutils (0.21.2+dfsg-2) ...
Processing triggers for ca-certificates (20250419) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Processing triggers for initramfs-tools (0.148.3) ...
update-initramfs: Generating /boot/initrd.img-6.12.41+deb13-amd64
Processing triggers for shim-signed:amd64 (1.47+15.8-1) ...
Errors were encountered while processing:
preload
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] dietpi-trixie-upgrade | APT dist-upgrade
- Command: apt-get -y dist-upgrade
[FAILED] dietpi-trixie-upgrade | Unable to continue, dietpi-trixie-upgrade
will now terminate.

MichaIng, 2025-08-16:
I checked the `postinst` script already, please see my edited message above.

famewolf, 2025-08-16:
Aug 16 14:48:30 DietPi systemd[1]: Starting preload.service - LSB: Adaptive readahead daemon...
Aug 16 14:48:30 DietPi preload[10069]: Starting Adaptive readahead daemon: preload.
Aug 16 14:48:30 DietPi systemd[1]: Started preload.service - LSB: Adaptive readahead daemon.

It prompted me to reboot after the update so if you have a way to see previous messages I can run that too....

MichaIng, 2025-08-16:
So repeating the script or step worked? At least now the `preload` service is running. I will run an update on a test system with `preload` package installed.

famewolf, 2025-08-16:
Also the script uninstalled bpftune without prompting me and if I had not noticed during review I would not have known it was removed.  I don't recall mention of any of the user installed packages being removed.

[ INFO ] dietpi-trixie-upgrade | APT purge **bpftune** cockpit-ws dbus-user-session fontconfig fontconfig-config fonts-dejavu-core libblockdev2:amd64 libfakeroot:amd64 libgdk-pixbuf-2.0-0:amd64 libglib2.0-0:amd64 libmagic1:amd64 libparted-fs-resize0:amd64 libpwquality-common libvdpau1:amd64 linux-image-6.1.0-35-amd64 mdadm network-manager packagekit policykit-1 **preload** shared-mime-info udisks2 wget, please wait...
Reading package lists...
Building dependency tree...
Reading state information...

MichaIng, 2025-08-16:
On test VM, upgrading with `preload` worked well:
```
Setting up preload (0.6.4-5+b2) ...
... ~50 lines in between
Processing triggers for ca-certificates (20250419) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
[  OK  ] dietpi-trixie-upgrade | APT dist-upgrade
```
Might depend on the amount and exact preloaded binaries, with the fact that basically everything changes on disk in a relatively short time. My test system was otherwise mostly empty. If someone else is running into the same issue, and service logs from before the reboot do not give a better hint, we can just stop and disable that service before the upgrade, and enable it again afterwards, but not start it, so it starts back up after reboot instead.
______________
Regarding `bpftune`, since this is available in Debian repos only [since Trixie](https://packages.debian.org/trixie/bpftune), where did you install it from?

Based on the packages which are actively "purge"d, not "autopurge"d, these must have been removed earlier already, so that only config files were left. I guess the `bpftune` package you installed from elsewhere had some conflict or missing dependency with the packages available on Trixie, hence was removed automatically as part of `apt-get dist-upgrade`.

Some of the packages are a bit surprising to be missing now indeed, like `network-manager`, `mdadm`, and `udisks2`. Also `dbus-user-session`. Possibly some core package of a desktop environment pulled them as hard dependency earlier, and does not require them anymore. But while they would have been flagged for autoremoval then, there is no reason why they would have been removed during `apt-get dist-upgrade`. Maybe you removed them manually before, but with `apt remove` instead of `apt purge`, so those config files were left from before the distro upgrade? At least I'll add an info message that this step purges only packages which were removed before already.

famewolf, 2025-08-16:
Regarding bpftune I don't recall.  If it's not available in bookworm I may
have installed the trixie package or built the .deb myself.  At one point I
was compiling it myself from oracle's github.   In any case I appreciate
the script upgrading my mini pc to trixie.

MichaIng, 2025-08-16:
The Trixie package depends on `libbpftune0` also available on Trixie only, which depends on `libc6` >= 2.38, while Bookworm ships `libc6` 2.36. Probably you compiled it from Debian sources then. I am wondering which dependency conflict caused its removal then, I see one only on 32-bit due to time64 transition.

In any case, especially when compiling or downloading DEB packages from other sources, this shows that keeping an eye on removed packages makes sense, just to assure nothing expected is missing, especially if it is no foreground UI application which would be obvious quickly. This includes 3rd party DEB packages installed from `dietpi-software` as well, but changes are good that we caught all potential issues with them already.

>  In any case I appreciate the script upgrading my mini pc to trixie.

And thanks for reporting the issues you faced 👍.

MichaIng, 2025-08-20:
@chrlns
Where is your 1 TiB disk mounted? Mount points and directories below `/mnt` are excluded by default, with the only exception of `/mnt/dietpi_userdata`, if that one is on the same SD card/drive of the OS itself. Otherwise, you'd need to exclude it with the filter option, adding
```
- /path/to/mountpoint
```
to the file.

> Error: pg_controldata not found, please install postgresql-15

We try to migrate PostgreSQL 15 clusters to v17 automatically. This requires PostgreSQL 15 as well as PostgreSQL 17 to be installed concurrently. During package upgrade, the maintainer scripts (of the package, not by us) create an APT preference which prevents the removal of PostgreSQL 15, at least it should do that, did so on various tests I did. So old clusters can still run until the migration is done, and that one can work etc. And PostgreSQL 15 can then be removed manually, after removing the APT block. In your case however, it looks like PostgreSQL 15 was uninstalled, despite the old cluster being there. So the maintainer scripts did not do what they are intended to do. Can you show the output of this:
```sh
cat /etc/apt/apt.conf.d/02autoremove-postgresql
dpkg -l | grep postgres
```
And do you still have the output of the whole `dietpi-trixie-upgrade` run, so we can see where PostgreSQL packages are upgraded and in case removed?

MichaIng, 2025-08-20:
@Joulinar 
That was not a DietPi user it seems, or at least he did not use our script, since we handle that automatically:
```sh
if apt-mark showmanual | grep -q '^libicu72$'
then
	G_DIETPI-NOTIFY 2 'Migrating from libicu72 to libicu76'
	G_AGI libicu76
	G_EXEC apt-mark auto libicu72
fi
```
Interesting that in his case, `libicu72` was removed. There are not conflicts between them, and it usually just remains on the system, even if `libicu76` is installed. Probably he just misinterpreted things as `libicu72` can probably not be loaded on Trixie due to incompatibilities not reflected in the package metadata. So the relevant step is to install `libicu76` indeed, while the `libicu72` purge is just a cleanup step.

MichaIng, 2025-08-21:
> The disk was mounted at /bpool.

Then it makes sense. For `dietpi-backup` you would then need to exclude that directory explicitly.

MichaIng, 2025-08-25:
Let's see. Um an issue is that none of the desktop related packages is on the RPi repo Trixie suite yet. Let's see how far we can get when only replacing the `gtk+3.0` source based packages:
```sh
dpkg -P --force-depends gir1.2-gtk-3.0 libgtk-3-0 libgtk-3-common
apt -f install
```

TheRaven500, 2025-08-25:
Hm, looks ok so far. And now? Restart the update script?

MichaIng, 2025-08-25:
> Uhhh, looks like "apt-get install dbus-x11" did the trick. :-)

Is `dbus-user-session` installed? That is the modern replacement for `dbus-x11` and `dbus-launch`.

For autostart as non-root user, whether with login prompt or not, LightDM is used, not `startx`. LightDM is then configured to do the autologin as non-root user. Maybe try with manual login first, to see whether LightDM shows up:
```sh
dietpi-autostart 16
```
Else check its logs:
```sh
journalctl -u lightdm
```

TheRaven500, 2025-08-25:
dbus-user-session -> is installed 👍 
dietpi-autostart 16 -> does not work 👎 No GUI, just CLI prompt
journalctl -u lightdm -> Looks like lightdm does not start. See attached error message.

[image]

MichaIng, 2025-08-25:
Is LightDM on latest version?
```sh
dpkg -l lightdm
```

The warning is unrelated. It can be muted by installing the `accountsservice` package, which is however a "suggestion" only, and was so on Bookworm as well.

> Edit: Ah, maybe a "wayland/xorg thing"?!

LightDM is X11 only, and we install X11 with desktop environments. So unless you installed Wayland manually and configured the desktop to work with it, that would not cause issues. Also Xfce AFAIK is X11 only.

TheRaven500, 2025-08-26:
dpkg -l lightdm

> ii  lightdm        1.32.0-6+rpt2 arm64        simple display manager

I vaguely remember that it was somehow possible to switch between x11 and Wayland. Unfortunately, I don't remember how. Maybe it is set to wayland but i need X11.

caio1007, 2025-08-27:
Hello, can using the script for 9.16 now be considered stable? My system is (Native PC x86_64)

TheRaven500, 2025-08-27:
> Where you facing the same issue on Bookworm, that it took 90 seconds for the LightDM login prompt to appear, or did you take respective action like masking the DRI device units?

Yes i had this problem on Bookworm on a Pi4 and a Pi400. It came with an update, but i can not say when.
I masked the two services and it booted fast like before the update.

> Maybe not the same issue, but to test:
> 
> sudo wget https://raw.githubusercontent.com/RPi-Distro/raspberrypi-sys-mods/refs/heads/bookworm/lib/udev/rules.d/61-drm.rules -O /etc/udev/rules.d/61-drm.rules
> sudo udevadm trigger /dev/dri/card0 /dev/dri/renderD128

Second command does not work, i get this error:
`Failed to open the device '/dev/dri/card0': No such device`

I also tried to unmask "dev-dri-renderD128.device" and "dev-dri-card0.device".
Unfortunately without success. :worried:

MichaIng, 2025-08-27:
Interested whether the added Wants/After units is the only difference between the LightDM package from RPi repo and Debian, but sadly, as often, their GitHub repos do not reflect what they ship with their APT repo: https://github.com/raspberrypi-ui/lightdm/blob/master/debian/lightdm.service

> `Failed to open the device '/dev/dri/card0': No such device`

Then you do not have KMS/DRM enabled:
```sh
/boot/dietpi/func/diepti-set_hardware rpi-opengl vc4-kms-v3d
```

Without it, I can indeed replicate the exact issue. And this also with Debian's LightDM. Looks like an upstream change, that LightDM works only with KMS/DRM backend now, even that the X server can still start with framebuffer devices. `/usr/sbin/lightdm` is a binary, so we cannot see how it invokes the X server exactly. Also with debug flag, it does not throw any error:
```console
root@DietPi:~# /usr/sbin/lightdm -d
[+0.00s] DEBUG: Logging to /var/log/lightdm/lightdm.log
[+0.00s] DEBUG: Starting Light Display Manager 1.32.0, UID=0 PID=948
[+0.00s] DEBUG: Loading configuration dirs from /usr/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /usr/share/lightdm/lightdm.conf.d/01_debian.conf
[+0.00s] DEBUG: Loading configuration dirs from /usr/local/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration dirs from /etc/xdg/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /etc/lightdm/lightdm.conf
[+0.00s] DEBUG: Registered seat module local
[+0.00s] DEBUG: Registered seat module xremote
[+0.00s] DEBUG: Using D-Bus name org.freedesktop.DisplayManager
[+0.00s] DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
[+0.00s] DEBUG: Monitoring logind for seats
[+0.00s] DEBUG: Acquired bus name org.freedesktop.DisplayManager
[+0.00s] DEBUG: Could not run plymouth --ping: Failed to execute child process ?plymouth? (No such file or directory)
[+0.01s] WARNING: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
[+0.01s] DEBUG: Loading user config from /etc/lightdm/users.conf
[+0.01s] DEBUG: User dietpi added
```
The `org.freedesktop.Accounts` warning is unrelated and can be ignored. Looks like when using LightDM, we need to forcefully enable KMS/DRM on Raspberry Pi.

TheRaven500, 2025-08-27:
Ok, both set and the display works. But still only CLI login.

MichaIng, 2025-08-27:
A more generic way to enable DSI displays:
```sh
G_CONFIG_INJECT 'display_auto_detect=' 'display_auto_detect=1' /boot/config.txt
```
I.e. an alternative to `dtoverlay=vc4-kms-dsi-7inch`.

> Wait, lightdm service did not start!
> Lot of errors:

LightDM really has horrible error output, i.e. no usable error output at all 😄.

> I try to have a closer look to morrow. Oh, i mean today... 00:06 😁

Same here. Have a good night. To get some more output tomorrow, run this at best from an SSH session:
```sh
sudo lightdm -d
```

danmoo117, 2025-08-27:
@MichaIng I've updated one of my rpi4s tonight and after the running of your upgrade script I now get the following error when trying to update Apt.

```shell
Hit:1 http://raspbian.raspberrypi.org/raspbian trixie InRelease
Hit:2 https://archive.raspberrypi.com/debian trixie InRelease
Hit:3 https://downloads.plex.tv/repo/deb public InRelease
Err:2 https://archive.raspberrypi.com/debian trixie InRelease
  Sub-process /usr/bin/sqv returned an error code (1), error message is: Missing key CF8A1AF502A2AA2D763BAE7E82B129927FA3303E, which is needed to verify signature.
Hit:4 https://dietpi.com/apt trixie InRelease
Hit:5 https://dietpi.com/apt all InRelease
Get:6 https://pkgs.tailscale.com/stable/debian trixie InRelease
Fetched 6579 B in 1s (11.3 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. OpenPGP signature verification failed: https://archive.raspberrypi.com/debian trixie InRelease: Sub-process /usr/bin/sqv returned an error code (1), error message is: Missing key CF8A1AF502A2AA2D763BAE7E82B129927FA3303E, which is needed to verify signature.
W: https://downloads.plex.tv/repo/deb/dists/public/InRelease: Policy will reject signature within a year, see --audit for details
E: Failed to fetch https://archive.raspberrypi.com/debian/dists/trixie/InRelease  Sub-process /usr/bin/sqv returned an error code (1), error message is: Missing key CF8A1AF502A2AA2D763BAE7E82B129927FA3303E, which is needed to verify signature.
E: Some index files failed to download. They have been ignored, or old ones used instead.
``` 
ChatGPT suggested the following but I thought I'd check in with you first.

Download and install the missing key
```shell
curl -fsSL https://archive.raspberrypi.com/debian/raspberrypi.gpg.key \
  | sudo gpg --dearmor -o /usr/share/keyrings/raspberrypi-archive-keyring.gpg
```
Open the repo entry
```shell
sudo nano /etc/apt/sources.list.d/raspi.list
```
Update it with this
```shell
deb [signed-by=/usr/share/keyrings/raspberrypi-archive-keyring.gpg] https://archive.raspberrypi.com/debian trixie main
```

Is this what you'd suggest?

MichaIng, 2025-08-27:
First of all the LightDM issue: https://github.com/MichaIng/DietPi/commit/62388aa
But we will probably just remove support for legacy graphics, and enable KMS/DRM OOTB, or at least as fast as any GUI application is installed.

@caio1007 
> Hello, using the script for version 9.16 can now be considered stable for production use? My system is (Native PC x86_64).

On x86_64 definitely safer than on any SBC. But there is no 100% guarantee that everything will work OOTB, especially 3rd party packages/APT repositories, which is why such distro upgrade is never automated, but with separate script and separate support issue here 😉. However, Debian Trixie itself has been released, hence is stable, and we tested all software options systematically. If any issue occurs, we can help quickly.

@danmoo117
The RPi repo key is missing? That is weird. Did you run a DietPi update or `apt update` any recently? That should have both failed with same error 🤔. Easy to fix, but lets check the situation, so I do not interpret something wrong:
```sh
dpkg -l raspberrypi-archive-keyring
ls -l /etc/apt/trusted.gpg.d/raspberrypi-archive.asc
```
This package is supposed to be installed on all RPis and contains the key. And this is also what I would suggest, instead of downloading (and keeping updated) the key manually. However, to bootstrap it as the key is not present, the package needs to be downloaded manually once. But as said, I wonder how this could not appear earlier.

danmoo117, 2025-08-27:
> [@danmoo117](https://github.com/danmoo117) The RPi repo key is missing? That is weird. Did you run a DietPi update or `apt update` any recently? That should have both failed with same error 🤔. Easy to fix, but lets check the situation, so I do not interpret something wrong:
> 
> dpkg -l raspberrypi-archive-keyring
> ls -l /etc/apt/trusted.gpg.d/raspberrypi-archive.asc
> This package is supposed to be installed on all RPis and contains the key. And this is also what I would suggest, instead of downloading (and keeping updated) the key manually. However, to bootstrap it as the key is not present, the package needs to be downloaded manually once. But as said, I wonder how this could not appear earlier.

Looks to be missing...

```shell
dpkg-query: no packages found matching raspberrypi-archive-keyring
ls: cannot access '/etc/apt/trusted.gpg.d/raspberrypi-archive.asc': No such file or directory
```
out of curiosity I updated another pi and the same has occurred. Didn't have an issue with apt update/upgrade prior to running the script.

[87 more replies]