MichaIng/DietPi #7652: Update 9.15.2 failed

niwre2019, 2025-07-31:
Hi,

unfortunately the update to the new version 9.15.2 does not work for me, I get the following error message.

[image]

What could be the reason for this?

Best, Niwre

MichaIng, 2025-07-31:
Thanks for reporting. You get the same error when running this command, right?
```sh
curl -I https://github.com
```
Can you show the output of this command:
```sh
echo | openssl s_client -connect github.com:443
```

MichaIng, 2025-07-31:
Seems to have solved itself. To be sure, try to download the exact archive before retrying the `dietpi-update`:
```sh
cd /tmp
curl -sSfLO 'https://github.com/MichaIng/DietPi/archive/master.tar.gz'
```

niwre2019, 2025-07-31:
Strange, that doesn't work.

> cd /tmp
curl -sSfLO 'https://github.com/MichaIng/DietPi/archive/master.tar.gz'
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.

niwre2019, 2025-07-31:
`curl -vfLO 'https://github.com/MichaIng/DietPi/archive/master.tar.gz'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 140.82.121.4:443...
* Connected to github.com (140.82.121.4) port 443 (#0)
* ALPN: offers h2,http/1.1
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [3135 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [79 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=github.com
*  start date: Feb  5 00:00:00 2025 GMT
*  expire date: Feb  5 23:59:59 2026 GMT
*  subjectAltName: host "github.com" matched cert's "github.com"
*  issuer: C=GB; ST=Greater Manchester; L=Salford; O=Sectigo Limited; CN=Sectigo ECC Domain Validation Secure Server CA
*  SSL certificate verify ok.
} [5 bytes data]
* using HTTP/2
* h2h3 [:method: GET]
* h2h3 [:path: /MichaIng/DietPi/archive/master.tar.gz]
* h2h3 [:scheme: https]
* h2h3 [:authority: github.com]
* h2h3 [user-agent: curl/7.88.1]
* h2h3 [accept: */*]
* Using Stream ID: 1 (easy handle 0x55a3abadc7a0)
} [5 bytes data]
> GET /MichaIng/DietPi/archive/master.tar.gz HTTP/2
> Host: github.com
> user-agent: curl/7.88.1
> accept: */*
> 
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* old SSL session ID is stale, removing
{ [5 bytes data]
< HTTP/2 302 
< date: Thu, 31 Jul 2025 13:18:38 GMT
< content-type: text/html; charset=utf-8
< content-length: 0
< vary: X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, X-Requested-With,Accept-Encoding, Accept, X-Requested-With
< location: https://codeload.github.com/MichaIng/DietPi/tar.gz/refs/heads/master
< cache-control: max-age=0, private
< strict-transport-security: max-age=31536000; includeSubdomains; preload
< x-frame-options: deny
< x-content-type-options: nosniff
< x-xss-protection: 0
< referrer-policy: no-referrer-when-downgrade
< content-security-policy: default-src 'none'; base-uri 'self'; child-src github.githubassets.com github.com/assets-cdn/worker/ github.com/assets/ gist.github.com/assets-cdn/worker/; connect-
[... 26 lines cut ...]
* Connected to codeload.github.com (127.0.0.1) port 443 (#1)
* ALPN: offers h2,http/1.1
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [6 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [1210 bytes data]
* TLSv1.3 (OUT), TLS alert, unknown CA (560):
} [2 bytes data]
* SSL certificate problem: unable to get local issuer certificate
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Closing connection 1
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above. `

MichaIng, 2025-07-31:
Eh wait ... IP `127.0.0.1`? Do you use an ad blocker like Pi-hole or so with this server?

niwre2019, 2025-07-31:
That was the problem, I'm very sorry for wasting your time.

Can I delete my personal data (URL) from the last post?

MichaIng, 2025-07-31:
> Can I delete my personal data (URL) from the last post?

Oh yes that makes sense to edit out.

tindecken, 2025-07-31:
Hi @MichaIng 

I've just tried to run `dietpi-update` without any checking and it returned successful
I didn't use Pi-hole or AdGuard Home

```
 dietpi-update   : Run now to update DietPi from v9.14.2 to v9.15.2

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@Tindecken:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking IPv6 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercont                       ent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.15.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v9.14.2
[ INFO ] DietPi-Update | Latest version  : v9.15.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=20082 MiB | requir                       ed=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : minio
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php8.2-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : proftpd

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 k                       B]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [59.4 kB]
Hit:5 https://download.docker.com/linux/debian bookworm InRelease
Hit:6 https://pkg.cloudflare.com/cloudflared any InRelease
Get:7 https://deb.debian.org/debian-security bookworm-security/main arm64 Packag                       es [268 kB]
Hit:8 https://dietpi.com/apt bookworm InRelease
Hit:9 https://dietpi.com/apt all InRelease
Ign:10 https://pkg.jenkins.io/debian binary/ InRelease
Hit:11 https://pkg.jenkins.io/debian binary/ Release
Get:13 https://archive.raspberrypi.com/debian bookworm InRelease [55.0 kB]
[... 73 lines cut ...]
Purging configuration files for linux-image-6.12.25+rpt-rpi-v8 (1:6.12.25-1+rpt1                       ) ...
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | systemctl daemon-reload
[  OK  ] DietPi-Update | Incremental patching to v9.15.2 completed
[ INFO ] DietPi-Update | Checking for new available live patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v9.15.2
[ INFO ] DietPi-Update | Latest version  : v9.15.2
[  OK  ] DietPi-Survey | Sending survey data
[  OK  ] DietPi-Update | sync
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : proftpd
[  OK  ] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php8.2-fpm
[  OK  ] DietPi-Services | restart : nginx
[  OK  ] DietPi-Services | restart : minio
[  OK  ] DietPi-Services | restart : docker
[  OK  ] DietPi-Services | restart : cron
root@Tindecken:~#
```

MichaIng, 2025-07-31:
Since it is solved for both of you, I am closing the issue. If we receive other reports, I'll probably reopen and merge them here.

closed 2025-07-31 by MichaIng