MichaIng/DietPi #7654: Upgrading an old system in-place?

danergo, 2025-08-03:
I have an old system on an RPi1. TBH it's a compute module without SD card, so dietpi is running from local eMMC.

By that time we installed it, it was Debian Stretch.

Now we wanted to update it, but at some point it failed, and now we are stuck:

```
# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

# cat /etc/debian_version 
9.13

DietPi v9.2.0

# cat /etc/apt/sources.list
#deb http://legacy.raspbian.org/raspbian/ buster main contrib non-free rpi

# cat /etc/apt/sources.list.d/raspi.list 
deb https://archive.raspberrypi.com/debian/ buster main ui

# dpkg -l base-files
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                      Version                   Architecture              Description
+++-=========================================-=========================-=========================-========================================================================================
ii  base-files                                9.9+rpi1+deb9u13          armhf                     Debian base system miscellaneous files

# uname -r
5.10.103+
```

In case of running dietpi-update:

```
APT update
- Command: apt-get -y -eany update
- Exit code: 100
- DietPi version: v9.2.0 (MichaIng/master) | HW_MODEL: 1 | HW_ARCH: 1 |
```

So dietpi-update can't be run.

```
# apt dist-upgrade 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  openssh-sftp-server
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

```

```
# apt-cache policy base-files
base-files:
  Installed: 9.9+rpi1+deb9u13
  Candidate: 9.9+rpi1+deb9u13
  Version table:
 *** 9.9+rpi1+deb9u13 100
        100 /var/lib/dpkg/status
```

What would worth trying?

This RPi1 is far away, I have no physical contact option. Only remote SSH.

Joulinar, 2025-08-03:
Can you share the entire error message?

danergo, 2025-08-03:
Anyway, this seemed rather problematic:

```
# cat /etc/apt/sources.list
#deb http://legacy.raspbian.org/raspbian/ buster main contrib non-free rpi
```

This is supposed to be the base debian (raspbian) system, and something has commented it out.
I have now changed this to:

```
# cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian buster main contrib non-free rpi
```

And did an `apt update`, and `apt list --upgradable`: head of the results:

```
adduser/oldoldstable 3.118 all [upgradable from: 3.115]
apt/oldoldstable 1.8.2.3 armhf [upgradable from: 1.4.11]
apt-transport-https/oldoldstable 1.8.2.3 armhf [upgradable from: 1.4.11]
apt-utils/oldoldstable 1.8.2.3 armhf [upgradable from: 1.4.11]
autoconf/oldoldstable 2.69-11 all [upgradable from: 2.69-10]
automake/oldoldstable 1:1.16.1-4 all [upgradable from: 1:1.15-6]
autotools-dev/oldoldstable 20180224.1 all [upgradable from: 20161112.1]
base-files/oldoldstable 10.3+rpi1+deb10u13 armhf [upgradable from: 9.9+rpi1+deb9u13]
...
```

But as you can see, packages are now going to be updated from deb9uxy to deb10uxy, so I believe this is the correct way of the upgrade from stretch to buster. Please correct me if I'm being wrong here.

MichaIng, 2025-08-03:
First of all, please comment or remove the line in `/etc/apt/sources.list` again. If you run any `apt upgrade` now, you would do an unintended Debian distro migration from Stretch to Buster. You will need to do that as a next step, but let's first get DietPi on the intended branch and version. Probably `raspbian-archive-keyring` adjusted it, since the `stretch` suite has been removed from the Raspbian repository: http://raspbian.raspberrypi.org/raspbian/dists/stretch/ (empty)
Also, comment the line in `/etc/apt/sources.list.d/raspi.list` do avoid mixing distros. Or alternatively replace "buster" there with "stretch", since the RPi repo still hosts the `stretch` suite: https://archive.raspberrypi.org/debian/dists/stretch/

You are on a DietPi branch that is not compatible with Debian Stretch anymore. If you run `dietpi-update`, it should switch from `master` to `8` branch at first, and then again from `8` to `stretch` branch, since the former also requires Debian Buster as minimum.

A problem now is that you are on DietPi v9.2.0 already, while the `8` branch is latest DietPi v8, and the `stretch` branch is latest DietPi v7. So `dietpi-update` won't find an available update on the fallback branches. An update to DietPi v9 or even v8 should have been prevented by the automatic branch switch, so I am not sure how the upgrade to DietPi v9.2.0 was possible. Without the complete console output we won't be able to find out.

Do you remember from with DietPi version you were updating?

Please try the following:
```sh
G_CONFIG_INJECT 'G_DIETPI_VERSION_CORE=' 'G_DIETPI_VERSION_CORE=7' /boot/dietpi/.version
G_CONFIG_INJECT 'G_DIETPI_VERSION_SUB=' 'G_DIETPI_VERSION_SUB=9' /boot/dietpi/.version
dietpi-update 1
```
`dietpi-update` will then think you are on DietPi v7.9.0. It should see that your Debian version is too old, switch to the `stretch` branch, see that v7.9.4 is available there, and perform the update.

Let me know once done, then you could proceed with the upgrade to Debian Buster.

_____________

Generally, since you would need to do 3 distro upgrades to be on current stable Debian 12 Bookworm, and Debian 13 Trixie is released in 6 days, I actually wonder whether starting over with a fresh DietPi Bookworm or even Trixie image isn't the better approach? However, I updated our compat branches and upgrade scripts (available from Bullseye => Bookworm on) just recently, so would be somewhat interested to see a real world example of upgrading through all of them, instead of local test VMs only.

danergo, 2025-08-03:
Thank you!

I'm almost finished finished with upgrading to buster (manually, without dietpi).

What would be my correct update-path now?

MichaIng, 2025-08-03:
I added the recursive compat branch switch now, to prevent that very old Debian versions get upgraded to too new DietPi versions in the future. But I recognized another issue in that case: Since the old Debian version is not getting a proper integer ID, if the DietPi version is too new already, the compat branch switch was skipped. This behavior was intended when added, to conservatively skip any switch if the distro version detection failed: https://github.com/MichaIng/DietPi/commit/1da4071
However, in this situation it is causing issues. Also, switching to a compat branch in case of a failed distro detection does not harm, since a downgrade is never performed. And the intermediate DietPi update might well fix the distro detection, in which case `dietpi-update` switches back to `master` branch automatically. I hence reverted that change: https://github.com/MichaIng/DietPi/commit/5f3f87a

So in your case, since you are on DietPi v9 already, this might actually prevent the switch to the intended compat branch, and would explain the whole situation in general. To fix it:
```sh
sed -i 's/\$G_DISTRO && \$G_DISTRO/\$G_DISTRO/' /boot/dietpi/dietpi-update
```
and then as of above:
```sh
G_CONFIG_INJECT 'G_DIETPI_VERSION_CORE=' 'G_DIETPI_VERSION_CORE=8' /boot/dietpi/.version
G_CONFIG_INJECT 'G_DIETPI_VERSION_SUB=' 'G_DIETPI_VERSION_SUB=25' /boot/dietpi/.version
dietpi-update 1
```

danergo, 2025-08-03:
Did all these now, but `-eany` is not understood:

[image]

danergo, 2025-08-03:
#### Details:
- Date           | 2025-08-03 20:03:36
- Program name   | DietPi-Update
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- DietPi version | v8.25.0 (MichaIng/master)
- Distro version | Unknown (ID=0,RASPBIAN=1)
- Kernel version | `Linux hs 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux`
- Architecture   | `armhf`
- Hardware model | RPi CM (armv6l) (ID=1)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
- Pre-image      | Raspbian Lite
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Command line option 'e' [from -eany] is not understood in combination with the other options.
```

danergo, 2025-08-03:
Also:

```
sed -i 's/\$G_DISTRO && \$G_DISTRO/\$G_DISTRO/' /boot/dietpi/dietpi-update
G_CONFIG_INJECT 'G_DIETPI_VERSION_CORE=' 'G_DIETPI_VERSION_CORE=8' /boot/dietpi/.version
[  OK  ] G_CONFIG_INJECT | Desired setting in /boot/dietpi/.version was already set: G_DIETPI_VERSION_CORE=8
G_CONFIG_INJECT 'G_DIETPI_VERSION_SUB=' 'G_DIETPI_VERSION_SUB=25' /boot/dietpi/.version
[  OK  ] G_CONFIG_INJECT | Desired setting in /boot/dietpi/.version was already set: G_DIETPI_VERSION_SUB=25
dietpi-update 1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/stretch/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 7.9.4
[ INFO ] DietPi-Update | Your Debian version is too low to update to the current branch: Unknown (0)
[ INFO ] DietPi-Update | We're switching to an alternative branch: jessie-support
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/jessie-support/.update/version
curl: (22) The requested URL returned error: 404 
[FAILED] DietPi-Update | Unable to get latest version.
[ INFO ] DietPi-Update | Please check the target Git repository and your network connection, then rerun dietpi-update.
[ INFO ] DietPi-Update | If this error persists, please report at: https://github.com/MichaIng/DietPi/issues
[ INFO ] DietPi-Update | APT update, please wait...
E: Command line option 'e' [from -eany] is not understood in combination with the other options.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -y -eany update
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
```

MichaIng, 2025-08-03:
First of all, the upgrade to Debian Buster was complete?
```sh
cat /etc/os-release
```

danergo, 2025-08-04:
Yes, buster update seems finished correctly:

```
# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

MichaIng, 2025-08-04:
Okay. Please try this to work around the update issue:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=8' /boot/dietpi.txt
G_CONFIG_INJECT 'G_DISTRO=' 'G_DISTRO=5' /boot/dietpi/.hw_model
sed -i 's|apt-get -y -eany update|apt-get -y update|' /boot/dietpi/func/dietpi-globals
dietpi-update 1
```

danergo, 2025-08-04:
I got this now:

#### Details:
- Date           | 2025-08-04 14:44:15
- Program name   | DietPi-Update
- Command        | `apt-get -y update`
- Exit code      | 100
- DietPi version | v8.24.0 (MichaIng/master)
- Distro version | Unknown (ID=5,RASPBIAN=1)
- Kernel version | `Linux hs 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux`
- Architecture   | `armhf`
- Hardware model | RPi CM (armv6l) (ID=1)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
- Pre-image      | Raspbian Lite
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 https://archive.debian.org/debian buster InRelease [122 kB]
Hit:2 https://archive.raspberrypi.com/debian buster InRelease
Err:1 https://archive.debian.org/debian buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY DCC9EFBF77E11517
Reading package lists...
W: GPG error: https://archive.debian.org/debian buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY DCC9EFBF77E11517
E: The repository 'https://archive.debian.org/debian buster InRelease' is not signed.
```

MichaIng, 2025-08-04:
Great, finally 😅. So now you can upgrade to Bullseye with the guide referenced in that message: https://dietpi.com/blog/?p=811#manual-upgrade

The reason I wanted to have the correct DietPi version installed is that you can now do all migration steps from the guide and in case the `dietpi-backup` without errors.