MichaIng/DietPi #7665: NanoPi Neo kernel upgrade results in boot failure (from 6.1.0-37 to 6.1.0-38)

dave0003, 2025-08-13:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
(can't access currently, here's the /etc/os-release)
Debian GNU/Linux 12 (bookworm)

- Kernel version | `uname -a`
Previous working:  vmlinuz-6.1.0-37-rt-armmp
Upgrade not working:  vmlinuz-6.1.0-38-rt-armmp
 
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
NanoPi Neo 512 MB model

- Power supply used | (EG: 5V 1A RAVpower)
5V 3A

- - SD card used | (EG: SanDisk ultra)
Sandisk High Endurance 32GB

#### Steps to reproduce

1. apt update
2. apt upgrade (new kernel)
3. reboot
4. The NanoPi Neo doesn't reappear on the network

#### Expected behaviour

NanoPi Neo should reboot and reappear on the network

#### Actual behaviour

Failing to boot properly after kernel upgrade

#### Extra details

I attached UART adapter to the NanoPi Neo.  Here's the full output from the UART adapter failed boot.

```
U-Boot SPL 2024.01-armbian-2024.01-S866c-P7738-Ha5c2-V2b78-Bb703-R448a (Mar 09 2025 - 03:54:42 +0000)
DRAM: 512 MiB
Trying to boot from MMC1
ns16550_serial serial@1c28000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2024.01-armbian-2024.01-S866c-P7738-Ha5c2-V2b78-Bb703-R448a (Mar 09 2025 - 03:54:42 +0000) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi NEO
DRAM:  512 MiB
Core:  66 devices, 18 uclasses, devicetree: separate
WDT:   Not starting watchdog@1c20ca0
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial,usbkbd
Out:   serial
Err:   serial
Net:   eth0: ethernet@1c30000
starting USB...
Bus usb@1c1a000: sun4i_usb_phy phy@1c19400: External vbus detected, not enabling our own vbus
USB EHCI 1.00
[... 378 lines cut ...]
[    8.737058] hub 8-0:1.0: 1 port detected
[    8.742504] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.12
[    8.750835] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    8.758100] usb usb3: Product: EHCI Host Controller
[    8.763021] usb usb3: Manufacturer: Linux 6.12.23-current-sunxi ehci_hcd
[    8.769775] usb usb3: SerialNumber: 1c1b000.usb
[    8.775605] hub 3-0:1.0: USB hub found
[    8.779490] hub 3-0:1.0: 1 port detected
[    8.795757] Freeing unused kernel image (initmem) memory: 1024K
[    8.802735] Run /init as init process
Loading, please wait...
Starting systemd-udevd version 252.38-1~deb12u1
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Begin: Will now check root file system ... fsck from util-linux 2.38.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p1] fsck.ext4 -a -C0 /dev/mmcblk0p1
/dev/mmcblk0p1: clean, 49775/1904000 files, 774733/7790720 blocks (check after next mount)
done.
mount: No such device
Failed to mount /dev/mmcblk0p1 as root file system.
(initramfs)

```

dave0003, 2025-08-13:
This is the complete history.log from the apt update before it broke.
```
Start-Date: 2025-08-07  21:15:44
Commandline: apt upgrade
Upgrade: openssh-client:armhf (1:9.2p1-2+deb12u6, 1:9.2p1-2+deb12u7), redis-server:armhf (5:7.0.15-1~deb12u4, 5:7.0.15-1~deb12u5), openssh-server:armhf (1:9.2p1-2+deb12u6, 1:9.2p1-2+deb12u7), libssl3:armhf (3.0.16-1~deb12u1, 3.0.17-1~deb12u1), libssl-dev:armhf (3.0.16-1~deb12u1, 3.0.17-1~deb12u1), redis-tools:armhf (5:7.0.15-1~deb12u4, 5:7.0.15-1~deb12u5), openssh-sftp-server:armhf (1:9.2p1-2+deb12u6, 1:9.2p1-2+deb12u7), openssl:armhf (3.0.16-1~deb12u1, 3.0.17-1~deb12u1)
End-Date: 2025-08-07  21:30:47

Start-Date: 2025-08-10  13:17:44
Commandline: apt upgrade
Upgrade: libssl3:armhf (3.0.17-1~deb12u1, 3.0.17-1~deb12u2), libssl-dev:armhf (3.0.17-1~deb12u1, 3.0.17-1~deb12u2), openssl:armhf (3.0.17-1~deb12u1, 3.0.17-1~deb12u2)
End-Date: 2025-08-10  13:17:49

Start-Date: 2025-08-13  19:37:48
Commandline: apt upgrade
Install: linux-image-6.1.0-38-rt-armmp:armhf (6.1.147-1, automatic)
Upgrade: linux-image-rt-armmp:armhf (6.1.140-1, 6.1.147-1), linux-libc-dev:armhf (6.1.140-1, 6.1.147-1)
End-Date: 2025-08-13  19:41:24
```

dave0003, 2025-08-13:
I did a restore to recovery my system.  But I looked at the sd card before I did the restore, and I can confirm the links in /boot were pointing to different versions, some to 6.12.23 and some to 6.1.0-38.  After the restore, they all point to 6.12.23.

I purged these:  linux-image-6.1.0-37-rt-armmp linux-image-rt-armmp

...and it also auto-purged wireguard, which is probably what installed the other kernel version in the first place.

But I can get by with just wireguard-tools, so that's what I'll do.

Thanks for your help.  It's booting properly again now.

MichaIng, 2025-08-13:
Oh, Armbian did that already: https://github.com/armbian/build/commit/94a72b9

I wonder why this did not cause issues before, unlikely to be a coincidence that someone recognized this at both sides within 2 days 🤔? However, solved.

MichaIng, 2025-08-13:
Anyway, since the WireGuard meta package has this dummy dependency, providing it with Armbian kernel package, like the Debian kernel packages do, is the correct fix. While it remains true that `wireguard-tools` is the only required package, you will not be the only one who intuitively installs the `wireguard` meta package instead, and doing so must not break any system which uses a different kernel package, of course.

Other than that, this dummy package dependency should probably removed from the WireGuard package. Since the kernel module is by default part of the Linux kernel since v5.6, IMO the `wireguard` package should become a transitional meta package to pull `wireguard-tools`, and ultimately replaced by it, getting rid of the `wireguard-modules` dummy entirely. `wireguard-dkms` can remain as option, if people use 3rd party kernel builds with WireGuard module disabled, but not pulled in by anything as dependency.