MichaIng/DietPi #7677: Open Beta v9.16 | Please help testing and hardening our upcoming release

MichaIng, 2025-08-16:
RC version | v9.16.0
----------- | -------
v9.16.0 PR | #7660
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2025-08-23**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av9.16
____
### Beta v9.16.0
_(2025-08-16)_

#### Enhancements
- **DietPi-Software** | **Domoticz**: This software option has been enabled for all Debian versions and architectures. We build and distribute own packages via our own APT server for this purpose. All files in `/mnt/dietpi_userdata/domoticz` are treated as `conffiles`, hence manual changes will never be overwritten by package upgrades. For fresh installs, however, the plain HTTP listener at port 8124 has been disabled, hence by default, access is possible via HTTPS on port 8424 only. A `sudoers` config has been added to make the system shutdown and restart options from the web UI work.
- **DietPi-Software** | **Kodi**: On Raspberry Pi Bookworm systems, Kodi 21 is now installed, provided as separate package from the RPi APT repository. If you want to upgrade your previously installed instance: `sudo dietpi-software reinstall 31`
- **DietPi-Software** | **Kubo**: The IPFS Node install option has been renamed to "Kubo". It has been renamed upstream a while ago: https://github.com/ipfs/helia/wiki/FAQ
- **DietPi-Software** | **Java**: Our implementation switched for 64-bit systems from Debian's OpenJDK packages to Adoptium's Temurin Java packages. This allows us and DietPi users to install any released Java version on any Debian version. Previously on older Debian version, some Java applications could not be used anymore, or only older versions. For now, Java 21 will be installed by `dietpi-software`, which is the latest LTS version. But users can add any other version via APT: https://packages.adoptium.net/ui/native/deb/pool/main/t/. However, the Adoptium repository does not support the 32-bit architecture anymore, hence those system will remain with Debian's OpenJDK packages and the related restrictions.
- **DietPi-Software** | **Java**: The Raspberry Pi repository maintainers went through the efforts to provide ARMv6-compatible Java 17 packages. Thanks to this fact, we can re-enable the Java installation options for RPi 1/Zero models, however with the limitations Java 17 has compared to Java 21, at time of writing: Minecraft max v1.20.4 and openHAB max v4.
- **DietPi-Software** | **PaperMC**: On 64-bit Debian Bullseye and Bookworm systems, PaperMC can be updated again to its latest version, using Adoptium Temurin Java 21. Previously, due to minimum Java requirements, PaperMC 1.20.4 was the latest supported version on these systems.
- **DietPi-Software** | **Nextcloud Talk**/**Synapse**/**WireGuard**: It is now possible to install these software options non-interactively, e.g. via automated first boot installs: `AUTO_SETUP_AUTOMATED=1` in `dietpi.txt`. For this, a new dietpi.txt setting `SOFTWARE_PUBLIC_DOMAIN_NAME` has been added to define the (planned) public domain name of the DietPi system, used as default for software options which require or benefit from it. For now, only these software options and `dietpi-letsencrypt` will make use of `SOFTWARE_PUBLIC_DOMAIN_NAME` for non-interactive setups, while installs from an interactive terminal will still show a dialogue to change the default.
- **DietPi-Software** | **Synapse**: In addition to `SOFTWARE_PUBLIC_DOMAIN_NAME`, for Synapse installations, the new `SOFTWARE_SYNAPSE_USERNAME` `dietpi.txt` option can be set to define the initial admin username on non-interactive installs, which defaults to "dietpi". Hence be sure to set this when using `AUTO_SETUP_AUTOMATED=1` and `AUTO_SETUP_INSTALL_SOFTWARE_ID=125`.
- **DietPi-Software** | **WireGuard**: In addition to `SOFTWARE_PUBLIC_DOMAIN_NAME`, for WireGuard installations, the new `SOFTWARE_WIREGUARD_MODE` and `SOFTWARE_WIREGUARD_PORT` `dietpi.txt` options can be set to define whether WireGuard will shall be installed as server or client, and in case which UDP port it shall listen to. The defaults are server mode and port 51820.

#### Bug fixes
- **Raspberry Pi** | Resolved issues due to conflicting APT packages for the Raspberry Pi OS desktop. The related packages are now prevented via APT pins on all Raspberry Pi systems, added as well with this DietPi update. This was previously done only on LXDE install, since its base packages from the RPi repo are highly customized for the RPi OS desktop. But e.g. related GTK packages with conflicting dependencies are pulled in by other desktop and GUI packages as well. Many thanks to @anijatsu and others for reporting related issues: https://github.com/MichaIng/DietPi/issues/6959
- **RISC-V** | The Docker Compose and Snapcast `dietpi-software` options are now available on RISC-V systems.
- **Debian Trixie** | The install of Xfce and GNUstep desktops, as well as enabling Amiberry standard boot via `dietpi-autostart` failed, due the attempt to install the `policykit-1` package, which is a dummy transitional on Bookworm and has been removed on Trixie. Instead, from Bookworm on, the successor `polkitd` package is installed, which provides the same features to elevate privileges for e.g. `poweroff`/`reboot` from desktop and `systemctl` execution. Many thanks to @0xB33r for reporting this issue: https://github.com/MichaIng/DietPi/issues/7650
- **DietPi-BugReport** | Resolved an issue where generating a bug report failed on recent images if (p)7zip was not installed. This became an issue since we stopped to pre-install it on our images. Bug reports are not compress with `xz` if present, else `gzip`. Many thanks to @0xB33r for reporting this issue: https://github.com/MichaIng/DietPi/issues/7650
- **DietPi-Display** | Resolved an issue where display with once stored settings were always shown as disabled, when starting the script again.
- **DietPi-Software** | **VSCodium**: This software options has been disabled on 32-bit ARM systems, since the project stopped distribution packages for those architectures with their APT repository.
- **DietPi-Software** | **Koel**: Resolved an issue where the installation failed on 32-bit systems from Debian Bookworm on, due to a pre-installed dependency which strictly requires 64-bit. The dependency is now downgraded on accepted systems via PHP Composer.
- **DietPi-Software** | **Kavita**: Resolved an issue where the installer tried to exit a non-existing config file, so that the daemon was not listening on port 2036 as expected, but port 5000 instead.
- **DietPi-Software** | **Docker Compose**: This software option has been enabled for the RISC-V architecture.
- **DietPi-Software** | **Syncthing**: Resolved an issue where the installation failed due to CLI changes in the latest release v2.0.0.
- **DietPi-Software** | **Raspotify**: Resolved an issue where Raspotify could cause and endless failure loop on shutdown, effectively preventing the shutdown procedure from ever finishing.
- **DietPi-Software** | **openHAB**: Resolved an issue where the latest package could not run anymore on Debian Bullseye and Bookworm systems, since the minimum Java version requirement raised to Java 21 with openHAB 5. This is now possible on 64-bit systems, as we moved to Adoptium Temurin Java packages. On 32-bit Bullseye and Bookworm systems, the latest openHAB 4 will be installed instead, and the DietPi update performs a downgrade if needed. Many thanks to @MDAR for reporting this issue: https://github.com/MichaIng/DietPi/issues/7657
- **DietPi-Software** | **Nukkit**: Resolved an issue where a reinstall/update overwrote the config file. It is now never overwritten if present.
- **DietPi-Software** | **Node.js**: Resolved an issue where the installed Node.js on ARMv6 Bullseye systems was not functional, since it required a newer C++ ABI version than provided by Raspbian Bullseye.

MDAR, 2025-08-17:
Thanks a million 🥳 

I'll run a copy of Beta V9.16 on an Odroid C4 

Is there anything specific that you'd like me to test  / monitor?

MichaIng, 2025-08-17:
Things where a closer look and test wouldn't hurt:
1. Domoticz: For those who have it installed, the DietPi update installs a new package from our APT server. Just check whether everything works well. It is supposed to have the internal updater disable, which would conflict with the DEB package. We'll update our builds quickly on upstream releases.
2. Kodi on Raspberry Pi: There are new packages provided by the RPi repository, including dedicated Kodi 21 packages for Bookworm, and first packages for Trixie uploaded just some days ago. For those who use it already:
  ```sh
  sudo dietpi-software reinstall 31
  ```
3. Java: 64-bit systems will use Adoptium Temurin Java packages from now on, and I found OpenJDK 17 builds for ARMv6, i.e. Java can again be installed on RPi 1/Zero. For those who use it already:
  ```sh
  sudo dietpi-software reinstall 196
  ```
  Check whether your Java applications and projects work as before. If someone has an idea to compare memory consumption, CPU usage, run benchmarks comparing Debian's OpenJDK builds with Temurin, that would be surely interesting. From our GitHub tests, I can at least say application startup time did not raise, which was pretty long in emulated containers.
4. Debian Trixie: Vast parts of this release are around enhancing Debian Trixie support among install options. So in general, and especially in combination with this beta, testing upgrade to and operation on Debian Trixie is valuable: #7644

matttttttttttttt, 2025-08-17:
when trying to update to the lastest beta of dietpi I get following:

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking IPv6 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/main/.update/version
curl: (22) The requested URL returned error: 404
[FAILED] DietPi-Update | Unable to get latest version.
[ INFO ] DietPi-Update | Please check the target Git repository and your network connection, then rerun dietpi-update.

matttttttttttttt, 2025-08-17:
Thank you, Michal! This worked immediately

MichaIng, 2025-08-20:
Beta v9.16.1 has been merged: #7681

squassina, 2025-08-24:
I just updated and now I'm getting:

```
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:1 and /etc/apt/sources.list.d/dietpi.sources:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:1 and /etc/apt/sources.list.d/dietpi.sources:1
W: Target Packages (nanopineo/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:2 and /etc/apt/sources.list.d/dietpi.sources:2
W: Target Packages (nanopineo/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:2 and /etc/apt/sources.list.d/dietpi.sources:2
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:1 and /etc/apt/sources.list.d/dietpi.sources:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:1 and /etc/apt/sources.list.d/dietpi.sources:1
W: Target Packages (nanopineo/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:2 and /etc/apt/sources.list.d/dietpi.sources:2
W: Target Packages (nanopineo/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi.list:2 and /etc/apt/sources.list.d/dietpi.sources:2
```

I noticed that there was a both /etc/apt/sources.list.d/dietpi.list and /etc/apt/sources.list.d/dietpi.list.bak. The other *.list files were successfully backed-up and migrated to sources.

MichaIng, 2025-08-24:
It seems you ran APT's modernizer at some point? That is not offered on Trixie anymore, is it? It was for a while, but then the notice was removed before the release. I hope all list files have been backed up? Restore them all. We will do a list => sources migration at a later point. No reason to rush, no benefits.

MichaIng, 2025-08-24:
> using the other script

Which other script?

MichaIng, 2025-08-24:
And at which point did this convert your APT lists?

As said, I know from `apt update` calls some months before Debian Trixie was released (and not on Debian Forky/testing), this notice:
```
Notice: Some sources can be modernized. Run 'apt modernize-sources' to do so.
```
But this command is never called by our script, and it should never be called by anything else automatically either.

squassina, 2025-08-24:
I really don't recall running it on this machine, but I have several. Also, the warning just happen after I update to 9.16, not before. That's what caught my attention. (I had issues with the update as I was running pihole + unbound, I reported it in the other thread, so this server may be updated in full to fix the error, and I may have run apt modernize-sources as part of the "fix").

I just update another one (still in bookworm) and the update was fine. I did run `dietpi-update` twice, just to make sure.

MichaIng, 2025-08-24:
> EDIT: Hum, I think I got it! When I reun `apt modernize-sources` it was fine, but the update re-created `dietpi.list` and that couses the warning.

Yes, as said, do not run `apt modernize-sources` in the first place. Our scripts expect/handle and in case create list files only. Migrating means you will run into the same issue earlier or later, and some needed patches to lists might fail to apply during DietPi updates.

So you never ran `apt modernize-sources` manually, and never saw the notice I quoted above? Maybe there is some package which problematically does this automatically 🤔.

squassina, 2025-08-25:
Thanks for the detailed explanation! As I have the .list.bak of the files, I may give the rollback a try and see how it goes.

squassina, 2025-08-28:
just to tell that I reverted the .list files and everything is running fine. No issues at all.