MichaIng/DietPi #7696: CPU scaling available frequencies missing

gitlankford, 2025-08-25:
### Creating a bug report/issue
- [Y] I have searched the existing open and closed issues

#### Required Information
- DietPi version 

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version

bookworm

- Kernel version

Linux 6.12.35-current-rockchip64 #1 SMP PREEMPT Fri Jun 27 10:11:46 UTC 2025 aarch64 GNU/Linux

- SBC model

NanoPi R2S (aarch64)

- Power supply used

5V 3A USB-C

- SD card used

Samsung EVO Select 256 MB

#### Additional Information (if applicable)
- Software title

dietpi-config

- Was the software title installed freshly or updated/migrated?

both

- Can this issue be replicated on a fresh installation of DietPi?

yes

- Bug report ID

b257dff9-19b4-410b-a580-31be9193a81a

#### Steps to reproduce
1. dietpi-config
2. 3 : Performance Options
3. CPU Frequency Limits
4. Try to select a limit, but no options available.

#### Expected behaviour
A list of available CPU frequencies should be choosable.

#### Actual behaviour
No options available. Plus error message to terminal is

/boot/dietpi/dietpi-config: line 1400: /sys/devices/system/cpu/cpufreq/policy[0-9]*scaling_available_frequencies: No such file or directory

#### Extra details
Not sure when this was introduced, but it was before 9.14.2 (reproduced on such a system that has not been updated past that version yet)

MichaIng, 2025-08-25:
A slash is missing. It should have been
```
/sys/devices/system/cpu/cpufreq/policy[0-9]*/scaling_available_frequencies
```
I will push a live patch to fix this for DietPi v9.16 as well.

MichaIng, 2025-08-25:
Live patch merged as well. So those who did not change the particular line already, `dietpi-update` offers to fix it.

closed 2025-08-25 by MichaIng