MichaIng/DietPi #7701: Rock 5B -

isarrider, 2025-08-27:
The Rock 5B boots with this image just fine from SD Card:
https://dietpi.com/downloads/images/DietPi_ROCK5B-ARMv8-Forky.img.xz
When using a SSD, it wont come up...
So I thought updating the SPI Flash would help,
but it doenst...
I think becaus SSD is only supported with a later Kernel...

Could you plz build a Rock 5B Forky and Trixie image with the latest Mainline Kernel?

isarrider, 2025-09-05:
(Would be even cooler and less work for you if I could do that myself - is this possible)?

MichaIng, 2025-09-06:
Do you need the image to have mainline kernel right from the start? Because joi can just install it from APT afterwards.

isarrider, 2025-09-07:
yes, otherwise the Kernel is too old and the NVMe SSD wont work...
Mainline seems also almost done anyway:
https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md

isarrider, 2025-09-09:
I interpret n/a as sth that the board doenst have, but it seems with a second look this is only party true...
Anyway -> #7337 it is ;)