MichaIng/DietPi #7705: Request: Add DietPi support for Radxa Cubie A5E (Allwinner A527 SoC)

fahimrahmanbooom, 2025-08-28:
Board Information
	•	Board name: Radxa Cubie A5E
	•	SoC: Allwinner A527
	•	Current OS options: Tina SDK Debian-lite, Armbian

Request:

I would like to request official DietPi support for the Radxa Cubie A5E SBC.

This board offers good performance and affordability, and it is starting to gain popularity among developers and hobbyists. Having an optimized DietPi image would:
	•	Provide a lightweight, efficient Debian-based OS
	•	Extend the same minimal footprint and optimizations DietPi offers on other SBCs
	•	Give the community a stable, well-maintained alternative to the current Tina SDK and Armbian builds

References
	•	Armbian has already added support for this board, which might help as a reference.

fahimrahmanbooom, 2025-09-09:
Hi, Is there any update? Thanks.

EvilOlaf, 2026-01-16:
Armbian now uses mainline with 6.18.y kernel for the A5E and a bunch of stuff on top: https://github.com/armbian/build/tree/main/patch/kernel/archive/sunxi-6.18/patches.backports
If it works, no clue. no hw on hand

EvilOlaf, 2026-02-22:
bump to 6.19.y is pending. backports was mostly wiped. Perhaps try that
https://github.com/armbian/build/pull/9381