MichaIng/DietPi #7714: Fresh install bug?

hotFIRESTORM, 2025-09-09:
[image]

RPi400
Fresh install. Ntp server set to europe, the default doesnt work. Everything else is default. Fresh install. What am I missing?

MichaIng, 2025-09-09:
If I may ask, where are you located?

There is a known issue with blocked `raw.githubusercontent.com` from China, and parts of India IIRC: #7555

Can you show the output of this please:
```sh
getent ahosts raw.githubusercontent.com
```

And does this solve the issue for you?
```sh
echo '185.199.111.133 raw.githubusercontent.com' | sudo tee -a /etc/hosts
```

hotFIRESTORM, 2025-09-09:
> If I may ask, where are you located?
> 
> There is a known issue with blocked `raw.githubusercontent.com` from China, and parts of India IIRC: [#7555](https://github.com/MichaIng/DietPi/issues/7555)
> 
> Can you show the output of this please:
> 
> getent ahosts raw.githubusercontent.com
> And does this solve the issue for you?
> 
> echo '185.199.111.133 raw.githubusercontent.com' | sudo tee -a /etc/hosts
> Similarly there was a report with `ntp.org` pool issues from China, even with the dedicated `cn.pool.ntp.org` and also Alibaba time servers: [#7010](https://github.com/MichaIng/DietPi/issues/7010) But that might be a dedicated issue, not sure whether related to DNS. Difficult to check since those pool server domains naturally give a large varying number of different IPs.

I'm located in Slovenia. All settings are default, but guess what. I decided to try again and reformat the usb again and all, and the issue wasnt there, yet i tried multiple times yesterday. Anyways. Solved.

MichaIng, 2025-09-10:
Okay, maybe it was just a temporary issue.