MichaIng/DietPi #7730: Upgrade to Trixie failed

Chriveloper, 2025-09-15:
#### Details:
- Date           | 2025-09-16 00:21:01
- Program name   | dietpi-trixie-upgrade
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- DietPi version | v9.16.3 (MichaIng/master)
- Distro version | bookworm (ID=7)
- Kernel version | `Linux DietPi 6.12.46-current-rockchip64 #1 SMP PREEMPT Tue Sep  9 16:58:27 UTC 2025 aarch64 GNU/Linux`
- Architecture   | `arm64`
- Hardware model | Orange Pi 3B (aarch64) (ID=87)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://download.docker.com/linux/debian trixie InRelease
Hit:2 https://deb.debian.org/debian trixie InRelease
Hit:3 https://deb.debian.org/debian trixie-updates InRelease
Hit:4 https://deb.debian.org/debian-security trixie-security InRelease
Hit:5 https://deb.debian.org/debian trixie-backports InRelease
Hit:6 https://dietpi.com/apt trixie InRelease
Hit:7 https://dietpi.com/apt all InRelease
Ign:8 https://apache.jfrog.io/artifactory/couchdb-deb trixie InRelease
Err:9 https://apache.jfrog.io/artifactory/couchdb-deb trixie Release
  404   [IP: 100.21.156.46 443]
Reading package lists...
E: The repository 'https://apache.jfrog.io/artifactory/couchdb-deb trixie Release' does not have a Release file.
```

MichaIng, 2025-09-16:
That repo does not support Trixie yet: https://apache.jfrog.io/ui/native/couchdb-deb/dists/
Often (but not always) the Bookworm packages work on Trixie. Revert that repo from the error handler to continue the upgrade:
```sh
sed -i 's/trixie/bookworm/g' /etc/apt/sources.list.d/couchdb.list
```
_List name just a guess, adjust accordingly if needed._

Looks like they merged support for Trixie just yesterday, hence next release, I guess will be added to a new Trixie APT suite as well: https://github.com/apache/couchdb-ci/pull/85

MichaIng, 2025-11-16:
The repository supports Debian Trixie in the meantime. If you applied above workaround, you can revert it:
```sh
sed -i 's/bookworm/trixie/g' /etc/apt/sources.list.d/couchdb.list
```