MichaIng/DietPi #7731: Home Assistant - fresh install on Trixie missing components

Dynamic5912, 2025-09-16:
Hi,

Following a fresh install of Debian Trixie on my RPi4, I've installed Home Assistant from fresh (not with any backup restore) and it seems some key components are missing following the completion of the installation:

[image]

There seems to be a reliance on asset_pipeline, as shown below:

[image]

[image]

Which looks like it requires ['pymicro-vad==1.0.1', 'pyspeex-noise==1.0.2'], however pyspeex-noise fails to install as per below error in Home Assistant:

[image]

No error is shown for a failed install of pymicro-vad though...

Joulinar, 2025-09-16:
Similar to this one https://dietpi.com/forum/t/home-assistant-unable-to-unable-to-install-package-pyspeex-noise-1-0-2-and-other-errors/24135

you would need to install `g++` package

```
apt install g++
```

Dynamic5912, 2025-09-16:
When you first open Home Assistant as a brand new user (as I did for testing) it seems the Voice Assistant is pre-installed as it asks you (as one of the first setup questions) if you want to use it as the default Voice Assistant service - not sure when it became a mandatory requirement to have.

closed 2025-09-20 by MichaIng