MichaIng/DietPi #7733: Home Assistant - Cannot Restore Backup

Dynamic5912, 2025-09-16:
Attempting to restore my Home Assistant Backup (either a local archive backup or cloud backup) results in the following error:

[image]

MichaIng, 2025-09-16:
Try to extract some more details from `journalctl -u home-assistant`. I'll try to find the relevant code parts upstream:
- If the backup contains the Python environment from within `./deps` even if that is a symlink, it should also restore the environment to the target of the symlink.
- If it backs up the symlink as symlink, not its content, it should also restore that symlink ... though, that would break it as well if the Python version changed in the meantime from a `dietpi-software reinstall 157`.
- The preferred and IMO most reasonable behavior would be to skip `./deps` completely, since that is never required and has only the potential to break things, even without symlink: the Python version may have changed, which does not only change the symlink in our case, but generally means that most modules need to be recompiled. If a backup from HA where the Python version was a different one requires modules from a different Python version, many won't work.

Since we use this symlink since years, the upstream change causing this must have been done very recently.

Dynamic5912, 2025-09-18:
> Try to extract some more details from `journalctl -u home-assistant`.

There's a lot - more than the terminal screen can handle!!

Any way to dump it out to a file/filter it for certain terms you're looking for in particular?

Dynamic5912, 2025-09-21:
Hi @MichaIng - let me know what you need for further analysis :)

MichaIng, 2025-09-26:
Somewhat related to the general concerns I have with `deps` being included in the backup: https://github.com/home-assistant/core/issues/130396

Python version and changed OS runtime libraries can break it, and restoring on a different system as aim to transfer the HA instance usually cannot work. When skipping `deps`, that should all be possible, with HA compiling all missing dependencies automatically, based on the loaded components.

I am travelling this weekend, but will try to check a few things:
- Is the backup an archive that can be extracted on the OS to check it's content? So we see how `deps` is included and might be able to identify/narrow down the cause of the issue, i.e. whether `deps` is included as directory with content or as symlink.

steals, 2025-12-02:
I've faced the same issue trying to migrate from HA installed on Raspberry PI to x86_64 PC. DietPI is installed on both systems.
If I extract `homeassistant.tar.gz` backup archive I see two folders: `deps` that is empty and `deps_venv` that is actually a symlink to `/home/homeassistant/.pyenv/versions/3.13.1/lib/python3.13/site-packages`
Will try to remove both from the archive and see what happens

Dynamic5912, 2026-01-09:
> I ended up with manual extraction of the backup and placing it in the HA folder. As the recent versions of the HA backups are encrypted and therefore if I modify something in the backup archive I need to re encrypt it again. For me it was more complicated than manual extraction and copy all internal files into the right folder.

Which folders did you need to copy to?