MichaIng/DietPi #7738: Top USB not working on Orange Pi 5 Max

DiegoBM, 2025-09-19:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues
The only issue that I reckon seemed to mention USB was https://github.com/MichaIng/DietPi/issues/7338#issuecomment-2570541908

#### Required Information
- DietPi version | v9.16.3
- Distro version | trixie
- Kernel version | Linux DietPi 6.1.115-vendor-rk35xx #1 SMP Tue Jun  3 13:10:51 UTC 2025 aarch64 GNU/Linux
- SBC model | Orange Pi 5 Max (aarch64)
- Power supply used | 5V 3A
- SD card used | Sandisk Extreme 64GB

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Flash lastest dietpi
2. Plug keyboard on top-left usb 3 port
3. Boot the distro

#### Expected behaviour
- Keyboard should work

#### Actual behaviour
- There is no response

#### Extra details
- `dmesg -w | grep usb` shows no events on that port only

DiegoBM, 2025-09-19:
lsusb with the keyboard connected to the top port
```
root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

lsusb with the keyboard connected to another port
```
root@DietPi:~# lsusb
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 045e:0750 Microsoft Corp. Wired Keyboard 600
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

pavel-kamaev, 2025-11-21:
It looks like this port is configured in device mode. Also it has separate power supply which is switched off by default (and it should be off in the device mode).

This issue also applies to Orange Pi 5 Ultra.

DiegoBM, 2025-11-21:
That's great news!
For those of us whom don't know much about building dietpi, is there any simple and easy to grasp tutorial out there that teaches you how to build linux for sbc with overlays and modifications? Hopefuly explaining those (and all related) conceps as well, so that we can build it ourselves?
Although I can see it has been included to the roadmap :)

MichaIng, 2025-11-21:
No need to rebuild Linux. Just run the above command to enable the overlay. All that is needed our end is doing this by default for our affected Orange Pi 5 variant images, since I find it very unreasonable to have a USB-A port disabled by default for some potential rare device mode usage, and ADB does not exist on Linux.

DiegoBM, 2025-11-21:
Thank you @MichaIng. And one of the 2 usb 3 at that! Not cool