MichaIng/DietPi #7741: Errors when doing a reboot

zordaz, 2025-09-20:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version: v9.6.13
- Distro version: trixie 0
- Kernel version: Linux DietPi 6.12.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.47-1+rpt1 (2025-09-16) aarch64 GNU/Linux
- SBC model: RPi 3 Model B (aarch64)
- Power supply used: unknown
- SD card used: 16GB Transcend

#### Additional Information (if applicable)
N/A

#### Steps to reproduce
Reboot the RPI with one of the following commands.

#### Expected behaviour
The system should reboot without error messages, instead it should give a proper 'rebooting' alike message. It might be a cosmetic error, but IMO it should be solved anyway.

#### Actual behaviour
When doing a reboot via 'shutdown -r now' or 'poweroff --reboot'  or 'systemctl reboot' only an ugly error is shown before the SSH connection is lost (partly in Dutch due to my systems locale):

Failed to connect to system scope bus via local transport: Bestand of map bestaat niet
Failed to connect to system scope bus via local transport: Bestand of map bestaat niet

#### Extra details

- ...

zordaz, 2025-09-20:
It's a fresh DietPi/Trixie install with an absolute minimum of adjustments (mainly fixed IP, SSH instead of Dropbear), so no different shell is being used. My former Bookworm installation also gave (different) errors when doing a 'shutdown -r now'.

```
sudo systemctl status systemd-logind
○ systemd-logind.service
     Loaded: masked (Reason: Unit systemd-logind.service is masked.)
    Drop-In: /usr/lib/systemd/system/systemd-logind.service.d
             └─dbus.conf
     Active: inactive (dead)
```

I haven't touched the systemd config at all.

MichaIng, 2025-09-20:
Can you show this:
```sh
declare -f poweroff
```

MichaIng, 2025-09-20:
And
```
poweroff --reboot
```
without any other argument throws these errors?

zordaz, 2025-09-20:
> When doing this as root or non-root user?
As a non root user. 

In the meantime I've tested it as root and then the system reboots without the error. 
polkitd isn't installed.

MichaIng, 2025-09-20:
Now I see it. Only in a login shell, systemd seems to try getting the needed privileges, maybe from polkitd (regardless whether installed or not), but as well via DBus.

A bit annoying that there is no way to configure it skipping DBus calls in general. I wished it would be a lot more modular with all those additional features optional. But Lennart made clear that for him this is all needed. I mean makes sense, this opinion reflected in systemd design/development in general is the major point of criticism from the Linux community, and why there is, despite the obvious benefits, still a strong core of people who consequently avoid systemd and use classic SysV or other simpler (in design, not usage) init systems.

So as you can see, those errors are visual only, and regular reboot/shutdown works just fine with root permissions, as expected.

If this annoys you for some reason, to get rid of them, you need logind and DBus:
```sh
systemctl unmask systemd-logind
apt install dbus
systemctl start dbus systemd-logind
```
After this, systemd can do all the stuff it tries to do via DBus and logind before performing the shutdown/reboot, hence won't throw these errors anymore.

MichaIng, 2025-09-26:
Jep I agree to both: Debian already splits parts of systemd into dedicated packages which are not usually all built together, but these unconditional DBus/logind calls for these very basic commands show well how limited the possibilities are, and how much several components are increasingly tied together, which could be all made optional by design.

And more components involved means more points of failures and more levels at which logs can be emitted. Like here:
- Dbus must be installled and active, else it always throws an error that it is missing, not saying for what it is called in the first place.
- logind must be active, else DBus throws an error that it is not there, AFAIK then saying as well that it was called for wall measages.
- When disabling wall messages via CLI the error does not change, since the CLI option does not prevent the logind call, but just tells logind to not send the wall message, while the call is done for at least shutdown inhibitors as well, which is however not shown. So error is irritating when actively disabling wall messages.
- Knowing what's wrong and why is rather difficult and unexpected for such a basic command.