MichaIng/DietPi #7750: DietPi installation 'resetting' (almost back to factory)

mdr92, 2025-09-24:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.17.2` (although happened over the last few versions)
- Distro version | `bookworm`
- Kernel version | `6.1.0-40-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.153-1 (2025-09-20) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)'  (N100 mini PC)
- Power supply used | Factory 12v/3a
- SD card used | Intel Octane 16GB NVME

#### Additional Information (if applicable)
About two months ago (probably on 9.14/9.15) I was out and connected to my wireguard instance and logged into ssh remotely and it was on the initial setup window/first software install screen that couldn't be escaped and was back each time I logged in so I just re-installed fail2ban to get past it. It obviously wasn't a clean install as I was connected to my wireguard. Once I'd re-installed fail2ban and it rebooted I could no longer connect until I got home and did a physical restart. OpenSSH had also switched back to dropbear. I switched that back again and all seemed to be ok for a while

Fast forward to this week and I couldn't connect to my wireguard again, on getting back and connecting ssh was back to dropbear again and none of my services (pi-hole, homeassistant, wireguard, proftpd) were running, it now looks like after a reboot I have to go to dietpi-services and restart all services to get them running.

Any ideas? I've done a scan and repair on the drive but it does it at reboot and there were no messages so I'm not sure if it found anything but it didn't sort the issue. I've taken backups of what I can but haven't restored it as if it's corruption then it will presumably just restore that too.

MichaIng, 2025-09-26:
Please show the output of these commands:
```
systemctl
df /tmp
```
RAM logging is shown as installed, but journald logs to `/var/log` which is a problem if that is a tmpfs and cleared hourly. If you need persistent system logs, please uninstall `dietpi-ramlog`, otherwise delete `/var/log/journal` so that journald logs to `/var/run/journal` instead (this is done automatically then). So we can see system logs and check which services start on boot, how long they take and why.

MichaIng, 2025-09-26:
Ah sorry I meant `df /var/log`. Yeah the journal is working, but with RAM logging enabled, it is cleared every hour, which cuts possible startuo logs of services or related errors, and also prevents Fail2Ban from functioning properly, since it can track login failures for only max 1 hour.

How do you know the SSH server reverted to Dropbear? This is nearly impossible without the whole rootfs reverting to an old state like when you restore a backup. Or was ot only that `dietpi-software` falsely showed Dropbear (the `.installed` output above contains OpenSSH, not Dropbear) while in fact it was still OpenSSH? Generally if the SSH server really changes, SSH clients would show a prompt that the host key changed which you need to confirm interactively.

Regarding service starts, interesting would be the timestamps to see what takes so long. Maybe also `journalctl` helps to see all other boot steps in case one is hanging. But for that journald should log to /run/log instead of /var/log or RAM log should to be disabled/uninstalled, so the journal is not possibly cleared before you read it.

mdr92, 2025-09-28:
```
root@DietPi-NUC:~# df /var/log
Filesystem     1K-blocks  Used Available Use% Mounted on
tmpfs              51200  1776     49424   4% /var/log
```

Maybe I'm wrong about ssh as it was a little while ago, but I'm 90% sure it had changed at the same time as I got the initial setup wizard screen and I had to go into dietpi-software and change it again (and WinSCP couldn't connect to SFTP until I switched too).

I can turn ramlog off I guess, I didn't manually turn it on though, it's as it is out the box, I thought that was recommended for flash storage?

I've checked journalctrl and it's absolutely full of this for proftpd and proftpd doesn't show as running in htop. Once I restart from dietpi-services it shows as running htop and there's no more instances of the below in journalctrl and all services restart straight away too rather than slowly like after a reboot. 

[image]

mdr92, 2025-10-02:
Ok, I've deleted /var/log/journal, added debuglevel to proftpd.conf, I checked and that's the correct name, there wasn't one in there so I just added it. In the journalctrl logs the first mention of proftpd is that deactivated error above. But if I check the proftpd log (/var/log/proftpd/proftpd.log) I get this so it looks like something else is grabbing port 21 but I'm not sure what, and why restarting all services sorts it either.

[image]